esp-rs/espflash #321: Flash size detection not working properly

JuliDi, 2022-12-14:
I mentioned this in issue https://github.com/esp-rs/espflash/issues/146, but I think it should get its own issue.
Copying what I wrote there:

If I use `espflash` directly, it correctly determines the flash size of 4 MB and the program, which is around 1 MB, uses up around 25% of the flash.
```bash
$ cargo run                                    
    Finished dev [optimized + debuginfo] target(s) in 0.15s
     Running `espflash --monitor target/riscv32imc-esp-espidf/debug/esp_pid`
New version of espflash is available: v2.0.0-rc.2

Detected 2 serial ports. Ports which match a known common dev board are highlighted.

Serial port: /dev/cu.usbmodem4201
Connecting...

Chip type:         ESP32-C3 (revision 3)
Crystal frequency: 40MHz
Flash size:        4MB
Features:          WiFi
MAC address:       34:b4:72:80:1d:e0
App/part. size:    992624/4128768 bytes, 24.04%
```

If I use `cargo espflash` on the same device however, it determines the flash size **correctly** but then my almost 1 MB program uses up 95% of the flash. If I add some more std bloat, I easily get beyond 1 MB and will refuse to flash.

```bash
cargo espflash --monitor /dev/cu.usbmodem4201                   
New version of cargo-espflash is available: v2.0.0-rc.2

Serial port: /dev/cu.usbmodem4201
Connecting...

    Finished dev [optimized + debuginfo] target(s) in 0.61s
Chip type:         ESP32-C3 (revision 3)
Crystal frequency: 40MHz
Flash size:        4MB
Features:          WiFi
MAC address:       34:b4:72:80:1d:e0
App/part. size:    992624/1048576 bytes, 94.66%
```

If I turn off optimization in debug build to make it exceed 1 MB, the following happens:

```bash
cargo espflash --monitor /dev/cu.usbmodem4201
New version of cargo-espflash is available: v2.0.0-rc.2

Serial port: /dev/cu.usbmodem4201
Connecting...

    Finished dev [unoptimized + debuginfo] target(s) in 0.72s
Chip type:         ESP32-C3 (revision 3)
Crystal frequency: 40MHz
Flash size:        4MB
Features:          WiFi
MAC address:       34:b4:72:80:1d:e0
App/part. size:    1748400/1048576 bytes, 166.74%
Error: 
  × Supplied elf image is too big and doesn't fit configured app partition
```

using espflash directly it works, but the **code will panic** a few lines into `main()`

```bash
cargo run       
    Finished dev [unoptimized + debuginfo] target(s) in 0.18s
     Running `espflash --monitor target/riscv32imc-esp-espidf/debug/esp_pid`
New version of espflash is available: v2.0.0-rc.2

Detected 2 serial ports. Ports which match a known common dev board are highlighted.

Serial port: /dev/cu.usbmodem4201
Connecting...

Chip type:         ESP32-C3 (revision 3)
Crystal frequency: 40MHz
Flash size:        4MB
Features:          WiFi
MAC address:       34:b4:72:80:1d:e0
App/part. size:    1748400/4128768 bytes, 42.35%
```

This is very concerning imo and I cannot tell whether the panic is due to disabled optimization (which would seem like a small disaster to me) or due to espflash doing something wrong.
Any ideas what is happening here?

MabezDev, 2022-12-14:
To me this looks like espflash and cargo espflash are using different values for the total size available. espflash is using the total flash space to calculate the percentage free, but cargo-espflash is (correctly imo) using the partition size.

The default partition size is 1MB and your debug build exceeds that, once flashed the bootloader will realise the app doesn't fit in the partition and will fail to boot.

JuliDi, 2022-12-14:
Well, it does boot but then panics. Thats not very desirable either.
Is there a way to make use of more than 1 MB for my program? How much space is the bootloader taking up? Shouldn't be 3 MB...

JuliDi, 2022-12-14:
Alright, so if I supply a partition table, I can manage to flash even with `cargo espflash --partition-table <path>`. It will correctly determine the 3MB then (I presume this will also flash the partition table).
The bad news, however, is that if I disable optimizations in debug mode (to force the binary size over 1MB), it will still panic shortly after entering main.

ianrgraham, 2022-12-18:
Thanks @MabezDev for the link to the docs! I was similarly confused when I started messing around with both esp-hal and esp-idf-hal and I saw that I had a quarter of the flash space when swtiching to esp-idf-hal. 

While I see now that there's plenty of documentation on changing the partitioning to solve my problem, it wasn't clear that I needed to do that. Could the documentation be updated to make the solution easier to find (I'm sure it's a pretty common mistake for esp noobies like me), or better yet the output of espflash could advise the user to modify the partition table if it doesn't fit in the app partition?

The current message tells you the the partitions is of course too small, but doesn't give any solutions.

> Chip type:         ESP32-C3 (revision 3)
Crystal frequency: 40MHz
Flash size:        4MB
Features:          WiFi
MAC address:       68:67:25:ed:26:68
App/part. size:    1111120/1048576 bytes, 105.96%
Error: 
  × Supplied elf image is too big and doesn't fit configured app partition

closed 2023-01-30 by jessebraham