esp-rs/espflash #326: `espflash` and `cargo espflash` are flashing different bootloader

andresv, 2022-12-20:
`espflash` flashes a bootloader that is bundled with `espflash`, but `cargo espflash` takes bootloader from: `target/xtensa-esp32-espidf/release/build/esp-idf-sys-049b1b8192557540/out/build/bootloader/bootloader.bin`

I tried to enable `CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE` and it worked well on my dev machine where I always flash with `cargo espflash`, but this feature did not work for devices that were flashed with `espflash`.

It took me quite some time to figure out that it is because of bootloader.

I noticed that after `cargo espflash` logs start with:
```
I (27) boot: ESP-IDF v4.4.2-dirty 2nd stage bootloader
I (27) boot: compile time 11:54:03
I (27) boot: chip revision: 1
I (30) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (37) boot.esp32: SPI Speed      : 40MHz
I (42) boot.esp32: SPI Mode       : DIO
I (46) boot.esp32: SPI Flash Size : 4MB
I (51) boot: Enabling RNG early entropy source...
I (56) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (75) boot:  1 otadata          OTA data         01 00 0000f000 00002000
I (82) boot:  2 phy_init         RF data          01 01 00011000 00001000
I (90) boot:  3 app_0            OTA app          00 10 00020000 001db000
I (97) boot:  4 app_1            OTA app          00 11 00200000 001db000
```

But after `espflash` they started with:
```
I (476) cpu_start: Pro cpu up.
I (477) cpu_start: Starting app cpu, entry point is 0x40082f2c
0x40082f2c - call_start_cpu1
    at /Users/firmware/.embuild/espressif/esp-idf/v4.4.2/components/esp_system/port/cpu_start.c:160
I (0) cpu_start: App cpu up.
I (491) cpu_start: Pro cpu start user code
I (491) cpu_start: cpu freq: 160000000
I (491) cpu_start: Application information:
I (495) cpu_start: Project name:     libespidf
I (500) cpu_start: App version:      1
I (505) cpu_start: Compile time:     Nov 10 2022 11:54:00
I (511) cpu_start: ELF file SHA256:  0000000000000000...
I (517) cpu_start: ESP-IDF:          v4.4.2-dirty
I (523) heap_init: Initializing. RAM available for dynamic allocation:
```

Not sure what is correct way to solve this confusion, maybe:
* print out which particular bootloader is being flashed
* add more info about his to docs

jessebraham, 2022-12-20:
This has been the case since `1.6.0`, and was added in #186. The behaviour is different between `cargo-espflash` and `espflash` because `espflash` does _not_ build the project and as such has no information regarding the location of the build artifacts. You're able to use the `--bootloader` option to provide a path to `espflash`.

I will document this for `cargo-espflash` for the time being.

jessebraham, 2022-12-20:
I've at least mentioned this auto-detection feature in the README with #327.

jessebraham, 2023-01-10:
I don't think there's really anything actionable here, so I'm going to close this issue. If you feel this is a mistake please feel free to re-open it, but other than mentioning this behaviour in the `README` I'm not sure what more to do. I will at least add some logging output to indicate the bootloader being used, that should be simple enough.

closed 2023-01-10 by jessebraham

andresv, 2023-01-10:
Yes, it would be perfect if it is just logged which one is actually used.

jessebraham, 2023-01-10:
Will be done in the next release, see https://github.com/esp-rs/espflash/pull/339.