esp-rs/espflash #329: espflash v2.0.0-rc.2 doesn't work as a cargo runner anymore

dimpolo, 2023-01-03:
```toml
# .cargo/config.toml
[target.xtensa-esp32s3-none-elf]
runner = "espflash --monitor"
```

```
$ cargo run --release
...
     Running `espflash --monitor target\xtensa-esp32s3-none-elf\release\control`
error: Found argument '--monitor' which wasn't expected, or isn't valid in this context
```

MabezDev, 2023-01-03:
espflash now requires the `flash` subcommand to be supplied too. Example usage here: https://github.com/esp-rs/esp-hal/blob/1f6803776c10cfbb0fa444b45d3d88f5b68d9121/esp32s3-hal/.cargo/config.toml#L2

closed 2023-01-03 by MabezDev

dimpolo, 2023-01-03:
Ah thank you, i missed that.
Maybe you'd like to update https://github.com/esp-rs/espflash/blob/main/espflash/README.md#cargo-runner