esp-rs/espflash #331: SPI flash autodetection fails on Adafruit ESP32 Feather V2

jeremiahrex, 2023-01-07:
I have an [Adafruit ESP32 Feather V2](https://www.adafruit.com/product/5400) and I'm trying to connect using espflash V1.7.0. I've verified that I can connect using both the Arduino IDE and esptool.py V4.4. Here is the output of espflash using log level trace:

```
espflash --log-level trace board-info
New version of espflash is available: v2.0.0-rc.2

Serial port: COM5
Connecting...

2023-01-07T22:00:01.084018Z DEBUG espflash::flasher: Attempting flash enable with: SpiAttachParams { clk: 0, q: 0, d: 0, hd: 0, cs: 0 }
2023-01-07T22:00:01.141322Z DEBUG espflash::flasher: Flash detect failed
2023-01-07T22:00:01.141419Z DEBUG espflash::flasher: Attempting flash enable with: SpiAttachParams { clk: 6, q: 17, d: 8, hd: 11, cs: 16 }
2023-01-07T22:00:01.189323Z DEBUG espflash::flasher: Flash detect failed
2023-01-07T22:00:01.189387Z DEBUG espflash::flasher: SPI flash autodetection failed
Error: espflash::flash_connect

  × Failed to connect to on-device flash
```

Connecting to the board using putty and resetting to see the boot up output I see:

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
```

Comparing the SPI connection parameters, they look the same. I've also tried `espflash --flash-size 8MB --flash-mode DIO --log-level trace board-info` but it produces the same error.

MabezDev, 2023-01-12:
Do you happen to know what flash chip is being used on this board? Perhaps passing the `--use-stub` command _might_ help.

jeremiahrex, 2023-01-13:
I do not know how to determine the flash chip being used. If you can provide some direction on how to figure it out I'd be happy to try.

Two updates... Using `--use-stub` fixes the problem on V1.7.0:
```
espflash board-info
New version of espflash is available: v2.0.0-rc.3

Serial port: COM5
Connecting...

Error: espflash::flash_connect

  × Failed to connect to on-device flash

espflash board-info --use-stub
New version of espflash is available: v2.0.0-rc.3

Serial port: COM5
Connecting...

Using flash stub
Chip type:         ESP32 (revision 3)
Crystal frequency: 40MHz
Flash size:        8MB
Features:          WiFi, BT, Dual Core, 240MHz, Embedded Flash, Embedded PSRAM, Coding Scheme None
MAC address:       e8:9f:6d:25:4e:d4
```
I was also able to try with V2.0.0.rc2 and it works as well:
```
cargo run --bin espflash -- -V
    Finished dev [unoptimized + debuginfo] target(s) in 0.24s
     Running `target\debug\espflash.exe -V`
espflash 2.0.0-rc.2
cargo run --bin espflash -- board-info
    Finished dev [unoptimized + debuginfo] target(s) in 0.24s
     Running `target\debug\espflash.exe board-info`
[2023-01-13T03:23:27Z INFO ] 🚀 A new version of espflash is available: v2.0.0-rc.3
[2023-01-13T03:23:27Z INFO ] Serial port: 'COM5'
[2023-01-13T03:23:27Z INFO ] Connecting...
[2023-01-13T03:23:28Z INFO ] Using flash stub
Chip type:         esp32 (revision v3.0)
Crystal frequency: 40MHz
Flash size:        8MB
Features:          WiFi, BT, Dual Core, 240MHz, Embedded Flash, Embedded PSRAM, Coding Scheme None
MAC address:       e8:9f:6d:25:4e:d4
```
I see that 2.0.0-rc.2 is using the flash stub automatically, which is probably the reason it's successful?

MabezDev, 2023-01-13:
I'm not particularly familiar with the differences between the ROM bootloader and using the stub, but I know that the stub is generally faster and I guess in this instance has more compatibility with FLASH chips.

As you've got it working I'm going to close this :).

closed 2023-01-13 by MabezDev