esp-rs/espflash #347: Integrate esp-serial-dbg into espflash

MabezDev, 2023-01-17:
Having https://github.com/bjoernQ/esp-serial-dbg inside espflash and being able to pass `--monitor --gdb` for example would be really cool! Something we could look at after the 2.0 release of espflash.

jessebraham, 2023-01-19:
Very interesting idea! Definitely want to hold off until `2.0.0` is released but after that I'll look into it!

farbodpm, 2023-04-19:
Great idea !  How can I start it ? I am asking because I think it is a big change and may be I can not do it by myself.

SergioGasquez, 2023-05-05:
> Great idea ! How can I start it ? I am asking because I think it is a big change and may be I can not do it by myself.

I think we should look into this after we finally release 2.0. Basically, I think we should use `esp-serial-dbg` as library and integrate it into our commands, see https://github.com/esp-rs/esp-serial-dbg#basic-usage

SergioGasquez, 2023-11-22:
Is there still value on adding this @esp-rs/espressif ? Specially, if we end up having support in probe-rs for all our targets

bjoernQ, 2023-11-27:
I'd say not anymore - esp-serial-dbg would need some love and given the progress with probe-rs I even consider archiving esp-serial-dbg

SergioGasquez, 2023-11-27:
I will be closing the issue then, if we end up changing our mind we can reopen it.

closed 2023-11-27 by SergioGasquez