esp-rs/espflash #348: Sign Windows executable release artefacts

SergioGasquez, 2023-01-31:
As @georgik requested, we could sign Windows executables [the same way we do it for `espup`](https://github.com/espressif/idf-env/blob/main/.github/workflows/sign-espup.yaml). 

It would be really cool if we had a signing mechanism in `esp-rs`, as doing it in `idf-env` repo is not the best solution (its a completely unrelated repo and it requires some manual steps)

jessebraham, 2023-01-31:
I think that essentially everybody installs via `cargo install` anyway, I've never heard anybody mention this issue before, so not sure what impact it will have in reality. But I suppose as long as it doesn't complicate the release process too much we can do it.

jessebraham, 2023-11-17:
I'm just going to go ahead and say this isn't worth pursuing at this time. We can revisit it in the future if it becomes a necessity.

closed 2023-11-17 by jessebraham