esp-rs/espflash #349: Implement different reset strategies based on host OS and communication method

jessebraham, 2023-02-01:
This is a newer feature in `esptool`, see: https://github.com/espressif/esptool/blob/cc062082fb8ba0ca40cdd32e14fc673562b7b38f/esptool/reset.py

jessebraham, 2023-02-03:
I had implemented this, but it doesn't actually seem to solve any problems. Given that additional code and bump in MSRV required, I'm not sure this is even worth implementing at this point.

AVee, 2023-04-15:
This reset strategy seems to be added to EspTool to fix https://github.com/espressif/esptool/issues/712. It looks like it all about the timing of setting the RTS and DTR pins which appears to differ depending on the combination of OS, USB ports and USB-Serial converter chip. I'm getting the impression it is (mostly?) and issue with usb 3.x ports and/or usb-c serial converters, so it might well become more common over time.

I think I can come up with a fix which at least doesn't need the MSRV bump, and might end up being a bit more compact. I'll create a pull request once I've tested and cleaned it.

AVee, 2023-04-16:
I created a pull request: https://github.com/esp-rs/espflash/pull/387

SergioGasquez, 2023-10-06:
Can anyone affected by this issue check if #487 fixes the issue?
```
cargo install --git https://github.com/SergioGasquez/espflash --branch fix/resets cargo-espflash espflash
```

closed 2023-11-16 by jessebraham