esp-rs/espflash #355: Monitor hangs on esp32-s2/Windows

ardave, 2023-02-16:
Hangs at 

**C:\git\s2-hello-world>espflash flash C:\git\s2-hello-world\target\xtensa-esp32s2-espidf\debug\s2-hello-world --monitor 
✔ Use serial port 'COM4' - USB Serial Device (COM4)? · yes
✔ Remember this serial port for future use? · no
[2023-02-16T00:26:08Z INFO ] Serial port: 'COM4'
[2023-02-16T00:26:08Z INFO ] Connecting...
Chip type:         esp32s2 (revision v0.0)
Crystal frequency: 40MHz
Flash size:        4MB
Features:          WiFi, Embedded Flash 4MB, Embedded PSRAM 2MB, ADC and temperature sensor calibration in BLK2 of efuse V1
MAC address:       84:f7:03:d8:c7:58
App/part. size:    446,928/4,128,768 bytes, 10.82%
[00:00:00] [========================================]       7/7       0x1000                                                                                                                           [00:00:00] [========================================]       1/1       0x8000                                                                                                                           [00:00:05] [========================================]     122/122     0x10000                                                                                                                          [2023-02-16T00:26:18Z INFO ] Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit**

with no output from running program, and also CTRL+R has no effect.

Repro steps:

1. Generate code project from [Hello World ESP-IDF template](https://github.com/esp-rs/esp-idf-template), selecting the ESP32-S2 build target.
2. Cargo Build
3. espflash flash [path to build output] --monitor

Board is: https://www.adafruit.com/product/5303

FWIW the same steps work fine when using this https://www.adafruit.com/product/5395 and selecting the vanilla ESP32 build target for the Hello World project template, which gives me some modicum of assurance in my PC, wiring, etc.

Let me know if I'm missing anything or can do more troubleshooting.  I'm quite interested in learning this space, including learning this flasher implementation in depth.

jessebraham, 2023-03-01:
Looking at [the schematic](https://cdn-learn.adafruit.com/assets/assets/000/110/822/original/adafruit_products_FESP3_sch.png?1649969039) for the ESP32-S2 Feather it appears they're using the `USB_SERIAL_JTAG` peripheral for USB, rather than a USB-to-UART bridge (CH9102F) like on the Qt Py. The `UART` is not connected to the USB connector on the Feather, so you won't see any output unless you print with the `USB_SERIAL_JTAG` peripheral instead.

ardave, 2023-03-02:
I will enjoy now researching the difference between the two.  Apologies for prompting you to read the schematic for my own device.  

May I ask how you can tell that the schematic indicates that a USB_SERIAL_JTAG peripheral is used?  Given that I don't spot a part # on the USB-to-Serial convertor section of the schematic, perhaps you are deducing it from the pinout shown ...

Regardless, thank you for looking into this.

closed 2023-03-02 by ardave

jessebraham, 2023-03-02:
Sorry I was a bit mixed up yesterday, the S2 doesn't have the `USB_SERIAL_JTAG` peripheral (the C3, S3, and C6 do) but it uses `USB-CDC` instead I think. I'm not super familiar with the S2, it's a bit of an odd chip in a lot of ways.

Regardless, this is not a traditional `UART`.

> May I ask how you can tell that the schematic indicates that a USB_SERIAL_JTAG peripheral is used? 

As I've mentioned I misspoke a bit about this, but regardless when using `USB-CDC` or the `USB_SERIAL_JTAG` peripheral, in the schematic the USB port's data lines are connected directly to the `D+`/`D-` pins (`IO20`/`IO19`) and do not go through a USB-to-UART bridge IC, which would connect to some UART pins instead.