esp-rs/espflash #361: `cargo-espflash` fails to flash when the build is too long

jnross, 2023-03-03:
This appears to be another issue related to the watchdog behavior.  `cargo-espflash` connects to the chip before performing the build, and holds the connection.  If the duration of the build is longer than the watchdog timer, the chip resets.  When `cargo-espflash` finishes the build, it attempts to use the now-broken connection to flash, which fails.   

I've prepared two different approaches to fix this, both of which resolve the issue in my testing:
1. Don't connect to the chip until after the build is complete.  [changes](https://github.com/esp-rs/espflash/compare/main...junelife:espflash:joe/dont-connect-before-build) 
2. Reconnect to the chip after the build is complete. [changes](https://github.com/esp-rs/espflash/compare/main...junelife:espflash:joe/reconnect-after-build)

Solution 2, reconnecting results in a tidier patch IMO.  So I've put up a PR with that second solution: 

Note that both solutions rely on reverting 1c054e84127c08a3ccfa60ada39a6185ed1c9ee0

jnross, 2023-03-03:
I can reproduce this easily with a project generated from `esp-idf-template`:

First build fails because the build duration exceeds the watchdog timer duration:
```
➜  esp-idf-template-hello-world git:(master) ✗ cargo +esp espflash flash --target xtensa-esp32s3-espidf "-Zbuild-std=std,panic_abort" --package esp-idf-template-hello-world --port /dev/cu.usbmodem3212201
[2023-03-03T19:04:54Z INFO ] Serial port: '/dev/cu.usbmodem3212201'
[2023-03-03T19:04:54Z INFO ] Connecting...
[2023-03-03T19:04:54Z INFO ] Using flash stub
    Updating crates.io index
   Compiling bindgen v0.60.1
   Compiling embuild v0.30.5
   Compiling esp-idf-sys v0.31.12
   Compiling esp-idf-template-hello-world v0.1.0 (/Users/joe/src/esp-idf-template-hello-world)
    Finished dev [optimized + debuginfo] target(s) in 24.39s
Error: espflash::serial_error

  × Error while connecting to device
  ├─▶ IO error while using serial port: No such device or address
  ╰─▶ No such device or address
```
Second build succeeds since the build is faster
```
➜  esp-idf-template-hello-world git:(master) ✗ cargo +esp espflash flash --target xtensa-esp32s3-espidf "-Zbuild-std=std,panic_abort" --package esp-idf-template-hello-world --port /dev/cu.usbmodem3212201
[2023-03-03T19:05:24Z INFO ] Serial port: '/dev/cu.usbmodem3212201'
[2023-03-03T19:05:24Z INFO ] Connecting...
[2023-03-03T19:05:24Z INFO ] Using flash stub
    Updating crates.io index
    Finished dev [optimized + debuginfo] target(s) in 0.25s
Chip type:         esp32s3 (revision v0.1)
Crystal frequency: 40MHz
Flash size:        8MB
Features:          WiFi, BLE
MAC address:       7c:df:a1:e2:4f:f8
Bootloader:        /Users/joe/src/esp-idf-template-hello-world/target/xtensa-esp32s3-espidf/debug/build/esp-idf-sys-d219997a19e22071/out/build/bootloader/bootloader.bin
Partition table:   /Users/joe/src/esp-idf-template-hello-world/target/xtensa-esp32s3-espidf/debug/build/esp-idf-sys-d219997a19e22071/out/build/partition_table/partition-table.bin
App/part. size:    469,600/1,048,576 bytes, 44.78%
[00:00:00] [========================================]      13/13      0x0                                             [00:00:00] [========================================]       1/1       0x8000                                          [00:00:05] [========================================]     238/238     0x10000                                         [2023-03-03T19:05:31Z INFO ] Flashing has completed!
```

jnross, 2023-03-03:
Looks like there might be other approaches here.  Like moving the call to:

```
target.begin(&mut self.connection).flashing()?; //disables the watchdog
```
before the build step.

jnross, 2023-03-05:
I've closed #362 because it has a few major drawbacks.  I'll noodle on this some more.

jessebraham, 2023-03-15:
Sorry for taking so long to get to this, been busy with other projects.

I believe this issue should be resolved by #359 (which has recently been merged), would you be able to verify whether the current `main` works for you?

jnross, 2023-03-15:
Hi @jessebraham:  Thanks for your response.  I totally understand how other priorities can limit response times for tickets like this.

I have been following along with the progress of #359, and I tested at commit a69116d3ff68abc4da60c5e4150e8f3e69741315.  I'll test again on `main` now that this has been merged and confirm, but I don't expect this behavior to be resolved.  What I see is that with my hardware (ESP32-S3), the flash stub is used but that the flash stub doesn't appear to disable the watchdog.  The code path [here](https://github.com/esp-rs/espflash/pull/359/commits/a69116d3ff68abc4da60c5e4150e8f3e69741315#diff-a4a619d1e5465e102ec644636f35777549000c08b9d1bdb3c11d254342637f0cR72) that disables the watchdog is not activated in my use.

I haven't dug deeply so the flash stubs are still a bit of a mystery to me - how do they work?  Where do these binaries come from?

In the meantime, I continue to use a local build that includes the changes in #363, and everything is working great for me.

PR #363 (2023-03-24): Disable watchdog timer before build.
Fixes #361
closed 2023-03-24 by jessebraham