esp-rs/espflash #365: espflash fails to enable terminal cursor when exited with ctrl-c

faern, 2023-03-08:
If I terminate `espflash` while in the [port selection state](https://github.com/esp-rs/espflash/blob/90056f4691c7188194c3197e47b247c49aa72a54/espflash/src/cli/serial.rs#L179) my terminal cursor remains disabled. I guess this is because `espflash` uses terminal raw mode to disable the cursor, and then don't catch and handle the `SIGINT` signal and enables the cursor before exiting?

closed 2023-03-24 by jessebraham

jessebraham, 2023-03-29:
Sorry I just realized I closed this issue without saying anything 😅 I've added a fix for this issue which will be available in `2.0.0-rc.4`, which we're hoping to publish in the next week or two.

faern, 2023-03-29:
Awesome! Thanks 😊 looking forward

PR #372 (2023-03-24): Restore the cursor when exiting from serial port selection via Ctrl-C
Closes #365

Additionally bumps the MSRV due to new dependencies requiring such.