esp-rs/espflash #367: option to auto-choose serial port if only one matches a known USB device

jneem, 2023-03-19:
Because I often have a few USB-serial devices connected, my esp device gets a path like `/dev/ttyACMn` but the `n` is unpredictable. This prevents me from using the `serial` option in my `espflash.toml` file, and the presence of multiple serial ports also prevents `cargo espflash` from auth-connecting to one. But I usually connect only a single esp device, and so `cargo espflash` always highlights a single "known common dev board", and it's always the one I want.

Would you accept a PR to add an option (either a CLI flag or something in the config file) to say that if there's a single known dev board then auto-connect to it?

jessebraham, 2023-03-24:
Sorry for taking so long to get to this, thanks for opening this issue.

I would be happy to accept a PR for this functionality, I think it makes sense from a UX perspective. I'm not entirely sure how to go about doing this, maybe if there's only a single recommended device we can just select it regardless, no setting/flag required. I'm not really sure if there are any downsides to this, I'll have to think about it a bit I suppose. If you have any further input on this I'd be glad to hear it!

jneem, 2023-03-25:
Yeah, I don't have a real motivation for making it a flag; I mainly just suggested it to be conservative. I'll put up a PR that enables the auto-selection with no flag, but if you decide you'd rather go with one I'd be happy to change it.

I think the only downside of having a flag is increased complexity. As far as UX goes, I was just planning to add `runner = "espflash --monitor --enable-auto-whatever"` to my `.cargo/config.toml`, so I'll never have to type out that flag anyway :shrug:

PR #374 (2023-04-04): If exactly one port matches, use it
Fixes #367.

Since I added the `matches` closure, it felt like some clean-up was appropriate. But if you'd prefer a more minimal PR I can do that too.

This also slightly changes the behavior of non-usb serial ports, for consistency. Previously, you could make them match by manually editing the config file and adding "vid=0, pid=0", but they wouldn't be sorted and highlighted as though they matched. After this change, non-usb serial ports won't auto-match at all.
closed 2023-04-04 by SergioGasquez