esp-rs/espflash #385: espflash monitor won't work properly

BrPrS, 2023-04-05:
Hi 
I am trying to run [esp-idf-template](https://github.com/esp-rs/esp-idf-template) 
after flashing the esp using `espflash /dev/ttyUSB0 target/[xtensa-esp32-espidf|xtensa-esp32s2-espidf|xtensa-esp32s3-espidf|riscv32imc-esp-espidf]/debug/<My-project-name>`
when I am trying to monitor , using `espflash monitor` I am getting this error:
```
Serial port: /dev/ttyUSB0
Connecting...

Chip type:         ESP32 (revision 3)
Crystal frequency: 40MHz
Flash size:        4MB
Features:          WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC address:       b8:d6:1a:b2:e4:d8
Error: 
  × No such file or directory (os error 2)
```

and if I want to specify the exact USB Address using `espflash monitor /dev/ttyUSB0` , I getting this error:
```
Error: espflash::connection_failed

  × Error while connecting to device
  ╰─▶ Serial port not found
  help: Ensure that the device is connected and your host recognizes the serial adapter

```

but when I am using `idf.py`  to monitor the esp, it works totally fine.

chrisboyce, 2023-04-24:
I had the same issue but was able to make progress by building locally from the latest version.
```
git clone https://github.com/esp-rs/espflash.git
cargo install --path espflash/espflash
```

SergioGasquez, 2023-05-05:
Hi @BrPrS, in which version did the error occurred? Have you tried with `main` branch? @chrisboyce just fyi, you can also install the latest version with `cargo install espflash --git https://github.com/esp-rs/espflash`

F-bh, 2023-05-07:
It worked for me after upgrading to v2.0.0-rc.3

gustavobat, 2023-05-23:
I had the same problem today and also solved by installing `v2.0.0-rc.3`. I suggest updating `latest` ASAP since it's broken.

jessebraham, 2023-06-08:
I've published `2.0.0-rc.4`, which should hopefully resolve this issue. If you continue to see problems please feel free to re-open this issue, or to open a new issue.

closed 2023-06-08 by jessebraham