esp-rs/espflash #393: Flashing with USB-SERIAL-JTAG flashes an unbootable image

MabezDev, 2023-04-27:
After installing RC4 I noticed that flashing with the USB-SERIAL-JTAG port (in my case, S3) no longer correctly flashes the chip. The second stage bootloader complains about the partition table having partition size. I bisected from a known working state to find that #375 seems to be the culprit. 

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0x18 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403cdb4a
0x403cdb4a - __default_naked_level_7_interrupt
    at ??:??
SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:2
load:0x3fce3810,len:0x1690
0x3fce3810 - _stack_end_cpu0
    at ??:??
load:0x403c9700,len:0xbd0
0x403c9700 - __default_naked_level_7_interrupt
    at ??:??
load:0x403cc700,len:0x2dc8
0x403cc700 - __default_naked_level_7_interrupt
    at ??:??
SHA-256 comparison failed:
Calculated: ae126cb96d962b79703c4a4144654f513bc92dbad93c921b2e20c5558d0914dc
Expected: d1c78d7325094d95e7bb525e04f54c3212a0971c36efb48e3f3f056daece1a5e
Attempting to boot anyway...
entry 0x403c9900
0x403c9900 - __default_naked_level_7_interrupt
    at ??:??
I (60) boot: ESP-IDF v5.0-beta1-764-gdbcf640261 2nd stage bootloader
I (60) boot: compile time 11:32:39
I (60) boot: chip revision: V000
I (64) boot.esp32s3: Boot SPI Speed : 40MHz
I (69) boot.esp32s3: SPI Mode       : SLOW READ
I (74) boot.esp32s3: SPI Flash Size : 32MB
I (79) boot: Enabling RNG early entropy source...
I (84) boot: Partition Table:
I (88) boot: ## Label            Usage          Type ST Offset   Length
I (95) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (102) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (110) boot:  2 factory          factory app      00 00 00010000 01ff0000
I (117) boot: End of partition table
E (122) esp_image: partition size 0x1ff0000 invalid, larger than 16MB
E (129) boot: Factory app partition is not bootable
E (134) boot: No bootable app partitions in the partition table
```

Looking at the changes in #375, I'm not sure what would cause this. @maxwase, could you test flashing on your S3 board via USB serial jtag to verify this is a real issue?

MabezDev, 2023-04-27:
Edit, I actually found this happens regardless of flashing via USB-SERIAL-JTAG or the USB serial converter.

MabezDev, 2023-04-27:
I found an S3 board which _doesn't_ have Octal flash on it, and it works just fine... Hmmm.

maxwase, 2023-04-29:
I tied to reproduce this, but got... the opposite results
I'm unable to flash S3 before my bootloader changes, so I tried to flash hello_world example from `esp-idf` repository. Surprisingly, my esp went into a bootloop with
`E (277) spi_flash: Detected size(512k) smaller than the size in the binary image header(32768k). Probe failed.` message. 
Out of curiosity  I selected octal flash mode using `idf.py menuconfig` -> `Serial flasher config` -> `Enable Octal Flash` and flashed the helloworld image once again and it booted without any problem (note that I've flashed bootloader before)

Next, I tried to flash rusty blinky image and got the same bootloop error, so I tried to specify octal flash settings in `sdkconfig.defaults`:
```
CONFIG_ESPTOOLPY_OCT_FLASH=y
CONFIG_ESPTOOLPY_FLASH_SAMPLE_MODE_STR=y
```
Magically, it booted successfully, although I did receive some warnings that I always had.

This behavior is especially interesting because I was able to flash both helloworld and rust blinky examples before without octal mode. If I remember correctly, idf and toolchain versions are the same as before.

maxwase, 2023-05-07:
UPD: Reproduced with explicit partition table that has >16 MB partition. This issue has not been seen before due to wrong size-freq config in the header

PR #398 (2023-05-29): limit default partition size
Some chips can have more than 16MB of flash, mine has 32MB. Historically espflash failed to detect this flash size (and defaulted to 4MB) but now it can. Without this change, the board will boot loop because the esp idf bootloader max size is only 16MB.

Closes #393
closed 2023-05-29 by jessebraham