esp-rs/espflash #394: espflash unable to connect

yshui, 2023-04-28:
I have a esp32-devkitc board from AZ-delivery. The chip is ESP32-WROOM-32. 

`espflash` is unable to connect to this board. `idf.py` from `esp-idf` works properly, I was able to upload the hello_world program with it.

yshui, 2023-04-28:
esptool log:

```
esptool esp32 -p /dev/ttyUSB0 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 2MB 0x1000 bootloader/bootloader.bin 0x10000 hello_world.bin 0x8000 partition_table/partition-table.bin
esptool.py v4.5.1
Serial port /dev/ttyUSB0
Connecting.....
Chip is ESP32-D0WD (revision v1.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 94:3c:c6:37:eb:f8
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00007fff...
Flash will be erased from 0x00010000 to 0x0003afff...
Flash will be erased from 0x00008000 to 0x00008fff...
Compressed 26720 bytes to 16752...
Writing at 0x00001000... (50 %)
Writing at 0x00007689... (100 %)
Wrote 26720 bytes (16752 compressed) at 0x00001000 in 0.8 seconds (effective 282.0 kbit/s)...
Hash of data verified.
Compressed 174624 bytes to 97266...
Writing at 0x00010000... (16 %)
Writing at 0x0001bfd8... (33 %)
Writing at 0x000219d7... (50 %)
Writing at 0x000270d8... (66 %)
Writing at 0x0002d393... (83 %)
Writing at 0x000350f9... (100 %)
Wrote 174624 bytes (97266 compressed) at 0x00010000 in 2.5 seconds (effective 547.9 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 103...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (103 compressed) at 0x00008000 in 0.1 seconds (effective 396.3 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
Done
```

espflash log:

```
$ espflash board-info /dev/ttyUSB0
New version of espflash is available: v2.0.0-rc.3

Serial port: /dev/ttyUSB0
Connecting...

Unable to connect, retrying with extra delay...
Unable to connect, retrying with default delay...
Unable to connect, retrying with extra delay...
Unable to connect, retrying with default delay...
Unable to connect, retrying with extra delay...
```

yshui, 2023-04-28:
can confirm this is fixed by #387

ede1998, 2023-05-07:
Also had this issue and could not connect to my ESP32-WROOM-32. #387 fixes that for me as well.
```
> espflash board-info
[2023-05-07T13:30:50Z INFO ] Serial port: '/dev/ttyUSB0'
[2023-05-07T13:30:50Z INFO ] Connecting...
[2023-05-07T13:30:50Z INFO ] Attempting UnixTight reset with default delay...
[2023-05-07T13:30:51Z INFO ] Using flash stub
Chip type:         esp32 (revision v3.1)
Crystal frequency: 40MHz
Flash size:        4MB
Features:          WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
```

Output of esptool.py:
```
> esptool.py chip_id
esptool.py v4.5.1
Found 2 serial ports
Serial port /dev/ttyUSB0
Connecting......
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting.....
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision v3.1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: a0:b7:65:53:ea:68
Uploading stub...
Running stub...
Stub running...
Warning: ESP32 has no Chip ID. Reading MAC instead.
MAC: a0:b7:65:53:ea:68
Hard resetting via RTS pin...
```

jessebraham, 2023-05-31:
@yshui @ede1998 sorry to be a bother, but would you two also mind testing out this branch as well please:  
https://github.com/jessebraham/espflash/tree/fixes/reset-to-flash

This is an updated of a previous PR I had opened, but closed for various reasons. I prefer this implementation, but need to be confident it actually works. This is copied almost verbatim from `esptool.py` though, so in theory it should be correct.

ede1998, 2023-05-31:
Bad news:
```
> RUST_LOG=debug target/release/espflash board-info
[2023-05-31T16:22:24Z DEBUG] BoardInfo(
ConnectArgs {
baud: None,
port: None,
no_stub: false,
},
)
[2023-05-31T16:22:24Z DEBUG] connecting to crates.io:443 at [2600:9000:2251:ac00:c:7ed3:240:93a1]:443
[2023-05-31T16:22:24Z DEBUG] No cached session for DnsName(DnsName(DnsName("crates.io")))
[2023-05-31T16:22:24Z DEBUG] Not resuming any session
[2023-05-31T16:22:24Z DEBUG] Using ciphersuite TLS13_AES_128_GCM_SHA256
[2023-05-31T16:22:24Z DEBUG] Not resuming
[2023-05-31T16:22:24Z DEBUG] TLS1.3 encrypted extensions: [ServerNameAck]
[2023-05-31T16:22:24Z DEBUG] ALPN protocol is None
[2023-05-31T16:22:24Z DEBUG] created stream: Stream(RustlsStream)
[2023-05-31T16:22:24Z DEBUG] sending request GET https://crates.io/api/v1/crates/espflash/versions
[2023-05-31T16:22:24Z DEBUG] writing prelude: GET /api/v1/crates/espflash/versions HTTP/1.1
Host: crates.io
User-Agent: ureq/2.6.2
Accept: */*
accept-encoding: gzip
[2023-05-31T16:22:24Z DEBUG] Ticket saved
[2023-05-31T16:22:24Z DEBUG] Chunked body in response
[2023-05-31T16:22:24Z DEBUG] response 200 to GET https://crates.io/api/v1/crates/espflash/versions
[2023-05-31T16:22:24Z DEBUG] dropping stream: Stream(RustlsStream)
[2023-05-31T16:22:24Z INFO ] Serial port: '/dev/ttyUSB0'
[2023-05-31T16:22:24Z INFO ] Connecting...
[2023-05-31T16:22:24Z DEBUG] Using UnixTight reset strategy with delay of 50ms
[2023-05-31T16:22:24Z DEBUG] Using UnixTight reset strategy with delay of 550ms
[2023-05-31T16:22:24Z DEBUG] Using Classic reset strategy with delay of 50ms
[2023-05-31T16:22:25Z DEBUG] Using Classic reset strategy with delay of 550ms
[2023-05-31T16:22:26Z DEBUG] Using UnixTight reset strategy with delay of 50ms
[2023-05-31T16:22:26Z DEBUG] Using UnixTight reset strategy with delay of 550ms
[2023-05-31T16:22:26Z DEBUG] Using Classic reset strategy with delay of 50ms
Error: espflash::connection_failed

× Error while connecting to device
╰─▶ Failed to connect to the device
help: Ensure that the device is connected and the reset and boot pins are not being held down

> git show --shortstat
commit 42e079d77ea8ee2243e88420dc889da2c4bf5b65 (HEAD -> reset-to-flash, jessebraham/fixes/reset-to-flash)
Author: Jesse Braham <jesse@beta7.io>
Date:   Wed May 31 08:38:29 2023 -0700

Bump the MSRV to 1.66

6 files changed, 8 insertions(+), 8 deletions(-)
```

ede1998, 2023-06-07:
Still no luck:
```
> cargo build --release
Compiling espflash v2.0.0-rc.4 (/home/erik/dev/rust/esp32/espflash/espflash)
Compiling cargo-espflash v2.0.0-rc.4 (/home/erik/dev/rust/esp32/espflash/cargo-espflash)
Finished release [optimized] target(s) in 23.92s

> RUST_LOG=debug target/release/espflash board-info
[2023-06-07T11:19:31Z DEBUG] BoardInfo(
ConnectArgs {
baud: None,
port: None,
no_stub: false,
},
)
[2023-06-07T11:19:31Z DEBUG] connecting to crates.io:443 at [2600:9000:2251:0:c:7ed3:240:93a1]:443
[2023-06-07T11:19:31Z DEBUG] No cached session for DnsName(DnsName(DnsName("crates.io")))
[2023-06-07T11:19:31Z DEBUG] Not resuming any session
[2023-06-07T11:19:32Z DEBUG] Using ciphersuite TLS13_AES_128_GCM_SHA256
[2023-06-07T11:19:32Z DEBUG] Not resuming
[2023-06-07T11:19:32Z DEBUG] TLS1.3 encrypted extensions: [ServerNameAck]
[2023-06-07T11:19:32Z DEBUG] ALPN protocol is None
[2023-06-07T11:19:32Z DEBUG] created stream: Stream(RustlsStream)
[2023-06-07T11:19:32Z DEBUG] sending request GET https://crates.io/api/v1/crates/espflash/versions
[2023-06-07T11:19:32Z DEBUG] writing prelude: GET /api/v1/crates/espflash/versions HTTP/1.1
Host: crates.io
User-Agent: ureq/2.6.2
Accept: */*
accept-encoding: gzip
[2023-06-07T11:19:32Z DEBUG] Ticket saved
[2023-06-07T11:19:32Z DEBUG] Chunked body in response
[2023-06-07T11:19:32Z DEBUG] response 200 to GET https://crates.io/api/v1/crates/espflash/versions
[2023-06-07T11:19:32Z DEBUG] dropping stream: Stream(RustlsStream)
[2023-06-07T11:19:32Z INFO ] Serial port: '/dev/ttyUSB0'
[2023-06-07T11:19:32Z INFO ] Connecting...
[2023-06-07T11:19:32Z DEBUG] Using UnixTight reset strategy with delay of 50ms
[2023-06-07T11:19:32Z DEBUG] Using UnixTight reset strategy with delay of 500ms
[2023-06-07T11:19:32Z DEBUG] Using Classic reset strategy with delay of 50ms
[2023-06-07T11:19:33Z DEBUG] Using Classic reset strategy with delay of 500ms
[2023-06-07T11:19:34Z DEBUG] Using UnixTight reset strategy with delay of 50ms
[2023-06-07T11:19:34Z DEBUG] Using UnixTight reset strategy with delay of 500ms
[2023-06-07T11:19:34Z DEBUG] Using Classic reset strategy with delay of 50ms
Error: espflash::connection_failed

× Error while connecting to device
╰─▶ Failed to connect to the device
help: Ensure that the device is connected and the reset and boot pins are not being held down

> git show --stat
commit dac6a90fd4473c0b7a505df8735a11a1e52da7ae (HEAD -> fix/reset-strategy, sergio/fix/reset-strategy)
Author: Sergio Gasquez <sergio.gasquez@gmail.com>
Date:   Wed Jun 7 11:20:26 2023 +0200

feat: ✨ Update classic reset function

espflash/src/connection/reset.rs | 7 ++++++-
1 file changed, 6 insertions(+), 1 deletion(-)
```

manio, 2023-07-22:
Hi,
I own `WROOM-32U`. I wanted to play with esp-rs on this. With latest espflash I've got:
```
❯ cargo espflash board-info
[2023-07-22T10:40:34Z INFO ] Serial port: '/dev/ttyUSB0'
[2023-07-22T10:40:34Z INFO ] Connecting...
[2023-07-22T10:40:35Z INFO ] Unable to connect, retrying with extra delay...
[2023-07-22T10:40:36Z INFO ] Unable to connect, retrying with default delay...
[2023-07-22T10:40:37Z INFO ] Unable to connect, retrying with extra delay...
[2023-07-22T10:40:38Z INFO ] Unable to connect, retrying with default delay...
[2023-07-22T10:40:39Z INFO ] Unable to connect, retrying with extra delay...
[2023-07-22T10:40:41Z INFO ] Unable to connect, retrying with default delay...
[2023-07-22T10:40:42Z INFO ] Unable to connect, retrying with extra delay...
Error: espflash::connection_failed

  × Error while connecting to device
  ╰─▶ Failed to connect to the device
  help: Ensure that the device is connected and the reset and boot pins are not being held down
```
While with https://github.com/esp-rs/espflash/pull/387 it's working fine:
```
❯ cargo espflash board-info      
[2023-07-22T10:39:44Z INFO ] 🚀 A new version of cargo-espflash is available: v2.0.1
[2023-07-22T10:39:44Z INFO ] Serial port: '/dev/ttyUSB0'
[2023-07-22T10:39:44Z INFO ] Connecting...
[2023-07-22T10:39:44Z INFO ] Attempting UnixTight reset with default delay...
[2023-07-22T10:39:44Z INFO ] Using flash stub
Chip type:         esp32 (revision v3.0)
Crystal frequency: 40MHz
Flash size:        4MB
Features:          WiFi, BT, Dual Core, 240MHz, Coding Scheme None
MAC address:       xx:xx:xx:xx:xx:xx
```

nweaver, 2023-07-27:
I'm having this problem as well:  
`nweaver@espbuild:~/hello $ cargo espflash board-info
[2023-07-27T15:01:05Z INFO ] Detected 2 serial ports
[2023-07-27T15:01:05Z INFO ] Ports which match a known common dev board are highlighted
[2023-07-27T15:01:05Z INFO ] Please select a port
[2023-07-27T15:01:07Z INFO ] Serial port: '/dev/ttyAMA0'
[2023-07-27T15:01:07Z INFO ] Connecting...
[2023-07-27T15:01:09Z INFO ] Unable to connect, retrying with extra delay...
[2023-07-27T15:01:10Z INFO ] Unable to connect, retrying with default delay...
[2023-07-27T15:01:11Z INFO ] Unable to connect, retrying with extra delay...
[2023-07-27T15:01:13Z INFO ] Unable to connect, retrying with default delay...
[2023-07-27T15:01:14Z INFO ] Unable to connect, retrying with extra delay...
[2023-07-27T15:01:16Z INFO ] Unable to connect, retrying with default delay...
^C[2023-07-27T15:01:17Z INFO ] Unable to connect, retrying with extra delay...
`

It works with idf.py, esptool reports version 2.0.1.

Platform is Raspberry Pi (64b Rasbian, latest install), target is ESP32S3 on a SEEED XIAO ESP32S3 board.

SergioGasquez, 2023-10-06:
Can anyone affected by this issue check if #487 fixes the issue?
```
cargo install --git https://github.com/SergioGasquez/espflash --branch fix/resets cargo-espflash espflash
```

manio, 2023-10-07:
@SergioGasquez 
Thanks! (Un)fortunately few days ago I moved my development ESP to target location and started running "in production" so it is a little difficult to test it like before (when I have it on my desk). But if nobody will test it - I will go there with a laptop and try to check it out :)

ede1998, 2023-10-07:
> Can anyone affected by this issue check if #487 fixes the issue?
> 
> ```
> cargo install --git https://github.com/SergioGasquez/espflash --branch fix/resets cargo-espflash espflash
> ```

Just tried it out and can confirm that it works. I tried both a release and debug build with the `board-info` subcommand and I successfully flashed an application using your fixed version.

Thanks for the effort!

```
> git show --stat
commit 322f5a92a9fa32ff328b47c4643ae53cfd732a9e (HEAD -> fix/resets, sergio/fix/resets)
Author: Sergio Gasquez <sergio.gasquez@gmail.com>
Date:   Fri Oct 6 12:01:54 2023 +0200

docs: Update changelog

CHANGELOG.md | 1 +
1 file changed, 1 insertion(+)

> cargo build
Finished dev [unoptimized + debuginfo] target(s) in 0.45s

> RUST_LOG=debug target/debug/espflash board-info
[2023-10-07T17:01:54Z DEBUG] BoardInfo(
ConnectArgs {
baud: None,
port: None,
no_stub: false,
},
)
[2023-10-07T17:01:54Z DEBUG] connecting to crates.io:443 at [2600:9000:225f:400:c:7ed3:240:93a1]:443
[2023-10-07T17:01:54Z DEBUG] No cached session for DnsName("crates.io")
[2023-10-07T17:01:54Z DEBUG] Not resuming any session
[2023-10-07T17:01:54Z DEBUG] Using ciphersuite TLS13_AES_128_GCM_SHA256
[2023-10-07T17:01:54Z DEBUG] Not resuming
[2023-10-07T17:01:54Z DEBUG] TLS1.3 encrypted extensions: [ServerNameAck]
[2023-10-07T17:01:54Z DEBUG] ALPN protocol is None
[2023-10-07T17:01:54Z DEBUG] created stream: Stream(RustlsStream)
[2023-10-07T17:01:54Z DEBUG] sending request GET https://crates.io/api/v1/crates/espflash/versions
[2023-10-07T17:01:54Z DEBUG] writing prelude: GET /api/v1/crates/espflash/versions HTTP/1.1
Host: crates.io
User-Agent: ureq/2.8.0
Accept: */*
accept-encoding: gzip
[2023-10-07T17:01:54Z DEBUG] Chunked body in response
[2023-10-07T17:01:54Z DEBUG] response 200 to GET https://crates.io/api/v1/crates/espflash/versions
[2023-10-07T17:01:54Z DEBUG] dropping stream: Stream(RustlsStream)
[2023-10-07T17:01:54Z INFO ] Serial port: '/dev/ttyUSB0'
[2023-10-07T17:01:54Z INFO ] Connecting...
[2023-10-07T17:01:54Z DEBUG] Using UnixTight reset strategy with delay of 50ms
[2023-10-07T17:01:55Z INFO ] Using flash stub
[2023-10-07T17:01:55Z DEBUG] Loading flash stub for chip: Esp32
[2023-10-07T17:01:55Z DEBUG] Write 3416 byte stub text
[2023-10-07T17:01:55Z DEBUG] Write 156 byte stub data
[2023-10-07T17:01:55Z DEBUG] Finish stub write
[2023-10-07T17:01:55Z DEBUG] Stub written!
[2023-10-07T17:01:55Z DEBUG] Re-detected chip: Esp32
[2023-10-07T17:01:55Z DEBUG] Attempting flash enable with: SpiAttachParams { clk: 0, q: 0, d: 0, hd: 0, cs: 0 }
[2023-10-07T17:01:55Z DEBUG] Flash detect OK!
Chip type:         esp32 (revision v3.1)
Crystal frequency: 40MHz
Flash size:        4MB
Features:          WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC address:       a0:b7:65:53:ea:68
```

SergioGasquez, 2023-10-09:
Thanks, @ede1998 @manio! Sadly, the PR introduces a breaking change in the public API, so we will need to wait until we start working on the 3.0.0 release (which should be soon) to merge it

closed 2023-11-16 by jessebraham