esp-rs/espflash #395: Unclear how to update espflash

phansel, 2023-04-28:
`New version of espflash is available: v2.0.0-rc.3` is printed every time espflash is run, but it's not clear how to update. On both aarch64 and x86_64, `cargo install espflash` simply re-installs v1.7.0. `cargo update` does not change anything.

jessebraham, 2023-04-29:
`cargo` requires you specify the version for pre-releases:
```bash
cargo install espflash@2.0.0-rc.3
```

phansel, 2023-04-30:
Thank you @jessebraham! Feel free to close, though maybe leaving it open would be helpful for other people with the same question.

tingox, 2023-05-15:
This issue was helpful for my understanding too.

jessebraham, 2023-06-08:
There's not really anything actionable here IMO, so I'm going to close this issue. I feel that this is explained by the `cargo` documentation already and I expect users to know how to use these tools :)

closed 2023-06-08 by jessebraham