esp-rs/espflash #404: Can't exit espflash if firmware is stuck

bugadani, 2023-05-11:
The exact cause is unknown at this point, but here is the setup:

 - using windows 11 and windows terminal
 - using an ESP32-S3 with the internal USB/JTAG interface
 - using espflash 2.0.0 with the #399 patch
 - trying to run the code from https://github.com/card-io-rs/card-io-fw/tree/68d58393aa12d0ed56ff9de52428f8fab9de7c12

The serial output includes the bootloader messages and the "Hello, world!" printed from the main task, but does not include the state printing a few lines below. In this state (i.e. probably while stuck in Board::initialize, which is trying to configure a display over SPI, but disregarding the bug in my code), espflash doesn't react to `Ctrl+C` or `Ctrl+R` and does not exit.

Manually resetting the MCU exits espflash and then the terminal prints the control characters:
[image]

bugadani, 2023-05-11:
And just after posting this, the problem goes away by itself, of course...

closed 2023-05-11 by bugadani

cacharle, 2024-08-28:
I have the same issue on Alacritty and the default terminal of RaspberryPiOS. Only way to stop espflash is to kill it in another terminal.

It doesn't even stop on SIGQUIT `Ctrl+\`.

Anyone knows a fix?