esp-rs/espflash #406: `espflash` can't flash using the JTAG port, but `esptool.py` can

flippette, 2023-05-16:
`espflash -V`: `espflash 2.0.0-rc.3`
`cargo espflash -V`: `cargo-espflash-espflash 2.0.0-rc.3`

I've got an ESP32-C6-DevKitC-1 and I've never gotten `espflash` to flash the board through the JTAG port. UART works fine though.

`espflash` with the JTAG port (`cargo espflash` does the same thing):

```
> espflash flash --format direct-boot --monitor target/riscv32imac-unknown-none-elf/release/hello-esp 
[2023-05-16T19:46:17Z INFO ] Detected 2 serial ports
[2023-05-16T19:46:17Z INFO ] Ports which match a known common dev board are highlighted
[2023-05-16T19:46:17Z INFO ] Please select a port
✔ Remember this serial port for future use? · no
[2023-05-16T19:46:18Z INFO ] Serial port: '/dev/ttyACM0'
[2023-05-16T19:46:18Z INFO ] Connecting...
[2023-05-16T19:46:19Z INFO ] Using flash stub
Error: espflash::timeout

  × Error while connecting to device
  ╰─▶ Timeout while running command
```

`espflash` with the UART port (`cargo espflash` does the same thing):

```
> espflash flash --format direct-boot --monitor target/riscv32imac-unknown-none-elf/release/hello-esp
[2023-05-16T19:47:31Z INFO ] Detected 2 serial ports
[2023-05-16T19:47:31Z INFO ] Ports which match a known common dev board are highlighted
[2023-05-16T19:47:31Z INFO ] Please select a port
✔ Remember this serial port for future use? · no
[2023-05-16T19:47:33Z INFO ] Serial port: '/dev/ttyUSB0'
[2023-05-16T19:47:33Z INFO ] Connecting...
[2023-05-16T19:47:33Z INFO ] Using flash stub
Chip type:         esp32c6 (revision v0.0)
Crystal frequency: 40MHz
Flash size:        4MB
Features:          WiFi 6, BT 5
MAC address:       40:4c:ca:40:04:dc
App size:          36,948 bytes
[00:00:01] [========================================]      21/21      0x0                                                                                                    [2023-05-16T19:47:36Z INFO ] Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
```

`esptool.py` with the JTAG port:

```
esptool.py --chip esp32c6 --port /dev/ttyACM0 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size keep 0x10000 target/riscv32imac-unknown-none-elf/release/hello-esp
esptool.py v4.5.1
Serial port /dev/ttyACM0
Connecting...
Chip is ESP32-C6 (revision v0.0)
Features: WiFi 6, BT 5, IEEE802.15.4
Crystal is 40MHz
MAC: 40:4c:ca:40:04:dc
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00010000 to 0x0001afff...
Compressed 43928 bytes to 21524...
Wrote 43928 bytes (21524 compressed) at 0x00010000 in 0.6 seconds (effective 618.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

flippette, 2023-05-16:
Update: It seems to work when I pass `--no-stub`, no idea why.

jessebraham, 2023-05-29:
The initial version of the C6's stub which we added to `espflash` seemed to have some issues, I recall needing to use `--no-stub` as well. Updating the stub resolves this issue, so this should be fixed in `2.0.0-rc.4` when it gets released.

closed 2023-06-08 by jessebraham