esp-rs/espflash #408: Reset button doesn't restart program when flashed with espflash

seanaye, 2023-05-21:
I am flashing an esp32-c6-WROOM-1 devkit using espflash. The program runs fine but when I click the reset button on the board the board freezes but does not restart the program. E.g. a blinky application will freeze the LED in the state at the time of the reset button push. The program does not start up again.

SergioGasquez, 2023-05-22:
Hi! Few questions:
- Can you open the monitor after flashing?
  - If so, have you tried resetting the board with `Ctrl+R`?
- Have you tried reproducing this behavior with [esptool](https://github.com/espressif/esptool) 

Not sure if this is related with espflash if the flashing/monitoring does not throw any errors.

seanaye, 2023-05-29:
Yes I can open the monitor after flashing. Ctrl + R does restart the board. I can flash the board see the serial out put use ctrl r to restart. When I click the reset button the board freezes (blinky will freeze)  and the serial monitor disconnects. I can still re-connect the monitor and use ctrl + R to restart the program but the reset button will never restart the program.

Will try with esptool

seanaye, 2023-05-30:
I should also note that when I click the reset button the board completely freezes in whatever state it is in. e.g. if the LED is on the board will freeze with the LED on and there is no way to get it to go off without unplugging. No combination of BOOT or RESET button pushes will turn off the light

seanaye, 2023-05-31:
I am unsure how to use esptool to flash the board, the documentation for `write_flash` references a `bootloader.bin` but my rust program does not output a file like that. I just have `/target/riscv32imac-esp-espidf/release/CRATE_NAME` after running `cargo build --release --target riscv32imac-esp-espidf.json`

SergioGasquez, 2023-05-31:
> I am unsure how to use esptool to flash the board, the documentation for `write_flash` references a `bootloader.bin` but my rust program does not output a file like that. I just have [...]

See [`elf2image`](https://docs.espressif.com/projects/esptool/en/latest/esp8266/esptool/basic-commands.html#convert-elf-to-binary-elf2image) command and, in case you need the bootloader, you can download it from https://github.com/esp-rs/espflash/tree/main/espflash/resources/bootloaders

seanaye, 2023-05-31:
Will try tonight, thanks

SergioGasquez, 2023-06-08:
> Will try tonight, thanks

Hi! Do you have any updates?

SergioGasquez, 2023-10-06:
Hello @seanaye, do you have any updates?

SergioGasquez, 2023-10-11:
I will be closing the issue as it's inactive, feel free to reopen it if the issue is still present

closed 2023-10-11 by SergioGasquez