esp-rs/espflash #409: espflash thinks that dev/cuau0 and /dev/cuaU0 are the same (usb) serial port

tingox, 2023-05-23:
If I don't specify a port (but select the correct one when asked) espflash board-info works
```
tingo@kg-core1:~/personal/projects/2023/rust/embedded/esp/espflash $ target/release/espflash board-info 
[2023-05-23T21:05:05Z INFO ] Detected 2 serial ports
[2023-05-23T21:05:05Z INFO ] Ports which match a known common dev board are highlighted
[2023-05-23T21:05:05Z INFO ] Please select a port
[2023-05-23T21:05:09Z INFO ] Serial port: '/dev/cuaU0'
[2023-05-23T21:05:09Z INFO ] Connecting...
[2023-05-23T21:05:09Z INFO ] Attempting UnixTight reset with default delay...
[2023-05-23T21:05:13Z INFO ] Attempting Classic reset with default delay...
[2023-05-23T21:05:13Z INFO ] Using flash stub
Chip type:         esp32c3 (revision v0.3)
Crystal frequency: 40MHz
Flash size:        4MB
Features:          WiFi, BLE
MAC address:       58:cf:79:05:5c:e8
```
if I try to specify the port, espflash autocorrects it to the wrong port
```
tingo@kg-core1:~/personal/projects/2023/rust/embedded/esp/espflash $ target/release/espflash board-info --port /dev/cuaU0
[2023-05-23T21:06:41Z INFO ] Serial port: '/dev/cuau0'
[2023-05-23T21:06:41Z INFO ] Connecting...
[2023-05-23T21:06:41Z INFO ] Attempting UnixTight reset with default delay...
[2023-05-23T21:06:42Z INFO ] Attempting Classic reset with default delay...
[2023-05-23T21:06:43Z INFO ] Attempting UnixTight reset with extra delay...
[2023-05-23T21:06:44Z INFO ] Attempting Classic reset with extra delay...
[2023-05-23T21:06:45Z INFO ] Attempting UnixTight reset with default delay...
[2023-05-23T21:06:46Z INFO ] Attempting Classic reset with default delay...
[2023-05-23T21:06:46Z INFO ] Attempting UnixTight reset with extra delay...
[2023-05-23T21:06:47Z INFO ] Attempting Classic reset with extra delay...
[2023-05-23T21:06:49Z INFO ] Attempting UnixTight reset with default delay...
[2023-05-23T21:06:49Z INFO ] Attempting Classic reset with default delay...
[2023-05-23T21:06:50Z INFO ] Attempting UnixTight reset with extra delay...
[2023-05-23T21:06:51Z INFO ] Attempting Classic reset with extra delay...
[2023-05-23T21:06:52Z INFO ] Attempting UnixTight reset with default delay...
[2023-05-23T21:06:53Z INFO ] Attempting Classic reset with default delay...
Error: espflash::connection_failed

  × Error while connecting to device
  ╰─▶ Failed to connect to the device
  help: Ensure that the device is connected and the reset and boot pins are not being held down
```
using quotes doesn't help
```
tingo@kg-core1:~/personal/projects/2023/rust/embedded/esp/espflash $ target/release/espflash board-info --port "/dev/cuaU0"
[2023-05-23T21:07:47Z INFO ] Serial port: '/dev/cuau0'
[2023-05-23T21:07:47Z INFO ] Connecting...
[2023-05-23T21:07:47Z INFO ] Attempting UnixTight reset with default delay...
[2023-05-23T21:07:48Z INFO ] Attempting Classic reset with default delay...
[2023-05-23T21:07:48Z INFO ] Attempting UnixTight reset with extra delay...
[2023-05-23T21:07:49Z INFO ] Attempting Classic reset with extra delay...
[2023-05-23T21:07:51Z INFO ] Attempting UnixTight reset with default delay...
[2023-05-23T21:07:51Z INFO ] Attempting Classic reset with default delay...
[2023-05-23T21:07:52Z INFO ] Attempting UnixTight reset with extra delay...
[2023-05-23T21:07:53Z INFO ] Attempting Classic reset with extra delay...
[2023-05-23T21:07:54Z INFO ] Attempting UnixTight reset with default delay...
[2023-05-23T21:07:55Z INFO ] Attempting Classic reset with default delay...
[2023-05-23T21:07:55Z INFO ] Attempting UnixTight reset with extra delay...
[2023-05-23T21:07:57Z INFO ] Attempting Classic reset with extra delay...
[2023-05-23T21:07:58Z INFO ] Attempting UnixTight reset with default delay...
[2023-05-23T21:07:58Z INFO ] Attempting Classic reset with default delay...
Error: espflash::connection_failed

  × Error while connecting to device
  ╰─▶ Failed to connect to the device
  help: Ensure that the device is connected and the reset and boot pins are not being held down
```
using the environment variable doesn't help
```
tingo@kg-core1:~/personal/projects/2023/rust/embedded/esp/espflash $ ESPFLASH_PORT="/dev/cuaU0" target/release/espflash board-info
[2023-05-23T21:09:01Z INFO ] Serial port: '/dev/cuau0'
[2023-05-23T21:09:01Z INFO ] Connecting...
[2023-05-23T21:09:01Z INFO ] Attempting UnixTight reset with default delay...
[2023-05-23T21:09:02Z INFO ] Attempting Classic reset with default delay...
[2023-05-23T21:09:03Z INFO ] Attempting UnixTight reset with extra delay...
[2023-05-23T21:09:04Z INFO ] Attempting Classic reset with extra delay...
[2023-05-23T21:09:05Z INFO ] Attempting UnixTight reset with default delay...
[2023-05-23T21:09:06Z INFO ] Attempting Classic reset with default delay...
[2023-05-23T21:09:06Z INFO ] Attempting UnixTight reset with extra delay...
[2023-05-23T21:09:07Z INFO ] Attempting Classic reset with extra delay...
[2023-05-23T21:09:09Z INFO ] Attempting UnixTight reset with default delay...
[2023-05-23T21:09:09Z INFO ] Attempting Classic reset with default delay...
[2023-05-23T21:09:10Z INFO ] Attempting UnixTight reset with extra delay...
[2023-05-23T21:09:11Z INFO ] Attempting Classic reset with extra delay...
[2023-05-23T21:09:12Z INFO ] Attempting UnixTight reset with default delay...
[2023-05-23T21:09:13Z INFO ] Attempting Classic reset with default delay...
Error: espflash::connection_failed

  × Error while connecting to device
  ╰─▶ Failed to connect to the device
  help: Ensure that the device is connected and the reset and boot pins are not being held down
```
version
```
tingo@kg-core1:~/personal/projects/2023/rust/embedded/esp/espflash $ target/release/espflash --version
espflash 2.0.0-rc.4
```
older versions behave the same way. This is on FreeBSD:
```
tingo@kg-core1:~/personal/projects/2023/rust/embedded/esp/espflash $ freebsd-version -ku
13.2-RELEASE
13.2-RELEASE
tingo@kg-core1:~/personal/projects/2023/rust/embedded/esp/espflash $ uname -a
FreeBSD kg-core1.kg4.no 13.2-RELEASE FreeBSD 13.2-RELEASE releng/13.2-n254617-525ecfdad597 GENERIC amd64
```

jessebraham, 2023-05-29:
Sorry about that, don't think any of the maintainers/contributors are BSD users 😅 

I think the issue stems from this line:  
https://github.com/esp-rs/espflash/blob/main/espflash/src/cli/serial.rs#L82

Under the hood `eq_ignore_ascii_case` converts both items to lowercase then compares them, so it's no wonder it's not working correctly.

Obviously changing this to `eq` instead should resolve the issue, but then makes any user input on the command-line case sensitive. This probably isn't the end of the world, though.

I suppose we could `cfg` in a special case for BSD here as well.

Let me sit on this issue for a bit, and I'll push a fix when I decide how I want to address it.

jessebraham, 2023-05-29:
Do you happen to know if the same format is used for serial devices in `dragonfly`, `openbsd`, and/or `netbsd` as well?

tingox, 2023-05-29:
Hmm, let me check
Dragonfly [ucom(4)](https://leaf.dragonflybsd.org/cgi/web-man?command=ucom) - yes
NetBSD [ucom(4)](https://man.netbsd.org/ucom.4) - yes
OpenBSD [ucom(4)](https://man.openbsd.org/ucom.4) - yes
thanks for looking into this.

PR #415 (2023-05-31): Fix handling of serial ports on BSD systems
@tingox would you mind verifying this fixes the issue for you, please?

Closes #409