esp-rs/espflash #436: Timeout while running FlashDeflateData command

hbhong, 2023-06-21:
My mcu is ESP32C3 with 16MB flash size, when I am running the following command to flash I got a timeout error 
> cargo espflash flash --release  --partition-table ./default_16MB_one_ota.csv 

the output:
[2023-06-21T14:06:50Z INFO ] Serial port: '/dev/ttyACM0'
[2023-06-21T14:06:50Z INFO ] Connecting...
[2023-06-21T14:06:51Z INFO ] Using flash stub
    Finished release [optimized] target(s) in 0.21s
Chip type:         esp32c3 (revision v0.3)
Crystal frequency: 40MHz
Flash size:        16MB
Features:          WiFi, BLE
MAC address:       60:55:f9:77:6d:6c
Bootloader:        /drn/RustProjects/pomodoro/target/riscv32imc-esp-espidf/release/build/esp-idf-sys-9c7a5783fe78fcfb/out/build/bootloader/bootloader.bin
Partition table:   ./default_16MB_one_ota.csv
App/part. size:    8,298,752/12,976,128 bytes, 63.95%
[00:00:01] [========================================]      13/13      0x0                                                                                                                                                                                                 
[00:00:00] [========================================]       1/1       0x8000                                                                                                                                                                                              
Error: espflash::timeout

  × Communication error while flashing device
  ╰─▶ Timeout while running FlashDeflateData command
here is my partition table file:
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0xc60000,
spiffs,   data, spiffs,  0xc70000,0x370000,

I just found that when the size of my app is reduced, the program can be flashed successfully. However, when I add more codes, timeout issues occur.
I would greatly appreciate any suggestions or assistance you can provide！

jessebraham, 2023-06-22:
Thank you for opening an issue, sorry that you're having issues.

Just out of curiosity, would you be able to share which version of `espflash` you are using as well as which development kit or board you are using? If you are not using the latest version of `espflash` (which is `2.0.0`) then please try again using this version. I don't believe we offer an ESP32-C3 module with 16MB of flash, so it seems to me that this is some third-party offering.

This problem is reminiscent of a previous issue we faced where the watchdog timer was causing a reset, however this issue should have been resolved in a previous PR. However it's possible there is another issue which we had missed previously.

hbhong, 2023-06-22:
@jessebraham Thank for your reply.  My `espflash` is `2.0.0` version. and I'm using  third-party development board, its default flash size is 4MB, because my app is a little bigger, so I replaced it with 16MB external flash.
[image]

MabezDev, 2023-06-23:
@hbhong How long does it flash for before reporting this error? The timeout for erase is 120 seconds, perhaps erasing 16MB could take longer?

hbhong, 2023-06-24:
@MabezDev with my current partition table, it takes 02:25. But I think that flash memory produced by different manufacturers may have speed differences.

hbhong, 2023-06-24:
I resolved this problem by recompiling the espflash,  I change the value of `ERASE_WRITE_TIMEOUT_PER_MB` from 40 to 400 (I know this value is too large, but I'm sorry I don't have much time to measure how much time is more appropriate. For me, this value ensures that I don't have to worry about such incidents happening again.)

jessebraham, 2023-06-24:
Thank you for the additional information! I will discuss this issue with some colleagues next week and try to determine what the best path forward for this is; possibly it's been solved in another tool already.

Perhaps we can simply increase the timeout length, determined by some experimentation. Otherwise, as you suggested maybe we can allow the user to specify this value in some way.

brainstorm, 2023-10-06:
I just bumped into this issue with a LOLIN S3 1.0.0 board:

```
% cargo run
    Finished dev [unoptimized + debuginfo] target(s) in 0.31s
     Running `espflash flash --monitor target/xtensa-esp32s3-none-elf/debug/pancake-legend-lights`
[2023-10-06T03:53:46Z INFO ] Detected 8 serial ports
[2023-10-06T03:53:46Z INFO ] Ports which match a known common dev board are highlighted
[2023-10-06T03:53:46Z INFO ] Please select a port
[2023-10-06T03:53:53Z INFO ] Serial port: '/dev/cu.usbserial-12420'
[2023-10-06T03:53:53Z INFO ] Connecting...
[2023-10-06T03:53:53Z INFO ] Using flash stub
Chip type:         esp32s3 (revision v0.1)
Crystal frequency: 40MHz
Flash size:        16MB
Features:          WiFi, BLE
MAC address:       68:b6:b3:28:6c:bc
App/part. size:    613,024/16,384,000 bytes, 3.74%
[00:00:01] [========================================]      14/14      0x0
[00:00:00] [========================================]       1/1       0x8000
Error: espflash::timeout

  × Communication error while flashing device
  ╰─▶ Timeout while running FlashDeflateData command
  ```

SergioGasquez, 2023-10-11:
Hi! Would any of the affected users mind testing #488?
```
cargo install --git https://github.com/SergioGasquez/espflash --branch fix/deflate-timeout cargo-espflash espflash
```

closed 2023-10-11 by jessebraham

bugadani, 2023-11-20:
Issue persists with https://github.com/esp-rs/espflash/commit/2c7ee4c66cee14a0b3a1222ccbf08782fdab7a23

```
[2023-11-20T08:52:38Z INFO ] Using flash stub
   Compiling esp32c6-hal v0.6.0 (C:\_Hobby\esp-hal\esp32c6-hal)
   Compiling esp-wifi v0.1.0 (https://github.com/esp-rs/esp-wifi.git?rev=0db7a70#0db7a706)
   Compiling card_io_fw v0.1.0 (C:\_Hobby\CardIO\card-io-fw)
    Finished dev [optimized + debuginfo] target(s) in 8.64s
Chip type:         esp32c6 (revision v0.0)
Crystal frequency: 40MHz
Flash size:        8MB
Features:          WiFi 6, BT 5
MAC address:       40:4c:ca:41:db:68
Partition table:   \\?\C:\_Hobby\CardIO\card-io-fw\partitions.csv
[2023-11-20T08:52:47Z INFO ] Erasing otadata (Data(Ota))...
App/part. size:    1,373,072/2,097,152 bytes, 65.47%
[00:00:00] [========================================]      13/13      0x0                                                                                                                                                                                                           
[00:00:00] [========================================]       1/1       0x8000                                                                                                                                                                                                        
Error: espflash::timeout

  × Communication error while flashing device
  ╰─▶ Timeout while running FlashDeflateData command
```