esp-rs/espflash #440: SPI operation failed when flashing .crt being not multiple of 4

alessio-perugini, 2023-07-03:
:wave: I was trying to flash a x509 certificate bundle in a .crt format to `0x3c0000` address. Unfortunately, the tool returns the following error:
>Error:   × The bootloader returned an error
  ├─▶ Error while running FlashDeflateEnd command
  ╰─▶ SPI operation failed

Talking to other guys on the team we hypothesized that we could have some padding problems, so I added a few bytes at the end. Surprisingly making the .crt file multiple of 4 solved the problem, and the `espflash` worked perfectly!

`stat -c %s google.crt`
| before | after |
|--------|--------|
| 3649 Bytes | 3652 Bytes |
 
 ## Steps to reproduce the problem

```bash
openssl s_client -showcerts -connect google.com:443 </dev/null | sed -n -e '/-.BEGIN/,/-.END/ p' > google.pem
openssl x509 -outform der -in google.pem -out google.crt
espflash write-bin -p /dev/ttyACM0 -b 921600 0x3C0000 google.crt
```

>[2023-07-03T14:49:58Z INFO ] Serial port: '/dev/ttyACM0'
[2023-07-03T14:49:58Z INFO ] Connecting...
[2023-07-03T14:49:58Z INFO ] Using flash stub
[2023-07-03T14:49:59Z WARN ] Setting baud rate higher than 115,200 can cause issues
Chip type:         esp32s3 (revision v0.1)
Crystal frequency: 40MHz
Flash size:        8MB
Features:          WiFi, BLE
MAC address:       dc:54:75:c3:01:18
[00:00:00] [========================================]       2/2       0x3C0000                                                                                                 Error:   × The bootloader returned an error
  ├─▶ Error while running FlashDeflateEnd command
  ╰─▶ SPI operation failed
  
 ## Steps to fix the problem
```bash
openssl s_client -showcerts -connect google.com:443 </dev/null | sed -n -e '/-.BEGIN/,/-.END/ p' > google.pem
openssl x509 -outform der -in google.pem -out google.crt
dd if=/dev/zero of=google.crt bs=1 count=1 seek=3651
espflash write-bin -p /dev/ttyACM0 -b 921600 0x3C0000 google.crt
```
>[2023-07-03T15:01:31Z INFO ] Serial port: '/dev/ttyACM0'
[2023-07-03T15:01:31Z INFO ] Connecting...
[2023-07-03T15:01:32Z INFO ] Using flash stub
[2023-07-03T15:01:32Z WARN ] Setting baud rate higher than 115,200 can cause issues
Chip type:         esp32s3 (revision v0.1)
Crystal frequency: 40MHz
Flash size:        8MB
Features:          WiFi, BLE
MAC address:       dc:54:75:c3:01:18
[00:00:00] [========================================]       2/2       0x3C0000  

## Additional information
I'm running on Ubuntu 22.04
```bash
$ uname -a
Linux ale 5.19.0-45-generic #46~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Wed Jun 7 15:06:04 UTC 20 x86_64 x86_64 x86_64 GNU/Linux

$ espflash --version
espflash 2.0.0
```
I was trying the flash on one the new [Arduino Uno R4 WiFi (datasheet)](https://docs.arduino.cc/static/a6e2cf697c7479d3ba3eca17dbe6dc2a/ABX00087-datasheet.pdf)

SergioGasquez, 2023-12-11:
Hi! I just tried reproducing the issue and I was not able to:
```
❯ openssl s_client -showcerts -connect google.com:443 </dev/null | sed -n -e '/-.BEGIN/,/-.END/ p' > google.pem

depth=2 C = US, O = Google Trust Services LLC, CN = GTS Root R1
verify return:1
depth=1 C = US, O = Google Trust Services LLC, CN = GTS CA 1C3
verify return:1
depth=0 CN = *.google.com
verify return:1
DONE

❯ openssl x509 -outform der -in google.pem -out google.crt

❯ espflash write-bin -p /dev/ttyACM0 -b 921600 0x3C0000 google.crt

[2023-12-11T15:58:32Z INFO ] Serial port: '/dev/ttyACM0'
[2023-12-11T15:58:32Z INFO ] Connecting...
[2023-12-11T15:58:32Z INFO ] Using flash stub
[2023-12-11T15:58:33Z WARN ] Setting baud rate higher than 115,200 can cause issues
Chip type:         esp32s3 (revision v0.1)
Crystal frequency: 40MHz
Flash size:        8MB
Features:          WiFi, BLE
MAC address:       60:55:f9:f5:20:d4
[00:00:00] [========================================]       2/2       0x3C0000                                                                                                                                                   

❯ stat -c %s google.crt
3701
```

I also tried with other sizes:
```
❯ dd if=/dev/zero of=google.crt bs=1 count=1 seek=3748
1+0 records in
1+0 records out
1 byte copied, 1.329e-05 s, 75.2 kB/s

❯ stat -c %s google.crt
3749

❯ espflash write-bin -p /dev/ttyACM0 -b 921600 0x3C0000 google.crt

[2023-12-11T15:57:43Z INFO ] Serial port: '/dev/ttyACM0'
[2023-12-11T15:57:43Z INFO ] Connecting...
[2023-12-11T15:57:43Z INFO ] Using flash stub
[2023-12-11T15:57:44Z WARN ] Setting baud rate higher than 115,200 can cause issues
Chip type:         esp32s3 (revision v0.1)
Crystal frequency: 40MHz
Flash size:        8MB
Features:          WiFi, BLE
MAC address:       60:55:f9:f5:20:d4
[00:00:00] [========================================]       2/2       0x3C0000      
```
I am using espflash with that latest `main` changes (commit [71d7a63](https://github.com/esp-rs/espflash/commit/71d7a630275965d47219048c371702bc587de152)). We did some changes in `main`, could you try reproducing it with this espflash version (`cargo install espflash --git https://github.com/esp-rs/espflash --force`) and see if the issue still persists for you @alessio-perugini?

alessio-perugini, 2023-12-11:
@SergioGasquez Tested on 71d7a63 now it's working flawlessly! :rocket:

closed 2023-12-11 by alessio-perugini