esp-rs/espflash #442: can't connect to NINA-W102

br0kenpixel, 2023-07-06:
I'm trying to use espflash with a u-blox NINA-W102 module connected to an Arduino Nano RP2040 Connect.

My Arduino board is flashed with the [SerialNINAPassthrough](https://github.com/arduino-libraries/WiFiNINA/blob/master/examples/Tools/SerialNINAPassthrough/SerialNINAPassthrough.ino) example.
I can connect to the module using esptool.py just fine:
```
$ esptool.py --port /dev/cu.usbmodem141301 --baud 115200 --before no_reset flash_id
esptool.py v4.3
Serial port /dev/cu.usbmodem141301
WARNING: Pre-connection option "no_reset" was selected. Connection may fail if the chip is not in bootloader or flasher stub mode.
Connecting......
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6 (revision v1.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: [...]
Uploading stub...
Running stub...
Stub running...
Manufacturer: c2
Device: 2815
Detected flash size: 2MB
Hard resetting via RTS pin...
```

However, espflash can't connect to it:
```
$ cargo espflash board-info --baud 115200 --port /dev/cu.usbmodem141301
[2023-07-06T16:33:28Z INFO ] Serial port: '/dev/cu.usbmodem141301'
[2023-07-06T16:33:28Z INFO ] Connecting...
[2023-07-06T16:33:28Z INFO ] Unable to connect, retrying with extra delay...
[2023-07-06T16:33:29Z INFO ] Unable to connect, retrying with default delay...
[2023-07-06T16:33:30Z INFO ] Unable to connect, retrying with extra delay...
[2023-07-06T16:33:31Z INFO ] Unable to connect, retrying with default delay...
[2023-07-06T16:33:32Z INFO ] Unable to connect, retrying with extra delay...
[2023-07-06T16:33:33Z INFO ] Unable to connect, retrying with default delay...
[2023-07-06T16:33:34Z INFO ] Unable to connect, retrying with extra delay...
Error: espflash::connection_failed

  × Error while connecting to device
  ╰─▶ Failed to connect to the device
  help: Ensure that the device is connected and the reset and boot pins are not being held down
```

SergioGasquez, 2023-07-07:
Hi! Do you mind trying to use #387 and see if the issue is solved?
```
cargo install cargo-espflash --git https://github.com/AVee/espflash --branch add_unix_tight_reset
```

br0kenpixel, 2023-07-07:
Hi. I tried it and still no luck.

```
$ cargo espflash board-info --baud 115200 --port /dev/cu.usbmodem141301
[2023-07-07T09:16:31Z INFO ] 🚀 A new version of cargo-espflash is available: v2.0.0
[2023-07-07T09:16:31Z INFO ] Serial port: '/dev/cu.usbmodem141301'
[2023-07-07T09:16:31Z INFO ] Connecting...
[2023-07-07T09:16:31Z INFO ] Attempting UnixTight reset with default delay...
[2023-07-07T09:16:32Z INFO ] Attempting Classic reset with default delay...
[2023-07-07T09:16:32Z INFO ] Attempting UnixTight reset with extra delay...
[2023-07-07T09:16:34Z INFO ] Attempting Classic reset with extra delay...
[2023-07-07T09:16:35Z INFO ] Attempting UnixTight reset with default delay...
[2023-07-07T09:16:35Z INFO ] Attempting Classic reset with default delay...
[2023-07-07T09:16:36Z INFO ] Attempting UnixTight reset with extra delay...
[2023-07-07T09:16:37Z INFO ] Attempting Classic reset with extra delay...
[2023-07-07T09:16:38Z INFO ] Attempting UnixTight reset with default delay...
[2023-07-07T09:16:39Z INFO ] Attempting Classic reset with default delay...
[2023-07-07T09:16:40Z INFO ] Attempting UnixTight reset with extra delay...
[2023-07-07T09:16:41Z INFO ] Attempting Classic reset with extra delay...
[2023-07-07T09:16:42Z INFO ] Attempting UnixTight reset with default delay...
[2023-07-07T09:16:42Z INFO ] Attempting Classic reset with default delay...
Error: espflash::connection_failed

  × Error while connecting to device
  ╰─▶ Failed to connect to the device
  help: Ensure that the device is connected and the reset and boot pins are not being held down
```

I was thinking that maybe the issue is that `espflash` is trying to reset the board, which is not necessary in this case, as the [SerialNINAPassthrough](https://github.com/arduino-libraries/WiFiNINA/blob/master/examples/Tools/SerialNINAPassthrough/SerialNINAPassthrough.ino) sketch forces the ESP32 into download mode. This is why I had to use `--before no_reset` with `esptool.py`. Does `espflash` have a switch for skipping the reset?

SergioGasquez, 2023-10-06:
Hi! Sorry for the late reply, there is no `--before no_reset` equivalent in `espflash` at the moment. Maybe you can try using #487

SergioGasquez, 2023-11-16:
Hi! Would you mind verifying if the issue still persists with the changes on `main`?  
```
cargo install espflash cargo-espflash --git https://github.com/esp-rs/espflash
```