esp-rs/espflash #444: `cargo install` of cargo-espflash failing to build "cargo" dependency

mykmelez, 2023-07-13:
`cargo install` of cargo-espflash started failing for me this morning due to a build failure in the "cargo" dependency:

```
> cd cargo-espflash
> cargo install --path .
…
error[E0063]: missing field `rust_version` in initializer of `NewCrate`
   --> /Users/myk/.cargo/registry/src/index.crates.io-6f17d22bba15001f/cargo-0.71.0/src/cargo/ops/registry.rs:383:14
    |
383 |             &NewCrate {
    |              ^^^^^^^^ missing `rust_version`

For more information about this error, try `rustc --explain E0063`.
error: could not compile `cargo` (lib) due to previous error
error: failed to compile `cargo-espflash v2.0.1-dev (/Users/myk/Projects/espflash/cargo-espflash)`, intermediate artifacts can be found at `/Users/myk/Projects/espflash/target`
```

jessebraham, 2023-07-13:
I've published version `2.0.1` which includes the fix for this problem.

closed 2023-07-13 by jessebraham

kevin-june, 2023-07-13:
@jessebraham @mykmelez I have some fear that this issue will reappear: [cargo v0.72.0](https://crates.io/crates/cargo/0.72.0) was yanked not long after it was released!

For reference - it appears that this issue was introduced in https://github.com/rust-lang/cargo/pull/12041 and is being tracked in https://github.com/rust-lang/cargo/issues/12354.

jessebraham, 2023-07-14:
Thanks for the update, that's some unfortunate timing 😅 For the time being you can still install using `--locked` despite the dependency being yanked. I will keep an eye on the situation.