esp-rs/espflash #450: Error: espflash::connection_failed  (ESP32 + ESP-Prog + JTAG)

systec-ms, 2023-07-24:
Good day, I got an `ESP32 D1 MINI`, which I can program via `espflash` over the serial (micro-usb). 
Additionally, I can flash the esp `blinky` example via the VSCode extension over the ESP-Prog JTAG connection (`⚡️ Flashed Successfully (JTag)`).

However, if I try to flash my rust code via `espflash` over the EPS-Prog JTAG connection, I run into `Error: espflash::connection_failed`.

```
     Running `espflash flash --monitor target/xtensa-esp32-espidf/debug/blinky`
[2023-07-24T14:45:47Z INFO ] Detected 3 serial ports
[2023-07-24T14:45:47Z INFO ] Ports which match a known common dev board are highlighted
[2023-07-24T14:45:47Z INFO ] Please select a port
✔ Remember this serial port for future use? · no
[2023-07-24T14:45:50Z INFO ] Serial port: '/dev/ttyUSB0'
[2023-07-24T14:45:50Z INFO ] Connecting...
[2023-07-24T14:45:51Z INFO ] Unable to connect, retrying with extra delay...
[2023-07-24T14:45:52Z INFO ] Unable to connect, retrying with default delay...
[2023-07-24T14:45:53Z INFO ] Unable to connect, retrying with extra delay...
[2023-07-24T14:45:54Z INFO ] Unable to connect, retrying with default delay...
[2023-07-24T14:45:54Z INFO ] Unable to connect, retrying with extra delay...
[2023-07-24T14:45:56Z INFO ] Unable to connect, retrying with default delay...
[2023-07-24T14:45:56Z INFO ] Unable to connect, retrying with extra delay...
Error: espflash::connection_failed
```

I also tried https://github.com/esp-rs/espflash/pull/387 as mentioned by https://github.com/esp-rs/espflash/issues/394 and https://github.com/esp-rs/espflash/issues/394 using
 `cargo install --git https://github.com/AVee/espflash --branch add_unix_tight_reset espflash` with the same result.

```
    Finished dev [optimized + debuginfo] target(s) in 31.24s
     Running `espflash flash --monitor target/xtensa-esp32-espidf/debug/blinky`
[2023-07-24T14:32:58Z INFO ] 🚀 A new version of espflash is available: v2.0.1
[2023-07-24T14:32:58Z INFO ] Detected 2 serial ports
[2023-07-24T14:32:58Z INFO ] Ports which match a known common dev board are highlighted
[2023-07-24T14:32:58Z INFO ] Please select a port
✔ Remember this serial port for future use? · no
[2023-07-24T14:33:03Z INFO ] Serial port: '/dev/ttyUSB1'
[2023-07-24T14:33:03Z INFO ] Connecting...
[2023-07-24T14:33:03Z INFO ] Attempting UnixTight reset with default delay...
[2023-07-24T14:33:03Z INFO ] Attempting Classic reset with default delay...
[2023-07-24T14:33:04Z INFO ] Attempting UnixTight reset with extra delay...
[2023-07-24T14:33:05Z INFO ] Attempting Classic reset with extra delay...
[2023-07-24T14:33:06Z INFO ] Attempting UnixTight reset with default delay...
[2023-07-24T14:33:07Z INFO ] Attempting Classic reset with default delay...
[2023-07-24T14:33:08Z INFO ] Attempting UnixTight reset with extra delay...
[2023-07-24T14:33:09Z INFO ] Attempting Classic reset with extra delay...
[2023-07-24T14:33:10Z INFO ] Attempting UnixTight reset with default delay...
[2023-07-24T14:33:11Z INFO ] Attempting Classic reset with default delay...
[2023-07-24T14:33:11Z INFO ] Attempting UnixTight reset with extra delay...
[2023-07-24T14:33:12Z INFO ] Attempting Classic reset with extra delay...
[2023-07-24T14:33:14Z INFO ] Attempting UnixTight reset with default delay...
[2023-07-24T14:33:14Z INFO ] Attempting Classic reset with default delay...
Error: espflash::connection_failed

  × Error while connecting to device
  ╰─▶ Failed to connect to the device
  help: Ensure that the device is connected and the reset and boot pins are not being held down

```

I would appreciate any help, thanks in advance.

systec-ms, 2023-07-25:
Could it be that it is not possible to flash Xtensa chips via espflash and jtag (https://github.com/esp-rs/espflash/issues/23)? 

If so, how do I extract the `bootloader.bin`, `partion-tabe.bin` and `binary.bin` to flash it via openocd?

systec-ms, 2023-07-31:
The following works for me:

## Flash via ESP-PROG (JTAG)

### Erase

```bash
openocd-esp32openocd -f /usr/share/openocd-esp32/scripts/interface/ftdi/esp32_devkitj_v1.cfg -f /usr/share/openocd-esp32/scripts/target/esp32.cfg -c "init; reset halt; flash erase_address 0x1000 16773120; exit"
```

### Flash Bootloader

Use `fd bootloader.bin --no-ignore` to find `bootloader.bin` location (build by `esp-idf-sys`).
Typical locations are (note <https://github.com/esp-rs/esp-idf-sys/issues/97>):

```bash
target/xtensa-esp32-espidf/debug/build/esp-idf-sys-5088a6f49af329e1/out/build/bootloader/bootloader.bin
target/xtensa-esp32-espidf/release/build/esp-idf-sys-0f6f63297e12c38d/out/build/bootloader/bootloader.bin
```

```bash
openocd-esp32openocd -f /usr/share/openocd-esp32/scripts/interface/ftdi/esp32_devkitj_v1.cfg -f /usr/share/openocd-esp32/scripts/target/esp32.cfg -c "program bootloader.bin 0x1000; exit"
```

### Flash Partition Table

Use `fd partition-table.bin --no-ignore` to find `partition-table.bin` or use `espflash partition-table --to-binary` to export the partition-table binary.

Typical locations are (note <https://github.com/esp-rs/esp-idf-sys/issues/97>):

```bash
target/xtensa-esp32-espidf/debug/build/esp-idf-sys-5088a6f49af329e1/out/build/partition_table/partition-table.bin
target/xtensa-esp32-espidf/release/build/esp-idf-sys-0f6f63297e12c38d/out/build/partition_table/partition-table.bin
```

Export binary:

```bash
espflash partition-table --to-binary --output partition-table.bin .embuild/espressif/esp-idf/release-v4.4/components/partition_table/partitions_singleapp.csv
```

Flash binary:

```bash
openocd-esp32openocd -f /usr/share/openocd-esp32/scripts/interface/ftdi/esp32_devkitj_v1.cfg -f /usr/share/openocd-esp32/scripts/target/esp32.cfg -c "program partition-table.bin 0x8000; exit"
```

### Flash Application

Export binary (change `/debug/` to `/release/`):

```bash
espflash save-image --chip esp32 target/xtensa-esp32-espidf/release/app app.bin
```

Flash binary:

```bash
openocd-esp32openocd -f /usr/share/openocd-esp32/scripts/interface/ftdi/esp32_devkitj_v1.cfg -f /usr/share/openocd-esp32/scripts/target/esp32.cfg -c "program app.bin 0x10000 reset exit"
```

Should be `verify reset exit` but `verify` is currently unsupported

SergioGasquez, 2023-10-06:
Hi! Sorry for the late reply, Can try using https://github.com/esp-rs/espflash/pull/487 and see if it solves the issue? You should be able to flash Xtensa via JTAG

SergioGasquez, 2023-11-16:
Hi! Would you mind verifying if the issue still persists with the changes on `main`?  
```
cargo install espflash cargo-espflash --git https://github.com/esp-rs/espflash
```

systec-ms, 2023-11-16:
Sorry for the late replay, sadly I no luck:

```bash
❯ espflash -V
espflash 3.0.0-dev
```

```bash
❯ espflash board-info
[2023-11-16T10:11:46Z INFO ] Detected 2 serial ports
[2023-11-16T10:11:46Z INFO ] Ports which match a known common dev board are highlighted
[2023-11-16T10:11:46Z INFO ] Please select a port
✔ Remember this serial port for future use? · no
[2023-11-16T10:11:56Z INFO ] Serial port: '/dev/ttyUSB0'
[2023-11-16T10:11:56Z INFO ] Connecting...
Error: espflash::connection_failed

  × Error while connecting to device
  ╰─▶ Failed to connect to the device
  help: Ensure that the device is connected and the reset and boot pins are not being held down

❯ espflash board-info
[2023-11-16T10:12:05Z INFO ] Detected 2 serial ports
[2023-11-16T10:12:05Z INFO ] Ports which match a known common dev board are highlighted
[2023-11-16T10:12:05Z INFO ] Please select a port
✔ Remember this serial port for future use? · no
[2023-11-16T10:12:07Z INFO ] Serial port: '/dev/ttyUSB1'
[2023-11-16T10:12:07Z INFO ] Connecting...
Error: espflash::connection_failed

  × Error while connecting to device
  ╰─▶ Failed to connect to the device
  help: Ensure that the device is connected and the reset and boot pins are not being held down
```

Same result with `espflash flash --monitor target/xtensa-esp32-espidf/release/blinky`.
Flashing still works with:
```
openocd-esp32openocd -f /usr/share/openocd-esp32/scripts/interface/ftdi/esp32_devkitj_v1.cfg -f /usr/share/openocd-esp32/scripts/target/esp32.cfg -c "program target/app.bin 0x10000; reset; exit"
```

So unfortunately, the problem still exists.

SergioGasquez, 2023-11-20:
Have you tried to flash your application with [`esptool`](https://github.com/espressif/esptool)?

SergioGasquez, 2023-11-23:
#521 may fix this issue, mind testing again with `main`?
```
cargo install espflash cargo-espflash --git https://github.com/esp-rs/espflash
```
Sorry for the debugging procedure, but since I can't reproduce the issue, I would like to know if it's still there

systec-ms, 2023-11-24:
Sadly, same result as before.

SergioGasquez, 2023-12-12:
> Have you tried to flash your application with [`esptool`](https://github.com/espressif/esptool)?

Hi! By any chance, did you try `esptool`?

systec-ms, 2023-12-12:
> > Have you tried to flash your application with [`esptool`](https://github.com/espressif/esptool)?
> 
> Hi! By any chance, did you try `esptool`?

I tried it initially, and it worked, but never again since.

SergioGasquez, 2023-12-29:
Do you have anything connected into the [strapping pins](https://docs.espressif.com/projects/esptool/en/latest/esp32/advanced-topics/boot-mode-selection.html?highlight=pins#select-bootloader-mode)?

systec-ms, 2024-01-10:
Sorry again for the late reply, but I no longer have the hardware available, so feel free to close this issue.