esp-rs/espflash #451: Please don't automatically vendor OpenSSL

faern, 2023-07-25:
I tried to install `cargo-espflash` and got compilation errors for OpenSSL (I did not have Perl installed). My system has a perfectly valid OpenSSL installation it can use.

I finally got it to install with `OPENSSL_NO_VENDOR=1 cargo install cargo-espflash`. But I don't think that should be needed to set explicitly. IMHO using the local install should be the default, and vendoring should be the fallback.

https://github.com/esp-rs/espflash/blob/4e7c018954cef041cedc110c3aebb3e91cd15dde/cargo-espflash/Cargo.toml#L40

jessebraham, 2023-07-25:
This dependency was vendored in the first place due to build issues on Windows, see https://github.com/esp-rs/espflash/issues/392. I try to avoid Windows as much as possible so not really sure what the best path forward is, but I would imagine most users don't have a valid `perl` installation (unless this build dependency has changed since the aforementioned issue was closed).

SergioGasquez, 2023-07-26:
We decided to vendor openSSL since not vendoring it also caused issues for some users. I just created #452 adding some documentation in the README to clarify this and make it easy for user that want to opt out of the default vendored openSSL

closed 2023-07-26 by jessebraham

faern, 2023-07-26:
Thanks for the extra documentation :+1: