esp-rs/espflash #453: Error while running FlashDeflateBegin command on 2.0.1

Janrupf, 2023-07-26:
This is probably the same as #195, but I'm not 100% sure.

Writing large binaries to the flash fails **when not using the stub**.

Example invocation:
```
$> dd if=/dev/zero of=/tmp/zero.bin bs=8M count=1
1+0 records in
1+0 records out
8388608 bytes (8.4 MB, 8.0 MiB) copied, 0.00464577 s, 1.8 GB/s

$>  ~/.cargo/bin/espflash write-bin --port /dev/ttyUSB3 --baud 921600 --no-stub 0x10000 /tmp/zero.bin
[2023-07-26T15:23:12Z INFO ] Serial port: '/dev/ttyUSB3'
[2023-07-26T15:23:12Z INFO ] Connecting...
Chip type:         esp32 (revision v1.0)
Crystal frequency: 40MHz
Flash size:        16MB
Features:          WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC address:       9c:9c:1f:0b:45:68
Error:   × The bootloader returned an error
  ├─▶ Error while running FlashDeflateBegin command
  ╰─▶ Other
```

I'm trying to flash without the stub, since I have flash encryption in development mode enabled (so I need the ROM bootloader to transparently encrypt while I'm flashing).

The ESP32 I'm using has 16MB of flash, but I can also reproduce the problem if I reduce the `dd` size to `2M`.

Janrupf, 2023-07-26:
I'm more or less clueless what is happening, but here are my observations so far:
- `esptool.py` has no problems writing the file with `--no-stub`
- enabling or disabling compression doesn't change a thing (I modified `espflash` source code for that)
- the "Other" error code seems to be error `0x06`, which helpfully states "Failed to act on received message" [here](https://docs.espressif.com/projects/esptool/en/latest/esp32/advanced-topics/serial-protocol.html#rom-loader-errors)
- I couldn't find any indication of what `esptool.py` does differently from `espflash` in that regard, they seem to more or less issue the same commands

Janrupf, 2023-07-26:
Alright, I think I figured it out, `espflash` does not send the `SpiSetParams` command, I'll hopefully will get a PR ready soon.

georgik, 2023-08-07:
I've encountered similar problem during an attempt to flash 4 MB binary on S3-BOX calling `flasher.write_bin_to_flash` method. The big binary failed with:
```
Flash error: RomError(RomError { command: FlashDeflateBegin, kind: Other })
```
I was able to flash smaller file around 1 MB, but it failed with anything bigger than ~2 MB

SergioGasquez, 2023-10-06:
> Alright, I think I figured it out, `espflash` does not send the `SpiSetParams` command, I'll hopefully will get a PR ready soon.

Hi! I was about to take a look at this issue, did you have a chance to look into it?

SergioGasquez, 2023-10-06:
Seems like @georgik error is different from the original error of the issue. @georgik issue also happens with big binaries, e.g: [`spooky-maze-esp32-s3-box.bin`](https://github.com/georgik/esp32-spooky-maze-game/releases/download/v0.5.0/spooky-maze-esp32-s3-box.bin) and it's the following error:
```
❯ espflash write-bin 0x0 spooky-maze-esp32-s3-box.bin
[2023-10-06T10:44:53Z INFO ] Serial port: '/dev/ttyACM0'
[2023-10-06T10:44:53Z INFO ] Connecting...
[2023-10-06T10:44:54Z INFO ] Using flash stub
[2023-10-06T10:44:54Z WARN ] Setting baud rate higher than 115,200 can cause issues
Chip type:         esp32s3 (revision v0.1)
Crystal frequency: 40MHz
Flash size:        16MB
Features:          WiFi, BLE
MAC address:       7c:df:a1:e1:c0:c4
Error: espflash::timeout

  × Error while connecting to device
  ╰─▶ Timeout while running FlashDeflateData command
```

Increasing the [default timeout](https://github.com/esp-rs/espflash/blob/main/espflash/src/command.rs#L10) seems to fix this issue, not sure if this is the best solution though

SergioGasquez, 2023-10-11:
This should now be fixed with #489, if anyone reencounters the issue, feel free to reopen it

closed 2023-10-11 by SergioGasquez