esp-rs/espflash #454: GLIBC version mismatch when running pre-compiled 2.0.1 binaries on a Raspberry Pi 3B+

LeCyberDucky, 2023-07-29:
Howdy!

I'm trying to run espflash on my Raspberry Pi 3B+. I downloaded the 2.0.1 [espflash-armv7-unknown-linux-gnueabihf.zip](https://github.com/esp-rs/espflash/releases/download/v2.0.1/espflash-armv7-unknown-linux-gnueabihf.zip) binary here from GitHub and placed it on my Raspberry Pi. 

When executing `espflash` on the Pi, I get the following errors: 

```
espflash: /lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.33' not found (required by espflash)
espflash: /lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.32' not found (required by espflash)
espflash: /lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.34' not found (required by espflash)
```
Version 1.7.0 of `espflash` does not yield such errors. 

---
I have not yet tried reflashing my Raspberry Pi with a fresh version of Raspbian. What I have done, is following [this guide](https://www.makeuseof.com/tag/raspberry-pi-update-raspbian-os/) in an attempt to ensure that my Raspberry Pi is up-to-date. 

Running `ldd --version` on the Pi yields: 
```
ldd (Debian GLIBC 2.31-13+rpt2+rpi1+deb11u5) 2.31
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Written by Roland McGrath and Ulrich Drepper.
```

Further, `uname -a` gives:
`Linux raspberrypi 6.1.41-v7+ #1667 SMP Wed Jul 26 17:52:08 BST 2023 armv7l GNU/Linux`

And, finally, running `cat /etc/os-release` produces: 
```
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

I'm happy to provide more information, if needed.

LeCyberDucky, 2023-08-02:
As a small update, I can tell that creating a release targeting Ubuntu 20.04 instead of 22.04, as done in this example https://github.com/LeCyberDucky/espflash/commit/57265f5ca4d61a35857fb056f893071dd9343dd0 yields a binary that will run on my Raspberry Pi without GLIBC errors. https://github.com/LeCyberDucky/espflash/releases/tag/v2.0.1

Thanks to the esp-rs matrix channel for the help :)

jessebraham, 2023-09-18:
This should be resolved by #476, we will have a new release out in a week or two and new binaries will be built at that time.

closed 2023-09-18 by jessebraham