esp-rs/espflash #456: cargo install cargo-espflash fails with error[E0599]

malyjak, 2023-08-09:
While running `cargo install cargo-espflash` on Windows with a fresh rust installation today, I encountered this error:
```
   Compiling git2-curl v0.18.0
error[E0599]: `Vec<u8>` is not an iterator
   --> C:\Users\malyyjak\.cargo\registry\src\github.com-1ecc6299db9ec823\cargo-0.72.2\src/cargo\core\compiler\future_incompat.rs:275:18
    |
275 |                 .map(|v| String::from_utf8(v).expect("utf8"))
    |                  ^^^ `Vec<u8>` is not an iterator; try calling `.into_iter()` or `.iter()`
    |
   ::: C:\Users\malyyjak\.rustup\toolchains\nightly-2023-02-28-x86_64-pc-windows-msvc\lib/rustlib/src/rust\library\alloc\src\vec\mod.rs:400:1
    |
400 | pub struct Vec<T, #[unstable(feature = "allocator_api", issue = "32838")] A: Allocator = Global> {
    | ------------------------------------------------------------------------------------------------ doesn't satisfy `Vec<u8>: Iterator`
    |
    = note: the following trait bounds were not satisfied:
            `Vec<u8>: Iterator`
            which is required by `&mut Vec<u8>: Iterator`
            `[u8]: Iterator`
            which is required by `&mut [u8]: Iterator`
    = help: items from traits can only be used if the trait is in scope
help: the following trait is implemented but not in scope; perhaps add a `use` for it:
    |
36  | use winnow::parser::Parser;
    |

error[E0599]: `Vec<u8>` is not an iterator
    --> C:\Users\malyyjak\.cargo\registry\src\github.com-1ecc6299db9ec823\cargo-0.72.2\src/cargo\core\compiler\mod.rs:1644:26
     |
1644 |                         .map(|v| String::from_utf8(v).expect("utf8"))
     |                          ^^^ `Vec<u8>` is not an iterator; try calling `.into_iter()` or `.iter()`
     |
    ::: C:\Users\malyyjak\.rustup\toolchains\nightly-2023-02-28-x86_64-pc-windows-msvc\lib/rustlib/src/rust\library\alloc\src\vec\mod.rs:400:1
     |
400  | pub struct Vec<T, #[unstable(feature = "allocator_api", issue = "32838")] A: Allocator = Global> {
     | ------------------------------------------------------------------------------------------------ doesn't satisfy `Vec<u8>: Iterator`
     |
     = note: the following trait bounds were not satisfied:
             `Vec<u8>: Iterator`
             which is required by `&mut Vec<u8>: Iterator`
             `[u8]: Iterator`
             which is required by `&mut [u8]: Iterator`
     = help: items from traits can only be used if the trait is in scope
help: the following trait is implemented but not in scope; perhaps add a `use` for it:
     |
34   | use winnow::parser::Parser;
     |

error[E0599]: `Vec<u8>` is not an iterator
    --> C:\Users\malyyjak\.cargo\registry\src\github.com-1ecc6299db9ec823\cargo-0.72.2\src/cargo\core\compiler\mod.rs:1678:22
     |
1678 |                     .map(|v| String::from_utf8(v).expect("utf8"))
     |                      ^^^ `Vec<u8>` is not an iterator; try calling `.into_iter()` or `.iter()`
     |
    ::: C:\Users\malyyjak\.rustup\toolchains\nightly-2023-02-28-x86_64-pc-windows-msvc\lib/rustlib/src/rust\library\alloc\src\vec\mod.rs:400:1
     |
400  | pub struct Vec<T, #[unstable(feature = "allocator_api", issue = "32838")] A: Allocator = Global> {
     | ------------------------------------------------------------------------------------------------ doesn't satisfy `Vec<u8>: Iterator`
     |
     = note: the following trait bounds were not satisfied:
             `Vec<u8>: Iterator`
             which is required by `&mut Vec<u8>: Iterator`
             `[u8]: Iterator`
             which is required by `&mut [u8]: Iterator`
     = help: items from traits can only be used if the trait is in scope
help: the following trait is implemented but not in scope; perhaps add a `use` for it:
     |
34   | use winnow::parser::Parser;
     |

For more information about this error, try `rustc --explain E0599`.
error: could not compile `cargo` due to 3 previous errors
error: failed to compile `cargo-espflash v2.0.1`, intermediate artifacts can be found at `C:\Users\malyyjak\AppData\Local\Temp\cargo-installXErm2R`
```

My rustc version is:
```
rustc --version
rustc 1.69.0-nightly (7281249a1 2023-02-27)
```

Installing 1.7.0 instead of 2.0.1 solved the problem. Can someone look into this? Thanks!

SergioGasquez, 2023-08-10:
Can you try with a more recent Rust version? On Linux, I also get an error when installing it with the same nightly as you, but just tried with `1.73.0-nightly (03a119b0b 2023-08-07)` and `rustc 1.71.1 (eb26296b5 2023-08-03)`  and it worked fine.

malyjak, 2023-08-21:
Sorry for getting back to you this late.

Using latest `nightrly`: `rustc 1.74.0-nightly (5c6a7e71c 2023-08-20)` i get no error during installation. However, when I try to build the basic Rust project, I encounter this error:

```
   Compiling node v0.1.0 (/home/user/Projects/ota_toolkit/IoT_Applications/node_device/rust/node)
error: linking with `ldproxy` failed: exit status: 1
  |
  = note: LC_ALL="C" PATH="/home/user/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/bin:/home/user/.local/bin:/home/user/.cargo/bin:/home/user/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin" VSLANG="1033" "ldproxy" "/tmp/rustc718u2c/symbols.o" "/home/user/Projects/ota_toolkit/IoT_Applications/node_device/rust/node/target/riscv32imc-esp-espidf/debug/deps/node-9aa38a7b29a9e2da.13p6tamg2g86x0ao.rcgu.o" "/home/user/Projects/ota_toolkit/IoT_Applications/node_device/rust/node/target/riscv32imc-esp-espidf/debug/deps/node-9aa38a7b29a9e2da.1e4z63t8pkue7ce4.rcgu.o" "/home/user/Projects/ota_toolkit/IoT_Applications/node_device/rust/node/target/riscv32imc-esp-espidf/debug/deps/node-9aa38a7b29a9e2da.1uih5d333rfc6tdw.rcgu.o" "/home/user/Projects/ota_toolkit/IoT_Applications/node_device/rust/node/target/riscv32imc-esp-espidf/debug/deps/node-9aa38a7b29a9e2da.1xqvkx2xjswnlxu5.rcgu.o" "/home/user/Projects/ota_toolkit/IoT_Applications/node_device/rust/node/target/riscv32imc-esp-espidf/debug/deps/node-9aa38a7b29a9e2da.20a3tigesorundxy.rcgu.o" "/home/user/Projects/ota_toolkit/IoT_Applications/node_device/rust/node/target/riscv32imc-esp-espidf/debug/deps/node-9aa38a7b29a9e2da.23sg683232sts02b.rcgu.o" "/home/user/Projects/ota_toolkit/IoT_Applications/node_device/rust/node/target/riscv32imc-esp-espidf/debug/deps/node-9aa38a7b29a9e2da.29jqi0ctz7m9ojzx.rcgu.o" "/home/user/Projects/ota_toolkit/IoT_Applications/node_device/rust/node/target/riscv32imc-esp-espidf/debug/deps/node-9aa38a7b29a9e2da.2bwh867u3reqpgrl.rcgu.o" "/home/user/Projects/ota_toolkit/IoT_Applications/node_device/rust/node/target/riscv32imc-esp-espidf/debug/deps/node-9aa38a7b29a9e2da.2d04gsvi5c49fw9c.rcgu.o" "/home/user/Projects/ota_toolkit/IoT_Applications/node_device/rust/node/target/riscv32imc-esp-espidf/debug/deps/node-9aa38a7b29a9e2da.2dmorvzoab9ghnei.rcgu.o" "/home/user/Projects/ota_toolkit/IoT_Applications/node_device/rust/node/target/riscv32imc-esp-espidf/debug/deps/node-9aa38a7b29a9e2da.2g1cqf9sahafxn5v.rcgu.o" "/home/user/Projects/ota_toolkit/IoT_Applications/node_device/rust/node/target/riscv32imc-esp-espidf/debug/deps/node-9aa38a7b29a9e2da.2g9j3wlhk6qnhx4p.rcgu.o" "/home/user/Projects/ota_toolkit/IoT_Applications/node_device/rust/node/target/riscv32imc-esp-espidf/debug/deps/node-9aa38a7b29a9e2da.2o4eojn1zgj2fug7.rcgu.o" "/home/user/Projects/ota_toolkit/IoT_Applications/node_device/rust/node/target/riscv32imc-esp-espidf/debug/deps/node-9aa38a7b29a9e2da.2qhh1ziuq3216ss7.rcgu.o" "/home/user/Projects/ota_toolkit/IoT_Applications/node_device/rust/node/target/riscv32imc-esp-espidf/debug/deps/node-9aa38a7b29a9e2da.2rovc5q5cfy2had4.rcgu.o" "/home/user/Projects/ota_toolkit/IoT_Applications/node_device/rust/node/target/riscv32imc-esp-espidf/debug/deps/node-9aa38a7b29a9e2da.2s31alx28fcj5bai.rcgu.o" "/home/user/Projects/ota_toolkit/IoT_Applications/node_device/rust/node/target/riscv32imc-esp-espidf/debug/deps/node-9aa38a7b29a9e2da.2ve6n1o6on814vzr.rcgu.o" "/home/user/Projects/ota_toolkit/IoT_Applications/node_device/rust/node/target/riscv32imc-esp-espidf/debug/deps/node-9aa38a7b29a9e2da.2vssig4oxe9ihbsk.rcgu.o" "/home/user/Projects/ota_toolkit/IoT_Applications/node_device/rust/node/target/riscv32imc-esp-espidf/debug/deps/node-9aa38a7b29a9e2da.2ycobtjkcuqiw39g.rcgu.o" "/home/user/Projects/ota_toolkit/IoT_Applications/node_device/rust/node/target/riscv32imc-esp-espidf/debug/deps/node-9aa38a7b29a9e2da.2yp2tdti3ez6jl79.rcgu.o" "/home/user/Projects/ota_toolkit/IoT_Applications/node_device/rust/node/target/riscv32imc-esp-espidf/debug/deps/node-9aa38a7b29a9e2da.36rknmxo6ckasn0e.rcgu.o" "/home/user/Projects/ota_toolkit/IoT_Applications/node_device/rust/node/target/riscv32imc-esp-espidf/debug/deps/node-9aa38a7b29a9e2da.3gspqgyfhtjicszw.rcgu.o" "/home/user/Projects/ota_toolkit/IoT_Applications/node_device/rust/node/target/riscv32imc-esp-espidf/debug/deps/node-9aa38a7b29a9e2da.3kvkl3hw8nj4ku3j.rcgu.o" "/home/user/Projects/ota_toolkit/IoT_Applications/node_device/rust/node/target/riscv32imc-esp-espidf/debug/deps/node-9aa38a7b29a9e2da.3rytuszwopxgzhlg.rcgu.o" "/home/user/Projects/ota_toolkit/IoT_Applications/node_device/rust/node/target/riscv32imc-esp-espidf/debug/deps/node-9aa38a7b29a9e2da.3tme9of8jgz7ztns.rcgu.o" "/home/user/Projects/ota_toolkit/IoT_Applications/node_device/rust/node/target/riscv32imc-esp-espidf/debug/deps/node-9aa38a7b29a9e2da.3vieaypisfevkayn.rcgu.o" "/home/user/Projects/ota_toolkit/IoT_Applications/node_device/rust/node/target/riscv32imc-esp-espidf/debug/deps/node-9aa38a7b29a9e2da.3xyax8qij1uwcx0p.rcgu.o" "/home/user/Projects/ota_toolkit/IoT_Applications/node_device/rust/node/target/riscv32imc-esp-espidf/debug/deps/node-9aa38a7b29a9e2da.3zyad5chd91vasbf.rcgu.o" "/home/user/Projects/ota_toolkit/IoT_Applications/node_device/rust/node/target/riscv32imc-esp-espidf/debug/deps/node-9aa38a7b29a9e2da.40nh4o3k97hhrj9k.rcgu.o" "/home/user/Projects/ota_toolkit/IoT_Applications/node_device/rust/node/target/riscv32imc-esp-espidf/debug/deps/node-9aa38a7b29a9e2da.42w3jcowb [... cut ...]
          /home/user/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: failed to merge target specific data of file esp-idf/esp_ringbuf/libesp_ringbuf.a(ringbuf.c.obj)
          /home/user/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: -march=: ISA string must begin with rv32 or rv64
```

Which I didn't encountered using the `nightly-2023-02-28` from the [Official Espressif Guide](https://esp-rs.github.io/std-training/02_2_software.html)

closed 2023-09-14 by jessebraham