esp-rs/espflash #463: `cargo espflash` fails when installed

persello, 2023-08-21:
If I install `cargo-espflash` with `cargo install cargo-espflash` and I run `cargo espflash`, this is what I get:

```
*** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '-[__NSCFString stringByStandardizingPath]: unrecognized selector sent to instance 0x11ee09730'
*** First throw call stack:
(
        0   CoreFoundation                      0x0000000186404960 __exceptionPreprocess + 176
        1   libobjc.A.dylib                     0x0000000185efdeb4 objc_exception_throw + 60
        2   CoreFoundation                      0x00000001864b646c -[NSObject(NSObject) __retain_OA] + 0
        3   CoreFoundation                      0x000000018636eb24 ___forwarding___ + 1572
        4   CoreFoundation                      0x000000018636e440 _CF_forwarding_prep_0 + 96
        5   Foundation                          0x000000018743fd80 -[NSProcessInfo arguments] + 188
        6   CoreFoundation                      0x0000000186480094 __getDefaultArguments_block_invoke + 96
        7   libdispatch.dylib                   0x0000000186105910 _dispatch_client_callout + 20
        8   libdispatch.dylib                   0x000000018610714c _dispatch_once_callout + 32
        9   CoreFoundation                      0x000000018647fa30 _addBackstopValuesForIdentifierAndSource + 640
        10  CoreFoundation                      0x000000018633a3b4 __81-[_CFXPreferences(SourceAdditions) withNamedVolatileSourceForIdentifier:perform:]_block_invoke + 144
        11  CoreFoundation                      0x000000018647f6d8 -[_CFXPreferences withNamedVolatileSourceForIdentifier:perform:] + 272
        12  CoreFoundation                      0x0000000186340724 -[CFPrefsSearchListSource addNamedVolatileSourceForIdentifier:] + 136
        13  CoreFoundation                      0x00000001864be94c __108-[_CFXPreferences(SearchListAdditions) withSearchListForIdentifier:container:cloudConfigurationURL:perform:]_block_invoke.155 + 296
        14  CoreFoundation                      0x00000001864be5f4 -[_CFXPreferences withSearchLists:] + 84
        15  CoreFoundation                      0x000000018633bc78 __108-[_CFXPreferences(SearchListAdditions) withSearchListForIdentifier:container:cloudConfigurationURL:perform:]_block_invoke + 300
        16  CoreFoundation                      0x00000001864be7a0 -[_CFXPreferences withSearchListForIdentifier:container:cloudConfigurationURL:perform:] + 384
        17  CoreFoundation                      0x000000018633b5a0 -[_CFXPreferences copyAppValueForKey:identifier:container:configurationURL:] + 156
        18  CoreFoundation                      0x000000018633b4c8 _CFPreferencesCopyAppValueWithContainerAndConfiguration + 112
        19  SystemConfiguration                 0x00000001870c8488 SCDynamicStoreCopyProxiesWithOptions + 180
        20  cargo-espflash                      0x0000000104f8f99c Curl_macos_init + 16
        21  cargo-espflash                      0x0000000104f8e26c curl_global_init + 176
        22  cargo-espflash                      0x0000000104f8bf98 _ZN3std4sync4once4Once9call_once28_$u7b$$u7b$closure$u7d$$u7d$17h59f7dc18648c0c99E + 36
        23  cargo-espflash                      0x000000010535e3d8 _ZN3std10sys_common4once5queue4Once4call17hedf6a27f629de9f0E + 1132
        24  cargo-espflash                      0x0000000104f8c120 _ZN4curl9INIT_CTOR9init_ctor17h5cfae87caa907fb3E + 88
        25  dyld                                0x0000000185f555c8 ___ZZNK5dyld46Loader25findAndRunAllInitializersERNS_12RuntimeStateEENK3$_0clEv_block_invoke + 168
        26  dyld                                0x0000000185f9a920 ___ZNK5dyld313MachOAnalyzer18forEachInitializerER11DiagnosticsRKNS0_15VMAddrConverterEU13block_pointerFvjEPKv_block_invoke.209 + 340
        27  dyld                                0x0000000185f8dc60 ___ZNK5dyld39MachOFile14forEachSectionEU13block_pointerFvRKNS0_11SectionInfoEbRbE_block_invoke + 496
        28  dyld                                0x0000000185f352fc _ZNK5dyld39MachOFile18forEachLoadCommandER11DiagnosticsU13block_pointerFvPK12load_commandRbE + 300
        29  dyld                                0x0000000185f8cc98 _ZNK5dyld39MachOFile14forEachSectionEU13block_pointerFvRKNS0_11SectionInfoEbRbE + 192
        30  dyld                                0x0000000185f9a434 _ZNK5dyld313MachOAnalyzer18forEachInitializerER11DiagnosticsRKNS0_15VMAddrConverterEU13block_pointerFvjEPKv + 516
        31  dyld                                0x0000000185f51798 _ZNK5dyld46Loader25findAndRunAllInitializersERNS_12RuntimeStateE + 448
        32  dyld                                0x0000000185f57b14 _ZNK5dyld416JustInTimeLoader15runInitializersERNS_12RuntimeStateE + 36
        33  dyld                                0x0000000185f51b4c _ZNK5dyld46Loader23runInitializersBottomUpERNS_12RuntimeStateERN5dyld35ArrayIPKS0_EE + 220
        34  dyld                                0x0000000185f55654 _ZZNK5dyld46Loader38runInitializersBottomUpPlusUpwardLinksERNS_12RuntimeStateEENK3$_1clEv + 112
        35  dyld                                0x0000000185f51ccc _ZNK5dyld46Loader38runInitializersBottomUpPlusUpwardLinksERNS_12RuntimeStateE + 304
        36  dyld                                0x0000000185f76ad4 _ZN5dyld44APIs25runAllInitializersForMainEv + 464
        37  dyld                                0x0000000185f39f34 _ZN5dyld4L7prepareERNS_4APIsEPKN5dyld313MachOAnalyzerE + 3192
        38  dyld                                0x0000000185f38f44 start + 1948
)
libc++abi: terminating due to uncaught exception of type NSException
zsh: abort      cargo espflash
```

If I run `espflash` only, I don't run into this kind of issue.

Building v2.0.1 of `cargo-espflash` with `cargo run --release --bin cargo-espflash -- espflash` produces an executable that doesn't display this issue.

I am using macOS 14.0 23A5312d on arm64.

persello, 2023-08-22:
Running `cargo install --path .` from the `cargo-espflash` folder produces the same issue.

The result is the following: the `cargo-espflash` binary located in `target/release` and the one located in `/Users/persello/.cargo/bin/cargo-espflash` are the same (checked with `diff` and `cmp`). The first one runs without issues, the second one crashes.

I think that the issue may come from the location of the resulting binary, but I don't know how to debug it.

jessebraham, 2023-08-22:
macOS 14 has not been released, so I'm not surprised you're experiencing issues. `cargo-espflash` installs just fine using macOS 13.5.1 on both x86_64 and arm64.

I need to be productive so I will not be installing an unstable and unreleased version of macOS, if you want to debug the issue and submit a fix please feel free, but I will not be updating my OS until there has been at least the official release and one subsequent patch release, sorry.

1cedsoda, 2023-09-14:
Sonoma was just released. Problem still there

jessebraham, 2023-09-14:
There is now an upstream issue tracking this (https://github.com/mitsuhiko/rye/issues/428), I will leave this issue open until that is resolved, but my hands are tied until there is a new release with a fix.

ryandesign, 2023-09-20:
> Sonoma was just released. Problem still there

macOS Sonoma has not been released. It will be released September 26, 2023.

> There is now an upstream issue tracking this ([mitsuhiko/rye#428](https://github.com/mitsuhiko/rye/issues/428))

Upstream for this issue is curl, where I've now filed a bug report: https://github.com/curl/curl/issues/11893

brainstorm, 2023-10-06:
Running on OSX Sonoma (non-beta) here, works fine for me except some minor issues reported in https://github.com/esp-rs/espflash/issues/436#issuecomment-1749949042

SomeoneToIgnore, 2023-10-09:
👋 
Upstream curl issue had been fixed, new `curl-sys` version had been released and seems to work on Sonoma.
See https://github.com/alexcrichton/curl-rust/issues/524#issuecomment-1753070751 for more details.

jessebraham, 2023-10-10:
I updated my M1 Mac Mini to Sonoma today, and running `cargo install cargo-espflash espflash` installed both applications successfully.