esp-rs/espflash #464: Unable to exit while stuck connecting to wrong port using Ctrl + C

laundmo, 2023-08-23:
Afte accidentally choosing the wrong COM port, espflash got stuck `Connecting...` and i was not able to exit it using Ctrl + C

SergioGasquez, 2024-01-25:
Could you provide some more details? I guess that you are using Windows, but which `espflash` version, which terminal ...

MabezDev, 2024-01-25:
I've also seen this happen, mainly on connect and when the chip doesn't respond. I have to wait for it to timeout even if a spam Ctrl+C - not very helpful I know, but another data point at least.

laundmo, 2024-01-25:
> Could you provide some more details? I guess that you are using Windows, but which `espflash` version, which terminal ...

sorry i really should have
teminal: powershell in vscode
version: sorry i dont recall but it should have been the latest release as of the original post. installed using `cargo install`.

SergioGasquez, 2024-01-25:
We are using `crossterm` to catch this:  
https://github.com/esp-rs/espflash/blob/main/espflash/src/cli/monitor/mod.rs#L115-L136

They have the most common terminals tested: https://github.com/crossterm-rs/crossterm#tested-terminals. Not sure what we could do beside opening an issue upstream or switching the method we use

SergioGasquez, 2024-01-25:
> sorry i really should have teminal: powershell in vscode version: sorry i dont recall but it should have been the latest release as of the original post. installed using `cargo install`.

Could you try using `Console Host` as its a tested terminal from `crosster` and see if the issue is also present?

laundmo, 2024-01-25:
sorry, i don't have any device with me i could test with in the next week. maybe @MabezDev could?

closed 2025-02-19 by jessebraham