esp-rs/espflash #472: Esp8266 baud rate ignored with Flasher::connect

loucass003, 2023-09-14:
I am trying to flash an esp8266 board with a speed of 921600

but the flasher baud is ignored in the connect function

```rs
// Now that we have established a connection and detected the chip and flash
        // size, we can set the baud rate of the connection to the configured value.
        if let Some(baud) = speed {
            match flasher.chip {
                Chip::Esp8266 => (), // Not available
                _ => {
                    if baud > 115_200 {
                        warn!("Setting baud rate higher than 115,200 can cause issues");
                        flasher.change_baud(baud)?;
                    }
                }
            }
        }
   ```
   
this does not seem right as I am in fact able to call the function after using connect and the flashing does complete without issue. 
   
 ```rs
    let mut flasher = Flasher::connect(interface, info.clone(), Some(921600), true).expect("unable to connect");
    flasher.change_baud(921600).unwrap();
 ``` 
 
 is there a particular reason on why the esp8266 is ignored and not simply prompting the warning like the others?

jessebraham, 2023-09-15:
I don't know that there's any specific reason for this. Seems to have been this way [since 2020 when it was added](https://github.com/esp-rs/espflash/commit/cd4ded39d75cc59f35a3bcea0c9c7e6561edf0bf).

Looking at `esptool` it doesn't look like they support setting the baud rate for the ESP8266 either, unless I've missed something. It's been so long since I've used that chip I frankly don't remember much about it, sorry.

loucass003, 2023-09-15:
from esptool code
https://github.com/espressif/esptool/blob/master/esptool/loader.py#L1084-L1085
could it be that the STUB loader allows for the baud rate to be changed ?

loucass003, 2023-09-17:
Ok so after investigating a bit more on this, it looks like you can set the baudrate higher than 115200 after loading the stub. 

So initially you shoud:

- Open the connection at 115200 
- Upload the stub if supported
- after the stub is loaded you can set the baudrate higher like 921600

jessebraham, 2024-02-05:
We've decided to drop support for the ESP8266 (see #576), so closing this issue.

closed 2024-02-05 by jessebraham