esp-rs/espflash #479: Improve flash speed by skipping bootloader and partition tables

MabezDev, 2023-09-20:
See: https://github.com/esp-rs/esp-hal/issues/803#issuecomment-1727958673

But essentially, we could make espflash smart enough to flash the things that have changed, which is usually just the main app.

bugadani, 2023-09-20:
If it is possible to read the hash of a particular flash region, it should also be possible I believe to implement selective reflashing: for each flash block, read the hash, compare, and only reflash if differs.

Also, cc https://github.com/esp-rs/espflash/issues/259 as this issue is basically the meat of flash content verification.

bjoernQ, 2023-12-21:
There is  ` ESP_FLASH_VERIFY_MD5` command ( https://github.com/espressif/esptool/blob/8538b8983e5e04ad5d6f63620d609bfa63412eb3/flasher_stub/include/stub_flasher.h#L53 )

I think it's not yet supported by the Rust esp-flasherstub but by the esp-tool stub and in ROM (not for ESP-8266)

bjoernQ, 2023-12-21:
Started to look into `ESP_FLASH_VERIFY_MD5` ... as mentioned, we don't have it supported by the Rust esp-flasherstub which is not that good

Tried without a flasher stub and it needs quite a few changes to espflash internals because of the way we currently parse responses into `CommandResponse` (and how we currently cannot return something that is not u32)

Additionally, there are at least two different response formats for that command

PR #538 (2023-12-28): Implement skip and verify
Implements verify, closes #259 
Implements skip flashing of unchanged flash regions, closes #479
closed 2023-12-28 by SergioGasquez