esp-rs/espflash #509: Improve tests

SergioGasquez, 2023-11-15:
```[tasklist]
### Tasks
- [ ] https://github.com/esp-rs/espflash/issues/439
- [ ] Add integration tests for subcommands like `elf2image`, `merge_bin`, `save_image`...
- [ ] Add more unit tests if possible
```

jessebraham, 2024-10-09:
@SergioGasquez do you have any intention of working on this in the future? I think it is a fine idea, but given our limited resources and the fact we have not been focusing on this project for some time now, perhaps we can just close this?

SergioGasquez, 2024-10-09:
> do you have any intention of working on this in the future?

I don't have any plans to work on this, unless we change our priorities, as you mentioned.

jessebraham, 2024-10-09:
Okay thanks, will close this for now. We can revisit it in the future if needed.