esp-rs/espflash #511: Another flash size question

enelson1001, 2023-11-17:
When I flash my M5Stack I get the following output.
```
ed@ed-ideacentre-AIO-910-27ISH:~/esp-rust-projects/m5stack/rust-lvgl-hello$ cargo espflash flash --monitor
[2023-11-17T01:45:07Z INFO ] 🚀 A new version of cargo-espflash is available: v2.1.0
[2023-11-17T01:45:07Z INFO ] Detected 2 serial ports
[2023-11-17T01:45:07Z INFO ] Ports which match a known common dev board are highlighted
[2023-11-17T01:45:07Z INFO ] Please select a port
[2023-11-17T01:45:08Z INFO ] Serial port: '/dev/ttyUSB0'
[2023-11-17T01:45:08Z INFO ] Connecting...
   Compiling lvgl v0.6.2 (/home/ed/lv-binding-rust/lv_binding_rust/lvgl)
   Compiling rust-lvgl-hello v0.1.0 (/home/ed/esp-rust-projects/m5stack/rust-lvgl-hello)
    Finished dev [optimized + debuginfo] target(s) in 5.22s

Chip type:         esp32 (revision 3)
Crystal frequency: 40MHz
Flash size:        16M
Features:          WiFi, BT, Dual Core, 240MHz, Coding Scheme None
MAC address:       3c:61:05:0b:e3:fc
App/part. size:    755,776/1,048,576 bytes, 72.08%
[00:00:02] [========================================]      17/17      segment 0x1000                                                                                       
[00:00:00] [========================================]       1/1       segment 0x8000                                                                                       
[00:00:45] [========================================]     420/420     segment 0x10000                                                                                      [2023-11-17T01:46:02Z INFO ] Flashing has completed!
```

After the ```MAC address:       3c:61:05:0b:e3:fc``` it shows ```App/part. size:    755,776/1,048,576 bytes, 72.08%```

What is this line telling me?  

I'm confused because it also says I have ```Flash size:        16M```

Vollbrecht, 2023-11-17:
partition size  =/= flash size like on your computer

enelson1001, 2023-11-17:
Sorry about addition questions.
1. It looks like it is using the default partition.  Is that correct?
```
# ESP-IDF Partition Table
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x6000,
phy_init, data, phy,     0xf000,  0x1000,
factory,  app,  factory, 0x10000, 1M,
```
2. So is it telling me that the factory/app partition size is 1M and the program is taking up 755, 766 bytes or 72.08% of that partition.  Correct?
3. So how much should I increase the factory/ app partition size?
4. How do I do that?  Is there some config file I need to add to my rust project?

Vollbrecht, 2023-11-17:
you can use whatever you want from your flash to be used. There is a command in `espflash flash --flash-size=16MB` that you can use to get the correct size calculated, as espflash doesn't know what type of flash is build onto your board. espflash assumes a default partition table if you don't give it one explicitly

enelson1001, 2023-11-17:
When I added the flash-size to the commend line it still reports App/part size the same as before.
```
ed@ed-ideacentre-AIO-910-27ISH:~/esp-rust-projects/m5stack/rust-lvgl-hello$ cargo espflash flash --flash-size=16M --monitor
[2023-11-17T14:43:00Z INFO ] 🚀 A new version of cargo-espflash is available: v2.1.0
[2023-11-17T14:43:00Z INFO ] Detected 2 serial ports
[2023-11-17T14:43:00Z INFO ] Ports which match a known common dev board are highlighted
[2023-11-17T14:43:00Z INFO ] Please select a port
[2023-11-17T14:43:02Z INFO ] Serial port: '/dev/ttyUSB0'
[2023-11-17T14:43:02Z INFO ] Connecting...
    Finished dev [optimized + debuginfo] target(s) in 5.12s

Chip type:         esp32 (revision 3)
Crystal frequency: 40MHz
Flash size:        16M
Features:          WiFi, BT, Dual Core, 240MHz, Coding Scheme None
MAC address:       3c:61:05:0b:e3:fc
App/part. size:    755,776/1,048,576 bytes, 72.08%
[00:00:02] [========================================]      17/17      segment 0x1000                                                                                       
[00:00:00] [========================================]       1/1       segment 0x8000                                                                                       
[00:00:45] [========================================]     420/420     segment 0x10000                                                                                      [2023-11-17T14:43:55Z INFO ] Flashing has completed!
```
Is this a bug?

Vollbrecht, 2023-11-17:
no, just telling it that you are having a larger flash-size doesn't change the fact that it is using the default-partition table.

enelson1001, 2023-11-17:
For those people looking for answers.
I created a folder under my project called partition-table and created a file partitions.csv.
```
esp-rust-projects/m5stack/rust-lvgl-hello
├── build.rs
├── Cargo.lock
├── Cargo.toml
├── custom_fonts
│   ├── gotham_bold_80.c
│   └── gotham_bold_80.h
├── logs
│   ├── log_esp_idf_hal_objdump.txt
│   ├── log_lvgl_objdump.txt
│   ├── log.txt
│   └── objdump.txt
├── lvgl-configs
│   ├── lv_conf.h
│   ├── lv_drv_conf.h
│   └── lv_drv_conf old.h
├── partition-table
│   └── partitions.csv
├── rust-toolchain.toml
├── sdkconfig.defaults
├── src
│   └── main.rs
```
My partitions.csv flie contents
```
# http://esp-idf.readthedocs.io/en/latest/api-guides/partition-tables.html
# This file uses a 3M factory instead of the default 1M

# Espressif ESP32 Partition Table
# Name,       Type, SubType, Offset,  Size
nvs,          data, nvs,     0x9000,  0x6000
phy_init,     data, phy,     0xf000,  0x1000
factory,      app,  factory, 0x10000, 3M
# The size 528k isn't arbitrary - it is the minumim size when
# wear leveling sector size is 4k
app_storage,  data, fat,     ,        528k
```
And then the espflash command
```
$ cargo espflash flash --partition-table=partition-table/partitions.csv --monitor
```
Produces the following
```
Chip type:         esp32 (revision 3)
Crystal frequency: 40MHz
Flash size:        16M
Features:          WiFi, BT, Dual Core, 240MHz, Coding Scheme None
MAC address:       3c:61:05:0b:e3:fc
App/part. size:    755,776/3,145,728 bytes, 24.03%
[00:00:02] [========================================]      17/17      segment 0x1000                                                                                       
[00:00:00] [========================================]       1/1       segment 0x8000                                                                                       
[00:00:45] [========================================]     420/420     segment 0x10000                                                                                      

```
Which shows I have a app/part of 3M

closed 2023-11-17 by enelson1001