esp-rs/espflash #520: Add verbose defmt output like probe-rs

dzamlo, 2023-11-23:
When using probe-rs with `defmt` We get timing and source info for each print. For example 
```
0.000061 INFO  Hello World!
└─ blinky::____embassy_main_task::{async_fn#0} @ src/bin/blinky.rs:14
```

For coloring, only the level word (INFO, DEBUG, etc.) is colored. I found it more readable than coloring the whole line.

It would be nice to be able to get the same with `espflash`.

I think it should not be the default but should be behind a cli flag (`-L defmt-verbose` for example).

bugadani, 2023-11-23:
I intentionally didn't do this for subjective reasons but I completely agree that it should be available as a `-L` option.

SergioGasquez, 2023-11-28:
I've been working a bit on this feature, find changes in [my branch](https://github.com/SergioGasquez/espflash/tree/feat/defmt-verbose). Currently, when using `-L defmt-verbose`:
```
[WARN] - Hello world!
└─ /home/sergio/Documents/Espressif/tests/defmt/src/main.rs:16
[INFO] - Loop...
└─ /home/sergio/Documents/Espressif/tests/defmt/src/main.rs:18
[INFO] - Loop...
└─ /home/sergio/Documents/Espressif/tests/defmt/src/main.rs:18
[INFO] - Loop...
└─ /home/sergio/Documents/Espressif/tests/defmt/src/main.rs:18
```

- The color is now only applied to the log level, as you suggested for both `defmt` and `defmt-verbose`
- Works with UART and RTT.
- Timestamp and code_location are not yet implemented
  - Is this even possible at the moment? @bugadani

bugadani, 2023-11-28:
> as you suggested for both defmt

oh but I like my fully colored log lines, they make it easier for me to read 😱

> Is this even possible at the moment?

I don't know, I didn't look into it.

SergioGasquez, 2023-11-28:
> oh but I like my fully colored log lines, they make it easier for me to read 😱 ( I probably got used to the monitor in esp-idf coloring everything but whatever :) )

TBH, I don't have a hard opinion on this, I may slightly prefer to have only the log level colored but happy to revert the change.

> I don't know, I didn't look into it.

I'll take a deeper look at https://defmt.ferrous-systems.com/timestamps/https://defmt.ferrous-systems.com/design-timestamp and https://github.com/probe-rs/probe-rs/blob/master/probe-rs/src/bin/probe-rs/util/rtt.rs#L21-L57 (although our RTT implantation is being reevaluated)

bugadani, 2023-11-28:
> TBH, I don't have a hard opinion on this, I may slightly prefer to have only the log level colored but happy to revert the change.

There's probably no way to make everyone happy but IIRC the original behavior, i.e. when espflash was displaying colored esp-idf logs was to color everything. Deviating from that may mean inconsistent output based on what's being logged.

My intuition is that printing timestamps shouldn't be very hard and we're just not calling `display_timestamp` to add it to the output. I'm hoping there shouldn't be any changes necessary to any of the embedded code here, though if the question came up maybe things don't work as I imagine them?

Vollbrecht, 2023-11-28:
i also still have an eye on the issue. i think we should go with a different approach here - we should provide 2-3 default sane options here, and then also provide the following two things
  1. a flag that allows for a custom formater provided by the user like its possible [here](https://docs.rs/defmt-decoder/latest/src/defmt_decoder/log/mod.rs.html#132-154)
  2. a log level provider so a user can filter out without needing to reflash

Vollbrecht, 2023-11-28:
So what i am saying is we maybe shouldn't duplicate the work here and just work with the upstream defmt_decoder::log and potential fix/add things there?

Vollbrecht, 2023-11-28:
yes i understand this, and probably the simplest way would be to only have this features for the defmt path inside the handle_defmt and not touch defmt_raw.

SergioGasquez, 2023-11-28:
> My intuition is that printing timestamps shouldn't be very hard and we're just not calling `display_timestamp` to add it to the output. 

I gave it a try, but it was returning `None` not sure if I was doing something wrong.

bugadani, 2023-11-28:
> I gave it a try, but it was returning None not sure if I was doing something wrong.

I think it's because we don't output timestamps by default in esp-println/esp-hal. @Vollbrecht's esp-idf implementation does. If defmt allows distinguishing between the two, I'm pretty happy that we don't need to print a bunch of zeroes or require a timestamp for esp-hal based apps.

bugadani, 2023-11-28:
> if you want this feature in esp-println you just need to implement the defmt timestamp! macro in the same place where the global_logger is impl - for example providing it a way to access the systimer. The rest then should work automagically
> 

IMO we should leave the option to the users. I don't find timestamps generally useful except for some niche cases.

bugadani, 2023-11-28:
> yes but the problem is defmt really hates it when there are multiple potential places where it is implemented, so you probably should implement it in the highest crate in the dependancy chain possible, and gate it behind a feature flag
> 

It has to work in the user's firmware. Esp-println doesn't depend on esp-hal and reading a timestamp would need more code to be duplicated than what could be called elegant I think.

PR #818 (2025-03-25): Option to supply custom defmt formats
The formatting options are identical to probe-rs's, except in espflash we can't control timestamp/location with switches. The user can supply a custom format string if such control is necessary.

Closes #520 
Closes #781