esp-rs/espflash #522: merge_adjacent_segements fails if there is a gap between .text and .text_init segments

t-moe, 2023-11-23:
See https://github.com/rust-lang/rust/issues/118214 .

I'm trying to figure out, if rustc is allowed to do this (insert a gap at will), and we should handle it, or if it is a rustc bug.

t-moe, 2023-11-27:
After some discussion in the linked rustc issue, I think we probably have to fix it on our side.

(https://github.com/rust-lang/rust/issues/118214#issuecomment-1827351764)

t-moe, 2023-11-27:
@Vollbrecht The issue I'm having is that sometimes the .text segment is 16 byte(!!) aligned. When data is 16 byte aligned, it is also 16bit aligned :).  Individual functions (like semihosting::process::abort) request a such large alignment and rustc or llvm will then align the entire text segment on a 16-byte boundary. We need to handle that case accordingly, and merge adjacent flash segments, even if they have some alignment padding between them.

Vollbrecht, 2023-11-27:
reading comprehension failed :D yeah i only read byte as bit's everywhere, thanks for clearing that up.

jessebraham, 2023-12-04:
Sorry, work did not go as planned at all last week and I'm still quite busy with other tasks. If somebody else is able to investigate this that'd be very much appreciated, otherwise I probably can't get to it until the new year unfortunately.

bjoernQ, 2023-12-18:
when fixing it we need to keep this in mind: https://github.com/esp-rs/esp-hal/blob/260c882b088c416c457e0b3337d849338a0a56e2/esp-hal-common/ld/esp32c6/esp32c6.x#L72-L82

bjoernQ, 2023-12-18:
This is a minimal repro of the issue:

```rust
#![no_std]
#![no_main]

use esp_backtrace as _;
use esp32c3_hal::prelude::entry;

#[entry]
fn main() -> ! {
    unsafe {
        foo();
    }
    loop {}
}

#[inline]
pub(crate) unsafe fn foo() {
    unsafe {
        core::arch::asm!(
            ".balign 16",
            ".option push",
            ".option norvc",
            "nop",
            ".option pop",
        );
    }
}
```

We could also solve this by just moving
```
    KEEP(*(.init));
    KEEP(*(.init.rust));
    KEEP(*(.text.abort));
```

into `.text` (needs some adjustments where `.text_init` is mentioned) - or are there any downsides doing this?

MabezDev, 2023-12-18:
> or are there any downsides doing this

We did do it for a reason, but I can't find the reason nor remember it :D. Let's try it and see what happens.

bjoernQ, 2023-12-18:
https://github.com/esp-rs/esp-hal/pull/1038 should solve the problem (in a different way) - feel free to re-open if there are still issues

closed 2023-12-18 by bjoernQ

t-moe, 2024-01-04:
Is it possible that we have the same problem, when using esp_idf_hal ? @bjoernQ 

See the gap here:
```
# riscv32-esp-elf-objdump -h  target/riscv32imac-esp-espidf/debug/esp32_devkit

target/riscv32imac-esp-espidf/debug/esp32_devkit:     file format elf32-littleriscv

Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  .....
  5 .iram0.text   00009f66  40800000  40800000  00001000  2**8
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  6 .iram0.text_end 00000000  40809f66  40809f66  0007f7e0  2**0
                  CONTENTS
```

Not sure if this is a problem or not, but I have a binary here, that does not boot up as expected...

MabezDev, 2024-01-04:
I don't think it is the same issue, because even though they have different section alignments, the two sections are contiguous where as before there was a gap which caused the error. I suggest filling a separate issue.

t-moe, 2024-01-04:
Thank you @MabezDev. You're indeed right, there is no gap here. case closed.

(I've added a diagnostic that warns me of gaps, but it seems the diagnostic is wrong. And i posted the output without checking.... sorry :face_with_head_bandage:  )