esp-rs/espflash #526: ESP32-C2 26MHz logs don't work properly

JurajSadel, 2023-12-06:
Log from `embassy_hello_world"

```rust
Chip type:         esp32c2 (revision v1.0)
Crystal frequency: 26MHz
Flash size:        4MB
Features:          WiFi, BLE
MAC address:       08:3a:8d:48:7d:6c
App/part. size:    89,760/4,128,768 bytes, 2.17%
[00:00:01] [========================================]      11/11      0x0
[00:00:00] [========================================]       1/1       0x8000
[00:00:02] [========================================]      22/22      0x10000
[2023-12-06T10:30:24Z INFO ] Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

r�bl~�lr��
�nnr�
��
l��l
�nn�l�n�쎌
��l`�r�b

b�l����lllp��nn��
```

MabezDev, 2023-12-06:
I don't have a 26mhz board, but to be sure, you're passing the `26mhz_xtal` feature too?

SergioGasquez, 2023-12-08:
Id suggest building a bootloader in a esp-idf project and using it,see https://esp-rs.github.io/book/troubleshooting/espflash.html#rtc_clk_init-possibly-invalid-config_xtal_freq-setting.

I don't have a 26 MHz board but I can elaborate more on Monday if needed.

beeb, 2023-12-09:
Thanks @SergioGasquez , I had not found that resource in the book. Unfortunately, I have no `esp-idf-sys-*` folder in my `target/xtensa-esp32-none-elf/release/build` folder. Can't find any bootloader binary in fact...

beeb, 2023-12-09:
I'm trying to compile the project generated by `esp-idf-template` but it complains that `ERROR: C:\tb\.embuild\espressif\espidf.constraints.v5.1.txt doesn't exist. Perhaps you've forgotten to run the install scripts. Please check the installation guide for more information.`

But I don't see any install script. I used `espup` to install everything.

Vollbrecht, 2023-12-09:
we are currently revisiting how the bootloader is exposed in esp-idf-sys. This is already done on gh but not released. Currently you can find the bootloader in the build artifacts of esp-idf-sys. It might look like the following inside your target dir 
`target/riscv32imc-<or-your-target>/debug/build/esp-idf-sys-<random-number>/out/build/bootloader`

beeb, 2023-12-09:
Cool thanks, so I am now able to build and flash my project by specifying the bootloader that was built in the idf template project.

SergioGasquez, 2023-12-11:
Since this issue has appeared a few times, we could try to include a 26 MHz bootloader for ESP32 and ESP32-C2 and use it if we detect a 26 MHZ freq when flashing

closed 2024-01-24 by MabezDev