esp-rs/espflash #534: Arduino Nano ESP32-S3 Not connecting

SnappierSoap318, 2023-12-25:
after `cargo run`

```
> cargo run
    Finished dev [unoptimized + debuginfo] target(s) in 0.09s
     Running `espflash flash --monitor target\xtensa-esp32s3-none-elf\debug\analog-keeb`
[2023-12-25T09:45:03Z INFO ] Serial port: 'COM4'
[2023-12-25T09:45:03Z INFO ] Connecting...
[2023-12-25T09:45:04Z INFO ] Unable to connect, retrying with extra delay...
[2023-12-25T09:45:05Z INFO ] Unable to connect, retrying with default delay...
[2023-12-25T09:45:06Z INFO ] Unable to connect, retrying with extra delay...
[2023-12-25T09:45:07Z INFO ] Unable to connect, retrying with default delay...
[2023-12-25T09:45:08Z INFO ] Unable to connect, retrying with extra delay...
[2023-12-25T09:45:09Z INFO ] Unable to connect, retrying with default delay...
[2023-12-25T09:45:10Z INFO ] Unable to connect, retrying with extra delay...
Error: espflash::connection_failed

  × Error while connecting to device
  ╰─▶ Failed to connect to the device
  help: Ensure that the device is connected and the reset and boot pins are not being held down

error: process didn't exit successfully: `espflash flash --monitor target\xtensa-esp32s3-none-elf\debug\analog-keeb` (exit code: 1)
```
Trying to get Board info 
```
 > espflash board-info
[2023-12-25T09:41:40Z INFO ] Serial port: 'COM4'
[2023-12-25T09:41:40Z INFO ] Connecting...
[2023-12-25T09:41:41Z INFO ] Unable to connect, retrying with extra delay...
[2023-12-25T09:41:42Z INFO ] Unable to connect, retrying with default delay...
[2023-12-25T09:41:43Z INFO ] Unable to connect, retrying with extra delay...
[2023-12-25T09:41:44Z INFO ] Unable to connect, retrying with default delay...
[2023-12-25T09:41:44Z INFO ] Unable to connect, retrying with extra delay...
[2023-12-25T09:41:46Z INFO ] Unable to connect, retrying with default delay...
[2023-12-25T09:41:46Z INFO ] Unable to connect, retrying with extra delay...
Error: espflash::connection_failed

  × Error while connecting to device
  ╰─▶ Failed to connect to the device
  help: Ensure that the device is connected and the reset and boot pins are not being held down
```
OS: Windows 10 22H2 19045.3803

Can program the device using Arduino's IDE 2.0.

Board info using Arduino's IDE:
[image]

SergioGasquez, 2023-12-25:
Hi! Which `espflash` version are you using? Did you tried using `main`: `cargo install espflash --git https://github.com/esp-rs/espflash`

SnappierSoap318, 2023-12-26:
Current `espflash` version
```
> espflash.exe --version
espflash 2.1.0
```
after `cargo install espflash --git https://github.com/esp-rs/espflash`
```
> espflash.exe --version
espflash 3.0.0-dev
```
running `board-info`
```
> espflash.exe board-info
[2023-12-26T11:10:00Z INFO ] Serial port: 'COM5'
[2023-12-26T11:10:00Z INFO ] Connecting...
Error: espflash::connection_failed

  × Failed to open serial port COM5
  ├─▶ Failed to open serial port COM5
  ├─▶ Error while connecting to device
  ╰─▶ Serial port not found
  help: Ensure that the device is connected and your host recognizes the serial adapter
```

SergioGasquez, 2023-12-28:
It might be related with #532. Have you played with drivers at any point using Zadig or something similar?

SnappierSoap318, 2023-12-28:
Nope, i havent played with any drivers, just using the ones that came with arduino's IDE and I am using UART

SnappierSoap318, 2023-12-28:
After the flash has completed and running cargo run again, it now works 

```
cargo.exe run --package analog-keeb --bin analog-keeb 

warning: unused manifest key: toolchain
    Finished dev [unoptimized + debuginfo] target(s) in 0.10s
     Running `espflash flash --monitor target\xtensa-esp32s3-none-elf\debug\analog-keeb`
[2023-12-28T13:13:37Z INFO ] Serial port: 'COM3'
[2023-12-28T13:13:37Z INFO ] Connecting...
[2023-12-28T13:13:37Z INFO ] Using flash stub
Chip type:         esp32s3 (revision v0.1)
Crystal frequency: 40MHz
Flash size:        16MB
Features:          WiFi, BLE
MAC address:       34:85:18:7a:ad:14
App/part. size:    189,504/16,384,000 bytes, 1.16%
[00:00:00] [========================================]      14/14      0x0
[00:00:00] [========================================]       1/1       0x8000
[00:00:01] [========================================]      72/72      0x10000
[2023-12-28T13:13:40Z INFO ] Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit      

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021       
rst:0x15 (USB_UART_CHIP_RESET),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403789d0
0x403789d0 - core::num::<impl u128>::trailing_zeros
    at ??:??
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fce3818,len:0x16f8
0x3fce3818 - _ESP_HAL_DEVICE_PERIPHERALS
    at ??:??
load:0x403c9700,len:0x4
0x403c9700 - __user_exception
    at ??:??
load:0x403c9704,len:0xc00
0x403c9704 - __user_exception
    at ??:??
load:0x403cc700,len:0x2eb0
0x403cc700 - __user_exception
    at ??:??
SHA-256 comparison failed:
Calculated: 9e7b8c8d53c0514c12b6eb2f4d48f985182d9c70a488ad4caf71e08902c97372
Expected: 3499e4149a363c2b1ee4ee9709ca5031f50ee6b595a532f697c8659516763ae3
Attempting to boot anyway...
entry 0x403c9908
0x403c9908 - __user_exception
    at ??:??
I (46) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (47) boot: compile time Jun  7 2023 08:07:32
I (47) boot: Multicore bootloader
I (51) boot: chip revision: v0.1
I (55) boot.esp32s3: Boot SPI Speed : 40MHz
I (60) boot.esp32s3: SPI Mode       : DIO
I (65) boot.esp32s3: SPI Flash Size : 16MB
I (70) boot: Enabling RNG early entropy source...
I (75) boot: Partition Table:
I (79) boot: ## Label            Usage          Type ST Offset   Length
I (86) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (93) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (101) boot:  2 factory          factory app      00 00 00010000 00fa0000
I (108) boot: End of partition table
I (113) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=069b4h ( 27060) map
I (128) esp_image: segment 1: paddr=000169dc vaddr=3fc89888 size=00004h (     4) load
I (130) esp_image: segment 2: paddr=000169e8 vaddr=40378000 size=01888h (  6280) load
I (140) esp_image: segment 3: paddr=00018278 vaddr=00000000 size=07da0h ( 32160) 
I (154) esp_image: segment 4: paddr=00020020 vaddr=42000020 size=1e3f4h (123892) map
I (186) boot: Loaded app from partition at offset 0x10000
I (187) boot: Disabling RNG early entropy source...
Hello world!
Loop...
Loop...
Loop...
Loop...
Loop...
```

SergioGasquez, 2023-12-28:
Ahh, great to hear that it was the strapping pins! Not sure why `esptool` (via Arduino IDE) was working and `espflash` wasn't, probably some difference in the resets. Thanks for reporting the solution!

closed 2023-12-28 by SergioGasquez

undecidedapollo, 2024-08-24:
Adding this as it might be helpful for someone else. I was stuck on this problem myself for a few hours. To resolve it I followed the steps above of connecting b1 to ground (next to each other, used tweezers) and pressed the reset button. After this it showed up as `USB JTAG/serial debug unit` but was stuck on `waiting for download`. Then I just unplugged the board, replugged it in, and things started working. Not sure why doing this resolved it but both boards I tried it on this resolved it. After doing this once it doesn't seem like you have to do it again.

Skywalker8510, 2025-08-02:
> After shorting the B1 (boot0) pin to ground, i am able to get board info
> 
> ```
> > espflash.exe board-info 
> [2023-12-28T13:10:11Z INFO ] Serial port: 'COM3'
> [2023-12-28T13:10:11Z INFO ] Connecting...
> [2023-12-28T13:10:11Z INFO ] Using flash stub
> Chip type:         esp32s3 (revision v0.1)
> Crystal frequency: 40MHz
> Flash size:        16MB
> Features:          WiFi, BLE
> MAC address:       34:85:18:7a:ad:14
> ```
> 
> edit: after flashing using the B1 pin shorted to ground,
> 
> ```
> cargo.exe run --package analog-keeb --bin analog-keeb 
> 
> warning: unused manifest key: toolchain
>     Finished dev [unoptimized + debuginfo] target(s) in 0.79s
>      Running `espflash flash --monitor target\xtensa-esp32s3-none-elf\debug\analog-keeb`
> [2023-12-28T13:11:16Z INFO ] Serial port: 'COM3'
> [2023-12-28T13:11:16Z INFO ] Connecting...
> [2023-12-28T13:11:16Z INFO ] Using flash stub
> Chip type:         esp32s3 (revision v0.1)
> Crystal frequency: 40MHz
> Flash size:        16MB
> Features:          WiFi, BLE        
> MAC address:       34:85:18:7a:ad:14
> App/part. size:    189,504/16,384,000 bytes, 1.16%
> [00:00:00] [========================================]      14/14      0x0
> [00:00:00] [========================================]       1/1       0x8000
> [00:00:01] [========================================]      72/72      0x10000
> [2023-12-28T13:11:19Z INFO ] Flashing has completed!
> Commands:
>     CTRL+R    Reset chip
>     CTRL+C    Exit      
> 
> ESP-ROM:esp32s3-20210327
> Build:Mar 27 2021
> rst:0x15 (USB_UART_CHIP_RESET),boot:0x0 (DOWNLOAD(USB/UART0))
> Saved PC:0x4037a314
> 0x4037a314 - __user_exception
>     at ??:??
> waiting for download
> ```
> 
> it gets stuck at waiting for download

Can confirm this fixed it for me as well. Mine wasn’t flashing on arduino ide either before this but now works both ways. How does one find out you have to perform this ritual to flash.