esp-rs/espflash #539: Support reading flash

t-moe, 2024-01-03:
It would be nice if espflash could read back the flash into a file. 
Currently I have to rely on esptool.py to read back flash segments... Probe-rs is also not able to do this.

t-moe, 2024-01-22:
Surprising! I didn't anticipate the implementation (requested by me) to happen so quickly. Big thanks @SergioGasquez ! Everything is functioning seamlessly. 🎉

SergioGasquez, 2024-01-23:
> Surprising! I didn't anticipate the implementation (requested by me) to happen so quickly. Big thanks @SergioGasquez ! Everything is functioning seamlessly. 🎉
> 
> ((Yes, I still had to manually put the device into bootloader mode, because of [esp-rs/esp-flasher-stub#45](https://github.com/esp-rs/esp-flasher-stub/issues/45) ))

Thanks for the testing! I'll try to investigate the bootloader issue in the upcoming days

closed 2024-02-01 by MabezDev