esp-rs/espflash #541: Help provides insufficient documentation for command line arguments

spectrachrome, 2024-01-06:
After a few hours of search, I finally found out how the partition table is actually applied onto the ESP32 device, but only because of the small section in the project README where I almost accidentally stumbled upon the `--partition-table` CLI argument.

Please make sure to include all of the possible command-line arguments in the help output of the CLI application.

SergioGasquez, 2024-01-08:
Hi! Do you mind-expanding which part is not covered in the CLI help? If you use `espflash flash --help`, you get:
```
      --partition-table <FILE>
          Path to a CSV file containing partition table
```

spectrachrome, 2024-01-09:
Okay. I was now able to have it show the CLI argument documentation with `cargo-espflash espflash flash --help`. Thanks for pointing this out. Yes, in that sense the documentation is covered, but really hard to find.

SergioGasquez, 2024-01-09:
Do you have any suggestion on how to make it more discoverable? I'd say this is the standard way of using the `--help`

spectrachrome, 2024-01-09:
Okay. I was now able to have it show the CLI argument documentation with `espflash flash --help` or `cargo-espflash espflash flash --help`. Thanks for pointing this out. Yes, in that sense the documentation is covered.

Maybe a small pointer in the root help, pointing out the dedicated help sections would be something. Anyhow, thank you for your time and I'm sorry for not understanding how your CLI works 😄

closed 2024-01-09 by SergioGasquez