esp-rs/espflash #546: [ESP32C3] [USB_SERIAL_JTAG] [stub] Error while connecting to device

TYPEmber, 2024-01-12:
Same situation #352  with ESP32C3 on M1 Mac
espflash's version is 2.1.0

SergioGasquez, 2024-01-14:
Have you tried with current `3.0.0-dev` version from `main`?
```
cargo install espflash --git https://github.com/esp-rs/espflash
```

TYPEmber, 2024-01-15:
No, but after clone & install esp-idf c sdk, this problem no longger occurs with same version(2.1.0) espflash.

SergioGasquez, 2024-01-15:
That is very weird, connecting errors should not depend on that, it should not be related to the app

MabezDev, 2024-01-15:
For completeness, does it work when passing the `--no-stub` option?

TYPEmber, 2024-01-16:
Yes, before I worked with esp-idf, I can flash esp32c3 with `--no-stub`.
Further more, after install esp-idf c sdk, without `--no-stub` still works well. Besides, in this situation, the elf file I flashed is generated directly by esp-idf c sdk.

SergioGasquez, 2024-01-23:
I'll close the issue as it seems to be resolved, feel free to reopen it in the future if it reappears.

closed 2024-01-23 by SergioGasquez