esp-rs/espflash #552: Large .text segments are merged incorrectly?

t-moe, 2024-01-17:
I'm using  a esp32c6 here with 8MB of flash and I have a rust program that is around 2 Mb, based on the `esp-idf-*` crates.

My partition table:

```
# ESP-IDF Partition Table
# Name,    Type, SubType, Offset,  Size, Flags
nvs,       data, nvs,     0x9000,  0x6000,
phy_init,  data, phy,     0xf000,  0x1000,
factory,   app,  factory, 0x10000, 4M,
```

<details>
<summary> Output of  riscv32-esp-elf-objdump -h 
</summary>

```

target/riscv32imac-esp-espidf/debug/esp32_devkit:     file format elf32-littleriscv

Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  0 .rtc.text     00000000  50000000  50000000  0020a560  2**0
                  CONTENTS
  1 .rtc.force_fast 00000000  50000000  50000000  0020a560  2**0
                  CONTENTS
  2 .rtc_noinit   00000000  50000000  50000000  0020a560  2**0
                  CONTENTS
  3 .rtc.force_slow 00000000  50000000  50000000  0020a560  2**0
                  CONTENTS
  4 .rtc_reserved 00000018  50003fe8  50003fe8  0020afe8  2**3
                  ALLOC
  5 .iram0.text   0000c84e  40800000  40800000  00001000  2**8
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  6 .iram0.text_end 00000000  4080c84e  4080c84e  0020a560  2**0
                  CONTENTS
  7 .iram0.data   00000002  4080c84e  4080c84e  0000d84e  2**0
                  ALLOC
  8 .iram0.bss    00000000  4080c850  4080c850  0020a560  2**0
                  CONTENTS
  9 .dram0.dummy  0000c850  40800000  40800000  0000e000  2**0
                  ALLOC
 10 .dram0.data   00001f30  4080c850  4080c850  0000d850  2**3
                  CONTENTS, ALLOC, LOAD, DATA
 11 .noinit       00000000  4080e780  4080e780  00000000  2**0
                  ALLOC
 12 .dram0.bss    00001598  4080e780  4080e780  0000f780  2**4
                  ALLOC
 13 .flash.text   0016f3b6  42000020  42000020  00010020  2**1
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
 14 iram1.queue_as_raw 00000016  4216f3d6  4216f3d6  0017f3d6  2**1
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
 15 iram1.queue_recv_front 0000011a  4216f3ec  4216f3ec  0017f3ec  2**1
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
 16 .flash_rodata_dummy 00170000  42000020  42000020  00180020  2**0
                  ALLOC
 17 .flash.appdesc 00000100  42170020  42170020  00180020  2**4
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
 18 .flash.rodata 000333d8  42170120  42170120  00180120  2**4
                  CONTENTS, ALLOC, LOAD, DATA
 19 .eh_frame     00057068  421a34f8  421a34f8  001b34f8  2**3
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
 20 .flash.rodata_noload 00000000  421fa560  421fa560  0020a560  2**0
                  CONTENTS
 21 .dram0.heap_start 00000008  4080fd18  4080fd18  0000f780  2**0
                  ALLOC
 22 .debug_loc    0007f8f6  00000000  00000000  0020a560  2**0
                  CONTENTS, READONLY, DEBUGGING, OCTETS
 23 .debug_abbrev 00037efb  00000000  00000000  00289e56  2**0
                  CONTENTS, READONLY, DEBUGGING, OCTETS
 24 .debug_info   00588cb8  00000000  00000000  002c1d51  2**0
                  CONTENTS, READONLY, DEBUGGING, OCTETS
 25 .debug_aranges 0003fee8  00000000  00000000  0084aa10  2**3
                  CONTENTS, READONLY, DEBUGGING, OCTETS
 26 .debug_ranges 00077b30  00000000  00000000  0088a8f8  2**0
                  CONTENTS, READONLY, DEBUGGING, OCTETS
 27 .debug_str    00694ced  00000000  00000000  00902428  2**0
                  CONTENTS, READONLY, DEBUGGING, OCTETS
 28 .comment      00000094  00000000  00000000  00f97115  2**0
                  CONTENTS, READONLY
 29 .riscv.attributes 0000004d  00000000  00000000  00f971a9  2**0
                  CONTENTS, READONLY
 30 .debug_line   0034d211  00000000  00000000  00f971f6  2**0
                  CONTENTS, READONLY, DEBUGGING, OCTETS
 31 .debug_frame  00012e40  00000000  00000000  012e4408  2**2
                  CONTENTS, READONLY, DEBUGGING, OCTETS
 32 .debug_line_str 00002402  00000000  00000000  012f7248  2**0
                  CONTENTS, READONLY, DEBUGGING, OCTETS
 33 .debug_loclists 0000f5c9  00000000  00000000  012f964a  2**0
                  CONTENTS, READONLY, DEBUGGING, OCTETS
 34 .debug_rnglists 00001d7e  00000000  00000000  01308c13  2**0
                  CONTENTS, READONLY, DEBUGGING, OCTETS

```

</details>

When I run `espflash flash --monitor --partition-table ./partitions.csv <elf>`, I get the following output:

```
Chip type:         esp32c6 (revision v0.0)
Crystal frequency: 40MHz
Flash size:        8MB
...
App/part. size:    2,205,072/4,194,304 bytes, 52.57%
...
I (41) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (42) boot: compile time Jun  7 2023 08:02:08
I (43) boot: chip revision: v0.0
I (46) boot.esp32c6: SPI Speed      : 40MHz
I (51) boot.esp32c6: SPI Mode       : DIO
I (56) boot.esp32c6: SPI Flash Size : 4MB
I (61) boot: Enabling RNG early entropy source...
E (66) flash_parts: partition 2 invalid - offset 0x10000 size 0x400000 exceeds flash chip size 0x400000
E (76) boot: Failed to verify partition table
E (81) boot: load partition table error!
```

Note how the bootloader does not detect the chip size correctly. This is probably a small bug in espflash somewhere.

If I now invoke `espflash flash --monitor --partition-table ./partitions.csv --flash-size 8mb <elf>` I get the following output:
```
...
Flash size:        8MB
...
I (41) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (42) boot: compile time Jun  7 2023 08:02:08
I (43) boot: chip revision: v0.0
I (46) boot.esp32c6: SPI Speed      : 40MHz
I (51) boot.esp32c6: SPI Mode       : DIO
I (56) boot.esp32c6: SPI Flash Size : 8MB
I (61) boot: Enabling RNG early entropy source...
I (66) boot: Partition Table:
I (70) boot: ## Label            Usage          Type ST Offset   Length
I (77) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (84) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (92) boot:  2 factory          factory app      00 00 00010000 00400000
I (99) boot: End of partition table
I (103) esp_image: segment 0: paddr=00010020 vaddr=42000020 size=16f3b8h (1504184) map
I (417) esp_image: segment 1: paddr=0017f3e0 vaddr=40800000 size=0e780h ( 59264) load
E (432) esp_image: invalid segment length 0x1866
E (432) boot: Factory app partition is not bootable
E (433) boot: No bootable app partitions in the partition table
```

Now the bootloader detects the flash size correctly, but cannot load the segments.  
Without investigating in detail, to me it looks that we still have some problems with merging the segments.

I'm using espflash from  todays master branch (074f8bcdbcbc5890909932ec99a157837202f025)

SergioGasquez, 2024-01-31:
Hi!

The default `SPI Flash Size` is 4MB, that's the reason of the first log. Do you mind enabling debug logs `RUST_LOG=debug espflash...` so we see logs like the [`spi_autodetect`](https://github.com/esp-rs/espflash/blob/main/espflash/src/flasher/mod.rs#L726-L766)? Have you tried reproducing the same behavior with `esptool`, you would need to download the bootloader from https://github.com/esp-rs/espflash/tree/main/espflash/resources/bootloaders

SergioGasquez, 2024-02-20:
Hello @t-moe, could you try reproducing the behavior with `esptool` and enabling debug logs as suggested?

t-moe, 2024-02-21:
Hi @SergioGasquez ,
Thanks for following up here. 

I'll answer in two comments.

# First bug

If I run
`RUST_LOG=debug espflash flash --monitor <elf> --partition-table ./partitions.csv ` (**without** `--flash-size 8mb`)   

I get   the following error after boot:

```
I (22) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (23) boot: compile time Jun  7 2023 08:02:08
I (24) boot: chip revision: v0.0
I (28) boot.esp32c6: SPI Speed      : 40MHz
I (32) boot.esp32c6: SPI Mode       : DIO
I (37) boot.esp32c6: SPI Flash Size : 4MB
I (42) boot: Enabling RNG early entropy source...
E (47) flash_parts: partition 3 invalid - offset 0x10000 size 0x600000 exceeds flash chip size 0x400000
E (57) boot: Failed to verify partition table
E (62) boot: load partition table error!

```

<details>
<summary>
Full Log output for spi autodetect stuff
</summary>

```
[2024-02-21T16:02:08Z DEBUG] Flash(
        FlashArgs {
            connect_args: ConnectArgs {
                after: HardReset,
                baud: None,
                before: DefaultReset,
                chip: None,
                confirm_port: false,
                no_stub: false,
                port: None,
            },
            flash_config_args: FlashConfigArgs {
                flash_freq: None,
                flash_mode: None,
                flash_size: None,
            },
            flash_args: FlashArgs {
                bootloader: None,
                erase_parts: None,
                erase_data_parts: None,
                log_format: Serial,
                min_chip_rev: 0,
                monitor: true,
                monitor_baud: None,
                partition_table: Some(
                    "./partitions.csv",
                ),
                target_app_partition: None,
                partition_table_offset: None,
                ram: false,
                no_verify: false,
                no_skip: false,
            },
            image: "target/riscv32imac-esp-espidf/debug/esp32_devkit_wifi",
        },
    )
[2024-02-21T16:02:08Z DEBUG] connecting to crates.io:443 at 13.224.103.6:443
[2024-02-21T16:02:08Z DEBUG] No cached session for DnsName("crates.io")
[... 123 lines cut ...]
[0;31mE (63) boot: load partition table error![0m
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x3 (LP_SW_HPSYS),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x4001974a
SPIWP:0xee
mode:DIO, clock div:2
load:0x4086c410,len:0xd48
load:0x4086e610,len:0x2d68
load:0x40875720,len:0x1800
entry 0x4086c410
[0;32mI (22) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader[0m
[0;32mI (23) boot: compile time Jun  7 2023 08:02:08[0m
[0;32mI (24) boot: chip revision: v0.0[0m
[0;32mI (28) boot.esp32c6: SPI Speed      : 40MHz[0m
[0;32mI (32) boot.esp32c6: SPI Mode       : DIO[0m
[0;32mI (37) boot.esp32c6: SPI Flash Size : 4MB[0m
[0;32mI (42) boot: Enabling RNG early entropy source...[0m
[0;31mE (47) flash_parts: partition 3 invalid - offset 0x10000 size 0x600000 exceeds flash chip size 0x400000[0m
[0;31mE (57) boot: Failed to verify partition table[0m
[0;31mE (62) boot: load partition table error![0m
<..loops forever...>

```
</details>

t-moe, 2024-02-21:
# Second Bug

If I run
`RUST_LOG=debug espflash flash --monitor <elf> --partition-table ./partitions.csv  --flash-size 8mb`  (** with `--flash-size`) 

I get  the following, different error after boot:

```
I (22) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (23) boot: compile time Jun  7 2023 08:02:08
I (24) boot: chip revision: v0.0
I (28) boot.esp32c6: SPI Speed      : 40MHz
I (32) boot.esp32c6: SPI Mode       : DIO
I (37) boot.esp32c6: SPI Flash Size : 8MB
I (42) boot: Enabling RNG early entropy source...
I (47) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00003000
I (66) boot:  1 nvs_prod         WiFi data        01 02 0000c000 00003000
I (73) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (80) boot:  3 factory          factory app      00 00 00010000 00600000
I (88) boot: End of partition table
I (92) esp_image: segment 0: paddr=00010020 vaddr=42000020 size=1e7c48h (1997896) map
I (506) esp_image: segment 1: paddr=001f7c70 vaddr=40800000 size=0fff0h ( 65520) load
I (522) esp_image: segment 2: paddr=00207c68 vaddr=4080ffee size=06f2ch ( 28460) load
I (530) esp_image: segment 3: paddr=0020eb9c vaddr=40816f20 size=037e0h ( 14304) load
E (534) esp_image: invalid segment length 0x58da
E (535) boot: Factory app partition is not bootable
E (539) boot: No bootable app partitions in the partition table
```

<details>
<summary>
Full Log output 
</summary>

```
[2024-02-21T16:10:05Z DEBUG] Flash(
        FlashArgs {
            connect_args: ConnectArgs {
                after: HardReset,
                baud: None,
                before: DefaultReset,
                chip: None,
                confirm_port: false,
                no_stub: true,
                port: None,
            },
            flash_config_args: FlashConfigArgs {
                flash_freq: None,
                flash_mode: None,
                flash_size: Some(
                    _8Mb,
                ),
            },
            flash_args: FlashArgs {
                bootloader: None,
                erase_parts: None,
                erase_data_parts: None,
                log_format: Serial,
                min_chip_rev: 0,
                monitor: true,
                monitor_baud: None,
                partition_table: Some(
                    "./partitions.csv",
                ),
                target_app_partition: None,
                partition_table_offset: None,
                ram: false,
                no_verify: false,
[... 145 lines cut ...]
[0;32mI (24) boot: chip revision: v0.0[0m
[0;32mI (28) boot.esp32c6: SPI Speed      : 40MHz[0m
[0;32mI (32) boot.esp32c6: SPI Mode       : DIO[0m
[0;32mI (37) boot.esp32c6: SPI Flash Size : 8MB[0m
[0;32mI (42) boot: Enabling RNG early entropy source...[0m
[0;32mI (47) boot: Partition Table:[0m
[0;32mI (51) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (58) boot:  0 nvs              WiFi data        01 02 00009000 00003000[0m
[0;32mI (66) boot:  1 nvs_prod         WiFi data        01 02 0000c000 00003000[0m
[0;32mI (73) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (80) boot:  3 factory          factory app      00 00 00010000 00600000[0m
[0;32mI (88) boot: End of partition table[0m
[0;32mI (92) esp_image: segment 0: paddr=00010020 vaddr=42000020 size=1e7c48h (1997896) map[0m
[0;32mI (506) esp_image: segment 1: paddr=001f7c70 vaddr=40800000 size=0fff0h ( 65520) load[0m
[0;32mI (522) esp_image: segment 2: paddr=00207c68 vaddr=4080ffee size=06f2ch ( 28460) load[0m
[0;32mI (530) esp_image: segment 3: paddr=0020eb9c vaddr=40816f20 size=037e0h ( 14304) load[0m
[0;31mE (534) esp_image: invalid segment length 0x58da[0m
[0;31mE (535) boot: Factory app partition is not bootable[0m
[0;31mE (539) boot: No bootable app partitions in the partition table[0m
<loops forever>

```
</details>

Ping me on matrix if you need the elf file or the full output with all flashdefl data sections.

t-moe, 2024-02-21:
Regarding trying with esptool, I assume you mean something like the following:

Starting with the elf from the previous step:

```
espflash partition-table --to-binary  partitions.csv   > partition_table.bin
espflash save-image --chip esp32c6 --flash-size 8mb <elf> text.bin
esptool.py -p /dev/ttyACM0  -b 460800 --before default_reset --after hard_reset --chip esp32c6 write_flash --flash_mode dio --flash_size 8MB --flash_freq 40m 0x0 ~/Downloads/esp32c6-bootloader.bin 0x8000 partition_table.bin 0x10000 text.bin
```

But this outputs a warning during flashing  
```Warning: Image file at 0x0 is protected with a hash checksum, so not changing the flash size setting. Use the --flash_size=keep option instead of --flash_size=8MB in order to remove this warning, or use the --dont-append-digest option for the elf2image command in order to generate an image file without a hash checksum
``` 

and later `espflash monitor` shows:
```
I (22) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (23) boot: compile time Jun  7 2023 08:02:08
I (24) boot: chip revision: v0.0
I (28) boot.esp32c6: SPI Speed      : 40MHz
I (32) boot.esp32c6: SPI Mode       : DIO
I (37) boot.esp32c6: SPI Flash Size : 2MB
I (42) boot: Enabling RNG early entropy source...
E (47) flash_parts: partition 3 invalid - offset 0x10000 size 0x600000 exceeds flash chip size 0x200000
E (57) boot: Failed to verify partition table
E (62) boot: load partition table error!

```

How do I have to adapt my steps to test the thing with esptool?  Thank you

SergioGasquez, 2024-02-26:
> How do I have to adapt my steps to test the thing with esptool? Thank you

I think the steps would be the following:
1. 
```
esptool.py --chip esp32c6 elf2image  --flash_size 8MB <elf>
```
You may need adding ` --dont-append-digest` flag. More info in https://docs.espressif.com/projects/esptool/en/latest/esp32c3/esptool/basic-commands.html?highlight=elf2image#convert-elf-to-binary-elf2image
2. espflash partition-table --to-binary  partitions.csv   > partition_table.bin
  - I think you should also be able to pass the csv partition table to esptool, but just in case.
3. 
```
esptool.py -p /dev/ttyACM0  -b 460800 --before default_reset --after hard_reset --chip esp32c6 write_flash --flash_mode dio --flash_size 8MB --flash_freq 40m 0x0 <default_espflash_c6_booloader> 0x8000 partition_table.bin 0x10000 <esptool_generated_bin>
```
4. `espflash monitor`

SergioGasquez, 2024-02-27:
I just did another test, using a freshly generated template (targetting esp32c6 and not using advanced config), I tried using your partition table and flash it with espflash and esptool:
```
❯ espflash flash -M --flash-size 8mb --partition-table /home/sergio/Documents/Espressif/forks/espflash/issue552/partition.csv target/riscv32imac-unknown-none-elf/debug/esp32c6
[2024-02-27T10:42:29Z INFO ] Serial port: '/dev/ttyUSB0'
[2024-02-27T10:42:29Z INFO ] Connecting...
[2024-02-27T10:42:30Z INFO ] Using flash stub
[2024-02-27T10:42:31Z WARN ] Setting baud rate higher than 115,200 can cause issues
Chip type:         esp32c6 (revision v0.0)
Crystal frequency: 40 MHz
Flash size:        8MB
Features:          WiFi 6, BT 5
MAC address:       60:55:f9:f7:1b:6c
Partition table:   /home/sergio/Documents/Espressif/forks/espflash/issue552/partition.csv
App/part. size:    211,648/6,291,456 bytes, 3.36%
[2024-02-27T10:42:31Z INFO ] Segment at address '0x0' has not changed, skipping write
[2024-02-27T10:42:31Z INFO ] Segment at address '0x8000' has not changed, skipping write
[2024-02-27T10:42:31Z INFO ] Segment at address '0x10000' has not changed, skipping write
[2024-02-27T10:42:31Z INFO ] Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x4086c410,len:0xd48
load:0x4086e610,len:0x2d68
load:0x40875720,len:0x1800
entry 0x4086c410
I (23) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (24) boot: compile time Jun  7 2023 08:02:08
I (25) boot: chip revision: v0.0
I (29) boot.esp32c6: SPI Speed      : 40MHz
I (33) boot.esp32c6: SPI Mode       : DIO
I (38) boot.esp32c6: SPI Flash Size : 8MB
I (43) boot: Enabling RNG early entropy source...
I (49) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00003000
I (67) boot:  1 nvs_prod         WiFi data        01 02 0000c000 00003000
I (74) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (82) boot:  3 factory          factory app      00 00 00010000 00600000
I (89) boot: End of partition table
I (93) esp_image: segment 0: paddr=00010020 vaddr=42000020 size=2d9d0h (186832) map
I (140) esp_image: segment 1: paddr=0003d9f8 vaddr=40800000 size=00020h (    32) load
I (140) esp_image: segment 2: paddr=0003da20 vaddr=4202da20 size=04da4h ( 19876) map
I (150) esp_image: segment 3: paddr=000427cc vaddr=40800020 size=012c4h (  4804) load
I (155) boot: Loaded app from partition at offset 0x10000
I (160) boot: Disabling RNG early entropy source...
Hello world!
Loop...
```
With esptool:
```
❯ esptool.py -p /dev/ttyUSB0 --before default_reset --after hard_reset --chip esp32c6  write_flash --flash_size 8MB 0x0 /home/sergio/Documents/Espressif/forks/espflash/issue552/esp32c6-bootloader.bin 0x8000 /home/sergio/Documents/Espressif/forks/espflash/issue552/partition.csv 0x10000 target/riscv32imac-unknown-none-elf/debug/esp32c6
esptool.py v4.6.2
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP32-C6 (QFN40) (revision v0.0)
Features: WiFi 6, BT 5, IEEE802.15.4
Crystal is 40MHz
MAC: 60:55:f9:00:00:f7:1b:6c
BASE MAC: 60:55:f9:f7:1b:6c
MAC_EXT: 00:00
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x00010000 to 0x00603fff...
Warning: Image file at 0x0 is protected with a hash checksum, so not changing the flash size setting. Use the --flash_size=keep option instead of --flash_size=8MB in order to remove this warning, or use the --dont-append-digest option for the elf2image command in order to generate an image file without a hash checksum
Compressed 21264 bytes to 12899...
Wrote 21264 bytes (12899 compressed) at 0x00000000 in 1.5 seconds (effective 116.8 kbit/s)...
Hash of data verified.
Compressed 216 bytes to 130...
Wrote 216 bytes (130 compressed) at 0x00008000 in 0.1 seconds (effective 30.9 kbit/s)...
Hash of data verified.
Compressed 6239616 bytes to 1621672...
Wrote 6239616 bytes (1621672 compressed) at 0x00010000 in 142.7 seconds (effective 349.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

esp32c6 on  main [?] is 📦 v0.1.0 via 🦀 v1.78.0-nightly took 2m36s
❯ espflash monitor
[2024-02-27T10:46:29Z INFO ] Serial port: '/dev/ttyUSB0'
[2024-02-27T10:46:29Z INFO ] Connecting...
[2024-02-27T10:46:30Z INFO ] Using flash stub
[2024-02-27T10:46:31Z WARN ] Setting baud rate higher than 115,200 can cause issues
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

�ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x4086c410,len:0xd48
load:0x4086e610,len:0x2d68
load:0x40875720,len:0x1800
entry 0x4086c410
I (23) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (24) boot: compile time Jun  7 2023 08:02:08
I (25) boot: chip revision: v0.0
I (29) boot.esp32c6: SPI Speed      : 40MHz
I (33) boot.esp32c6: SPI Mode       : DIO
I (38) boot.esp32c6: SPI Flash Size : 2MB
I (43) boot: Enabling RNG early entropy source...
E (48) flash_parts: partition 0 invalid magic number 0x2023
E (54) boot: Failed to verify partition table
E (59) boot: load partition table error!
```

espflash is able to flash and run the project properly using your custom partition table while esptool complains about the magic number of partition 0

SergioGasquez, 2024-03-06:
A small summary with some findings:
- Flashing the provided elf and partition table results in the same as flashing the provided elf and a default partition table:
    - In espflash, it flashes but `No bootable app partitions in the partition table`
    - In esptool, it fails flashing the provided elf with `File <elf> (length 30640700) at offset 65536 will not fit in 8388608 bytes of flash. Use --flash_size argument, or change flashing address.`
- Flashing a small app with the provided partition table, fails to use the partition table:
    - In espflash, flash succeeded but `Factory app partition is not bootable. Factory app partition is not bootable`
    - In esptool, flash succeeded but `partition 0 invalid magic number 0x2023.Failed to verify partition table. load partition table error!`
    
Not sure if this is an `espflash` issue as similar things happen in `esptool` too, but might be worth to investigate more.

MabezDev, 2025-02-26:
> In esptool, it fails flashing the provided elf with File <elf> (length 30640700) at offset 65536 will not fit in 8388608 bytes of flash. Use --flash_size argument, or change flashing address.

esptool won't automatically convert elf files, like espflash does. To see if esptool can handle the elf, use `esptool.py elf2image`. To verify the image, you can then use `esptool.py image_info`.

bugadani, 2025-04-04:
esptool is unable to convert the elf:

> A fatal error occurred: Segment loaded at 0x421e7c7c lands in same 64KB flash mapping as segment loaded at 0x421e7c66. Can't generate binary. Suggest changing linker script or ELF to merge sections.

the espsegs output contains this:

`iram1.queue_as_raw     421e7c66` which is a function in its own segment, aligned to 2 bytes instead of 4. The bootloader validates that each section is 4-byte aligned, which this is not, so it rejects the binary. I believe this is an esp-idf issue, possibly related to the compressed (2-byte) instruction set used by the ESP32-C6.

bugadani, 2025-04-04:
As far as I can tell #831 fixes this issue, but I guess it's a bit late to ask for testing that PR. 🙈

t-moe, 2025-04-04:
> As far as I can tell [#831](https://github.com/esp-rs/espflash/pull/831) fixes this issue, but I guess it's a bit late to ask for testing that PR. 🙈

ehm, yeah, I'm in a different project right now. But I promise to retest this, as soon as I'm working with esp again.. 
You can close this issue in the meantime, if you think it should be resolved.

PR #831 (2025-04-04): Align segment lengths after merging
Fixes #552 - the problem is that the reproducer elf has 2-byte aligned segments (functions in IRAM), and espflash used to 4-byte align segment lengths prematurely, causing them to overlap. This overlap then prevented merging segments, and the end result was an elf that the bootloader was not able to load.

The firmware seems to require some propriatery, or at least project-specific data to be present, so the firmware immediately panics, but it can boot with this PR.
closed 2025-04-04 by jessebraham