esp-rs/espflash #557: espflash::timeout on esp32c6 via JTAG uart

t-moe, 2024-01-22:
I'm using espflash master to flash a esp-idf-* (master) project with espflash to a esp32c6.

```
espflash flash --monitor <ELF>
[2024-01-22T08:43:49Z INFO ] Serial port: '/dev/ttyACM0'
[2024-01-22T08:43:49Z INFO ] Connecting...
[2024-01-22T08:43:49Z INFO ] Using flash stub
Error: espflash::timeout

  × Error while connecting to device
  ╰─▶ Timeout while running ReadReg command
```

Some more debug output of espflash

```
[2024-01-22T08:43:36Z DEBUG] Finish stub write
[2024-01-22T08:43:36Z DEBUG] Writing command: MemEnd { no_entry: false, entry: 1082130432 }
[2024-01-22T08:43:36Z DEBUG] Stub written!
[2024-01-22T08:43:36Z DEBUG] Writing command: ReadReg { address: 1073745920 }
Error: espflash::timeout
```

I've recompiled the stubs with dprint enabled, and I get the following output from the stub:

```
Stub init! Transport detected: UsbSerialJtag
Stub sending greeting!
Waiting for command
```

I've also slightly modified src/flasher/mod.rs (espflash)

```
         debug!("Stub written!");
 
         match self.connection.read(EXPECTED_STUB_HANDSHAKE.len())? {
             Some(resp) if resp == EXPECTED_STUB_HANDSHAKE.as_bytes() => Ok(()),
             _ => Err(Error::Connection(ConnectionError::InvalidStubHandshake)),
         }?;
 
+        debug!("Handshake ok");
+        std::thread::sleep(Duration::from_secs(3));
+
         // Re-detect chip to check stub is up
         let magic = self.connection.read_reg(CHIP_DETECT_MAGIC_REG_ADDR)?;
         let chip = Chip::from_magic(magic)?;
         debug!("Re-detected chip: {:?}", chip);
 
         Ok(())
```

It seems the handshake is written, but then the chip redetect fails.

When I flash the same binary via Uart0 instead of USB JTAG, it works...

t-moe, 2024-01-22:
I'm still trying to reproduce this reliably. This issue here appears at random, and when it appears, it persists until ??.  Same goes for #556, which also appears at random and then persists until ??.

Not sure what ?? actually is. Maybe:
* erase the flash
* power cycle the chip
* put the chip into bootloader mode manually
* flash a different firmware
* or a combination of the above

Investigating....

SergioGasquez, 2024-01-22:
Can you try using `--no-stub` and see if the issue persists?

t-moe, 2024-01-22:
I can confirm that it flashes , boots as expected with `--no-stub` in this case. That is, given I get past the "waiting for download" issue (#556 )...