esp-rs/espflash #569: cargo espflash behaves differently to espflash when trying to flash bins > 1MB

Vollbrecht, 2024-02-01:
We promoting the usage of `cargo espflash` for simple and easy testing of --examples in a repository. Unfortunately when one of the examples is bigger than 1MB than `cargo espflash` wont be able to flash it. For example the usage of `cargo espflash flash --target riscv32imc-esp-espidf --example wifi --monitor` inside esp-idf-svc. 

The error that is reported  is as follows
```
Error: espflash::image_too_big (link)

  × Supplied ELF image of 1802560B is too big, and doesn't fit configured app partition of 1048576B
  help: Reduce the size of the binary or increase the size of the app partition.
  ```
  
  `espflash` is able the handle that issue and autoincrease the app partition itself. I think it would make sense if `cargo espflash` behave the same.

SergioGasquez, 2024-02-02:
[`cargo-espflash` uses the partition-table generated from ESP-IDF project built](https://github.com/esp-rs/espflash/tree/main/cargo-espflash#bootloader-and-partition-table), while `espflash` generates a partition table using `esp-idf-part` and uses it. If you add the partition table arg to espflash and point to the esp-idf generated one, the result should be the same.

MabezDev, 2024-02-03:
I think we shouldn't do anything here. I don't think we want to stop the default behaviour of using the bootloader from esp-idf-sys, therefore I think you should add a partitions.csv with a bigger default partition to run your examples.

Feel free to reopen if you have a better suggestion :).

closed 2024-02-03 by MabezDev

Vollbrecht, 2024-02-04:
I am fine with it, it was just a bit anoying that with `cargo espflash` you get a new option -> automatically getting the esp-idf-sys build partition.csv , but you cant get the feature for auto-resizing, the only alternative is that you have to manually specify the partition.csv.

On that note, i now would use the `espflash.toml` to setup the configuration in idf-hal/svc to workaround the issue. You talked about removing the .toml and unify the places for settings?

SergioGasquez, 2024-02-05:
> On that note, i now would use the espflash.toml to setup the configuration in esp-idf-hal/svc to workaround the issue. You were talking about removing the .toml and unify the places for settings?

Not sure what you mean by removing the `.toml`, but we only support `espflash.toml` settings now (locals or global), that means that, when using cargo-espflash, you can't specify settings in the package metadata anymore.