esp-rs/espflash #571: [feature request] flashing from monitor

JohnnyZig, 2024-02-02:
Is it possible to implement flashing when we are already in monitor?
That would optimize fast recompile&flash&try it cycle (without needing to leave espflash monitor).
Similar like esp-idf.py monitor does with Ctrl-] Ctrl-F?
I tried to implement it myself my my knowledge of Rust is not enough.

MabezDev, 2024-02-02:
This would only be possible in cargo-espflash I believe, as espflash by itself doesn't know how to rebuild the project.

IMO this _probably_ isn't worth the effort, at least my workflow is only two extra key presses:

1. Ctrl+C to quit the monitor
2. Up arrow to load the last command (cargo run blah blah)
3. Enter to build, flash and open the monitor.

Perhaps if someone submitted a PR we might accept it, but most likely something we won't work on ourselves.

SergioGasquez, 2024-02-05:
As @MabezDev we probably won't work on this in our end, so I will close this issue as not planned. That being said, feel free to send a PR with this feature.

closed 2024-02-05 by SergioGasquez