esp-rs/espflash #574: flash --monitor throws stack trace but works ok eventually

dlakatos847, 2024-02-05:
Hello,

I use an ESP32-C6-DevKitC-1 development board for educational purposes.

When I build the project then flash it, there is a difference between the two scenarios:

1. `espflash flash target/riscv32imac-unknown-none-elf/debug/flashit` then `espflash monitor` + CTRL+R works as expected
2. `espflash flash target/riscv32imac-unknown-none-elf/debug/flashit --monitor` shows this output:

```
$ espflash flash target/riscv32imac-unknown-none-elf/debug/flashit --monitor
[2024-02-05T09:51:22Z INFO ] Serial port: '/dev/ttyUSB0'
[2024-02-05T09:51:22Z INFO ] Connecting...
[2024-02-05T09:51:23Z INFO ] Using flash stub
Chip type:         esp32c6 (revision v0.0)
Crystal frequency: 40MHz
Flash size:        8MB
Features:          WiFi 6, BT 5
MAC address:       40:4c:ca:42:5b:80
App/part. size:    256,048/8,323,072 bytes, 3.08%
[00:00:01] [========================================]      13/13      0x0                                                                                                                   [00:00:00] [========================================]       1/1       0x8000                                                                                                                [00:00:08] [========================================]      91/91      0x10000                                                                                                               [2024-02-05T09:51:34Z INFO ] Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x4086c410,len:0xd48
0x4086c410 - __global_pointer$
    at ??:??
load:0x4086e610,len:0x2d68
0x4086e610 - _stack_start
    at ??:??
load:0x40875720,len:0x1800
0x40875720 - _stack_start
    at ??:??
SHA-256 comparison failed:
Calculated: 377c26481ad6ab0cce508450f353a55f4bd7eb9159aaf85fde3111071f164514
Expected: 0af544a033ab3492852b8232c904c578d5f07c9d4a423a64473f060db374ab32
Attempting to boot anyway...
entry 0x4086c410
0x4086c410 - __global_pointer$
    at ??:??
I (42) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (42) boot: compile time Jun  7 2023 08:02:08
I (43) boot: chip revision: v0.0
I (47) boot.esp32c6: SPI Speed      : 40MHz
I (52) boot.esp32c6: SPI Mode       : DIO
I (57) boot.esp32c6: SPI Flash Size : 8MB
I (61) boot: Enabling RNG early entropy source...
I (67) boot: Partition Table:
I (70) boot: ## Label            Usage          Type ST Offset   Length
I (78) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (85) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (92) boot:  2 factory          factory app      00 00 00010000 007f0000
I (100) boot: End of partition table
I (104) esp_image: segment 0: paddr=00010020 vaddr=42000020 size=36f50h (225104) map
I (159) esp_image: segment 1: paddr=00046f78 vaddr=40800000 size=00020h (    32) load
I (159) esp_image: segment 2: paddr=00046fa0 vaddr=42036fa0 size=06804h ( 26628) map
I (170) esp_image: segment 3: paddr=0004d7ac vaddr=40800020 size=01058h (  4184) load
I (174) boot: Loaded app from partition at offset 0x10000
I (179) boot: Disabling RNG early entropy source...
Hello world!
Loop...
Loop...
Loop...
...
```

I wonder why this output appears only when using the `--monitor` flag:

```
load:0x4086c410,len:0xd48
0x4086c410 - __global_pointer$
    at ??:??
load:0x4086e610,len:0x2d68
0x4086e610 - _stack_start
    at ??:??
load:0x40875720,len:0x1800
0x40875720 - _stack_start
    at ??:??
```

and

```
Attempting to boot anyway...
entry 0x4086c410
0x4086c410 - __global_pointer$
    at ??:??
```

Thank you.

Vollbrecht, 2024-02-05:
if you use `espflash monitor` standalone you also need to add the path to your .elf as without it `espflash` cant decode the stacktrace

SergioGasquez, 2024-02-05:
Also, looking at the logs and bootloader version, you are probably using `espflash 2.1`. You are also encountering https://github.com/esp-rs/espflash/issues/206 which is now fixed.

closed 2024-02-16 by SergioGasquez