esp-rs/espflash #580: espflash detects wrong flash size

brookman, 2024-02-11:
I have an esp-c3-32s-kit-v1.0 (version a) see [PDF page 11 ](https://docs.ai-thinker.com/_media/esp32/docs/esp-c3-32s-kit-v1.0_specification.pdf).

When flashing with espflash it prints this:

```
Chip type:         esp32c3 (revision v0.3)
Crystal frequency: 40MHz
Flash size:        2MB
Features:          WiFi, BLE
MAC address:       [redacted]
App/part. size:    1,071,792/2,031,616 bytes, 52.76%
[00:00:01] [========================================]      13/13      0x0                                                                                                    
[00:00:00] [========================================]       1/1       0x8000                                                                                                 
[00:01:03] [========================================]     608/608     0x10000                                                                                                
[2024-02-11T09:46:00Z INFO ] Flashing has completed!
```

It says 2MB, however the board is supposed to have 4MB.

I also have the issue that I can only flash and run it by this process:
- unplug the USB cable
- press and hold both buttons
- plug in the USB
- release the buttons
- flash
- unplug the USB cable
- plug in the USB

This only started happening recently. Before I could flash it without this workaround. I suspect something went wrong with the flash size or layout at some point.

MabezDev, 2024-02-11:
The datasheet for the board also says there is a 2MB verison. Could you check what esptool.py detects?

brookman, 2024-02-11:
Dear @MabezDev, I checked again and it is indeed the 2MB version. 🫢
I'm terribly sorry!

closed 2024-02-11 by brookman