esp-rs/espflash #583: cargo runner `error: unexpected argument`

phemmer, 2024-02-15:
Using the template `esp-rs/esp-template` to create a project defaults to using `espflash`. However the "Rust on ESP book" says that `cargo-espflash` should be preferred. After installing `cargo-espflash` and trying to switch to it, it generates an error:

```
# cargo run
    Finished dev [unoptimized + debuginfo] target(s) in 0.12s
     Running `cargo espflash flash -p '/dev/serial/by-id/usb-Espressif_USB_JTAG_serial_debug_unit_54:32:04:07:2E:94-if00' --monitor target/riscv32imac-unknown-none-elf/debug/pumps`
error: unexpected argument 'target/riscv32imac-unknown-none-elf/debug/pumps' found

Usage: cargo espflash flash [OPTIONS]

For more information, try '--help'.
```

From the following runner config in `config.toml`:
```
[target.riscv32imac-unknown-none-elf]
runner = "cargo espflash flash -p /dev/serial/by-id/usb-Espressif_USB_JTAG_serial_debug_unit_54:32:04:07:2E:94-if00 --monitor"
```

Looking at the `--help` of the two commands, it seems there is a difference in that `espflash` takes a path to an image as a positional argument, while `cargo-espflash` does not. It also does not appear that any of the flags (e.g. `--bin`) accept the path that `cargo` gives it.

Can we get some documentation on how to properly integrate `cargo-espflash` into the cargo config as a runner?

SergioGasquez, 2024-02-15:
Hi! `cargo-espflash` cant be used as a Cargo custom runner, if you do `cargo espflash flash --monitor` it will build, flash and monitor. `cargo espflash` basically adds `espflash` the building features. Hence, if you want to use a custom runner, use `espflash`.

> However the "Rust on ESP book" says that cargo-espflash should be preferred. 

Can you point me to where this is in the book?

phemmer, 2024-02-15:
https://esp-rs.github.io/book/troubleshooting/espflash.html

> You should also prefer using `cargo-espflash` over `espflash`.

SergioGasquez, 2024-02-15:
That statement does not applies to you:

> [If you are building an esp-idf-sys based project](https://esp-rs.github.io/book/troubleshooting/espflash.html#if-you-are-building-an-esp-idf-sys-based-project)

This is not your case, you are not building an `esp-idf-sys` based project.

closed 2024-02-15 by SergioGasquez