esp-rs/espflash #588: Skip reflashing similar blocks

SergioGasquez, 2024-02-16:
If it is possible to read the hash of a particular flash region, it should also be possible I believe to implement selective reflashing: for each flash block, read the hash, compare, and only reflash if differs.

Also, cc https://github.com/esp-rs/espflash/issues/259 as this issue is basically the meat of flash content verification.

_Originally posted by @bugadani in https://github.com/esp-rs/espflash/issues/479#issuecomment-1727965492_

bjoernQ, 2024-02-16:
Will be quite a bit of work so probably we should check how much we can benefit from it beforehand. We could create images via save-image and check the diff

bjoernQ, 2024-02-20:
I did a quick check with an esp-hal example (embassy_i2s_sound.rs on ESP32-C3) and changed a few things between release builds:
- change a gpio number
- changed size of a heap allocated buffer
- add a println

Then I checked the output of `espflash save-image` if there are same flash sectors.

The only sectors which are same are the gaps between code and data (since data needs to be 64k aligned). The fill-byte is `0x00`. When the gap is quite big compared to code and data, then this would give us a nice boost. Otherwise, it won't

I wonder if we could somehow optimize for this in another way? Could we just avoid flashing whatever is in that gap completely? (i.e. no need to even check the md5)

bjoernQ, 2024-02-20:
> I wonder if we could somehow optimize for this in another way? Could we just avoid flashing whatever is in that gap completely? (i.e. no need to even check the md5)

I tried changing the padding byte to 0xff and made the flasher-stub skip the writing in that case. No measurable difference for me. So probably not worth it

AnthonyGrondin, 2024-02-20:
> I did a quick check with an esp-hal example (embassy_i2s_sound.rs on ESP32-C3) and changed a few things between release builds:
>
>     * change a gpio number
>
>     * changed size of a heap allocated [...]

I was afraid the entropy of the generated image file would be too high for this kind of improvement to be possible :(
I might try some experiments around. How do you diff the images for sectors?

MabezDev, 2024-02-22:
Imo this is probably not worth pursuing for the v3 release, removing from the milestone.

jessebraham, 2025-02-19:
I think probably we will not go down this path at this time. We can always revisit this in the future if there is a need/desire for it.

closed 2025-02-19 by jessebraham