esp-rs/espflash #593: `espflash` should look at parent/workspace directories for `espflash.toml`

marsfan, 2024-02-21:
Looking at the release notes for espflash v3.0.0-rc1, it seems that the partition table can now be specified in the `espflash.toml` file. (As added in #551)

However, it seems that `espflash` will only look for this file in the current directory, or the user's main configuration directory ([See these lines of code](https://github.com/esp-rs/espflash/pull/551/files#diff-16aedab0b163d4e6c5350f585959afd77901030799b31de2a9c3c47e64b9c253R105-R114). It should support looking at parent directories so as to support cargo workspaces with multiple crates.  

This is especially useful if `cargo test` is used, as that command will execute in the individual crates sub-directories in the workspace. With the current search method used by `espflash`, this would require creating an additional `espflash.toml` in each crate's directory, instead of a single `espflash.toml` in the workspace directory.

SergioGasquez, 2024-02-21:
Hi @marsfan! Just opened #595, mind testing it and verifying that it solves your request?

closed 2024-02-21 by MabezDev

marsfan, 2024-02-22:
@SergioGasquez Two questions regarding testing. 

1. How do I test a version that is not released to crates.io yet (I'm still learning my way around the Rust ecosystem). 
2. When using options configured in the file (such as the option for a partition table file), will it search from the same directory that the file is in, or the directory that `espflash` was invoked from? I ask this as the `cargo test` command will invoke `espflash` from the individual crates directories, as opposed to the workspace directory.

SergioGasquez, 2024-02-22:
> 1. How do I test a version that is not released to [crates.io](http://crates.io/) yet (I'm still learning my way around the Rust ecosystem).

You have two options:
- [Install the crate from git](https://doc.rust-lang.org/cargo/commands/cargo-install.html#install-options): `cargo install <crate> --git <repo_url>`, you can also use the  `--branch/--rev` args if needed. Now that the pr is merged you could just `cargo install espflash  --git https://github.com/esp-rs/espflash`
- Clone the repo and run it locally with `cargo run`

> 2. When using options configured in the file (such as the option for a partition table file), will it search from the same directory that the file is in, or the directory that espflash was invoked [...]

If you use local path to file, you should specify it from the directory where you are executing `espflash...` I'd say.